Koenkk/zigbee2mqtt #22223: TuYa ZS06 not working

buiducnhat, 2024-04-16:
### What happened?

[image]

My project need to use this device (TuYa ZS06), it can be paired successfully.
But when I change the state to on/off, the error occurs.

[image]

Error detail from logs:
```Publish 'set' 'learn_ir_code' to '0xdc8e95fffef9b164' failed: 'Error: ZCL command 0xdc8e95fffef9b164/1 zosungIRControl.zosungControlIRCommand00({"data":{"type":"Buffer","data":[123,34,115,116,117,100,121,34,58,48,125]}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'```

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20211115

### Adapter

CC2531 stick

### Setup

Plain (NodeJS), Raspberry Pi 4

### Debug log

info  2024-04-16 16:23:44: Logging to console and directory: '/home/pi/projects/zigbee2mqtt/data/log/2024-04-16.16-23-44' filename: log.txt
info  2024-04-16 16:23:44: Starting Zigbee2MQTT version 1.36.1 (commit #ffc2ff1d)
info  2024-04-16 16:23:44: Starting zigbee-herdsman (0.40.3)
info  2024-04-16 16:23:45: zigbee-herdsman started (resumed)
info  2024-04-16 16:23:45: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20211115,"transportrev":2},"type":"zStack12"}'
info  2024-04-16 16:23:45: Currently 5 devices are joined:
info  2024-04-16 16:23:45: 0x84fd27fffe08117c (0x84fd27fffe08117c): HS1DS - HEIMAN Door sensor (EndDevice)
info  2024-04-16 16:23:45: 0xa4c1381f2ec65460 (0xa4c1381f2ec65460): BT400B - TuYa Zigbee Panic Button (EndDevice)
info  2024-04-16 16:23:45: 0xdc8e95fffef9b164 (0xdc8e95fffef9b164): ZS06 - TuYa Universal smart IR remote control (Router)
info  2024-04-16 16:23:45: 0x000d6ffffebf9ede (0x000d6ffffebf9ede): E8332SCN300ZB - Schneider Electric Curtain switch 2 Gang (Router)
info  2024-04-16 16:23:45: 0x000d6ffffe9710b4 (0x000d6ffffe9710b4): Not supported (Router)
info  2024-04-16 16:23:45: Zigbee: disabling joining new devices.
info  2024-04-16 16:23:45: Connecting to MQTT server at mqtt://localhost
info  2024-04-16 16:23:45: Connected to MQTT server
info  2024-04-16 16:23:45: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2024-04-16 16:23:45: Started frontend on port 8080
info  2024-04-16 16:23:45: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"ffc2ff1d","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20211115,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":11,"extendedPanID":"0x00124b0008c0a2ec","panID":6754},"permit_join":false,"version":"1.36.1"}'
info  2024-04-16 16:23:45: MQTT publish: topic 'zigbee2mqtt/0x84fd27fffe08117c', payload '{"battery":84.5,"battery_low":false,"contact":true,"tamper":false}'
info  2024-04-16 16:23:45: MQTT publish: topic 'zigbee2mqtt/0xa4c1381f2ec65460', payload '{"battery":100,"linkquality":13,"voltage":3000}'
info  2024-04-16 16:23:45: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffebf9ede', payload '{"linkquality":0,"state":"ON","state_10":"OFF","state_11":"ON"}'
info  2024-04-16 16:23:45: Zigbee2MQTT started!
error 2024-04-16 16:24:27: Publish 'set' 'state' to '0x000d6ffffebf9ede' failed: 'Error: ZCL command 0x000d6ffffebf9ede/10 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":1,"datatype":4,"data":[0]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
info  2024-04-16 16:24:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x000d6ffffebf9ede' failed: 'Error: ZCL command 0x000d6ffffebf9ede/10 manuSpecificTuya.dataRequest({\"seq\":1,\"dpValues\":[{\"dp\":1,\"datatype\":4,\"data\":[0]}]}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"0x000d6ffffebf9ede"},"type":"zigbee_publish_error"}'
error 2024-04-16 16:24:55: Publish 'set' 'position' to '0x000d6ffffebf9ede' failed: 'Error: ZCL command 0x000d6ffffebf9ede/10 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":2,"datatype":3,"data":[]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
info  2024-04-16 16:24:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'position' to '0x000d6ffffebf9ede' failed: 'Error: ZCL command 0x000d6ffffebf9ede/10 manuSpecificTuya.dataRequest({\"seq\":1,\"dpValues\":[{\"dp\":2,\"datatype\":3,\"data\":[]}]}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"0x000d6ffffebf9ede"},"type":"zigbee_publish_error"}'
error 2024-04-16 16:25:15: Publish 'set' 'position' to '0x000d6ffffebf9ede' failed: 'Error: ZCL command 0x000d6ffffebf9ede/10 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":2,"datatype":3,"data":[]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
info  2024-04-16 16:25:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'position' to '0x000d6ffffebf9ede' failed: 'Error: ZCL command 0x000d6ffffebf9ede/10 manuSpecificTuya.dataRequest({\"seq\":1,\"dpValues\":[{\"dp\":2,\"datatype\":3,\"data\":[]}]}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"0x000d6ffffebf9ede"},"type":"zigbee_publish_error"}'
error 2024-04-16 16:26:10: Publish 'set' 'motor_steering' to '0x000d6ffffebf9ede' failed: 'Error: ZCL command 0x000d6ffffebf9ede/10 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":8,"datatype":4,"data":[0]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
info  2024-04-16 16:26:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'motor_steering' to '0x000d6ffffebf9ede' failed: 'Error: ZCL command 0x000d6ffffebf9ede/10 manuSpecificTuya.dataRequest({\"seq\":1,\"dpValues\":[{\"dp\":8,\"datatype\":4,\"data\":[0]}]}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"0x000d6ffffebf9ede"},"type":"zigbee_publish_error"}'
info  2024-04-16 16:55:03: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/generate_external_definition', payload '{"data":{"id":"0xdc8e95fffef9b164","source":"const {identify, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');\n\nconst definition = {\n    zigbeeModel: ['TS1201'],\n    model: 'TS1201',\n    vendor: '_TZ3290_7v1k4vufotpowp9z',\n    description: 'Automatically generated definition',\n    extend: [identify(), onOff({\"powerOnBehavior\":false})],\n    meta: {},\n};\n\nmodule.exports = definition;"},"status":" [... cut ...] Control.zosungControlIRCommand00({\"data\":{\"type\":\"Buffer\",\"data\":[123,34,115,116,117,100,121,34,58,48,125]}}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"0xdc8e95fffef9b164"},"type":"zigbee_publish_error"}'

closed 2024-04-25 by buiducnhat

buiducnhat, 2025-10-21:
> [@buiducnhat](https://github.com/buiducnhat) - you closed this as completed, did you find a solution? I am having the same problem.

Actually I left this IoT project for over a year, so now I cannot remember anything xD. If you want, I can reopen this issue

hrefhref, 2025-12-14:
I am also having issues with Tuya ZS06. No command works. Unlike your log, when I press "on" for the learn_ir_code on the UI, nothing appears in the logs. When directly publishing to mqtt, nothing appears in the log either.

When using the UI to send an IR code, nothing appears in the logs.