Koenkk/zigbee2mqtt #22231: Group scene id 0 invalid

markchalloner, 2024-04-16:
### What happened?

When trying to add a scene with id 0 (e.g. for the [Philips Hue Dimmer V2](https://www.zigbee2mqtt.io/devices/929002398602.html#philips-929002398602)) the following error is displayed and the scene not added:

```
Publish 'set' 'scene_store' to 'Test' failed: 'Error: ID must be at least 1'
```

Workaround was to downgrade to 1.36.0.

Zigbee2MQTT version
1.36.1 commit: unknown
Coordinator type
zStack3x0
Coordinator revision
20230507
Frontend version
0.6.161
zigbee_herdsman_converters_version
19.11.2
zigbee_herdsman_version
0.40.3

### What did you expect to happen?

No error message displayed and the scene dded to the group.

### How to reproduce it (minimal and precise)

In any group create a scene id 0

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20230507

### Adapter

SLZB-06

### Setup

Add-on to Home Assistant OS in ProxMox

malaakso, 2024-04-17:
Why would you add scene to a dimmer? They should be added to the bulbs.

markchalloner, 2024-04-17:
The scene is added to the group for use by a scene capable switch such as the [Philips Hue Dimmer V2](https://www.zigbee2mqtt.io/devices/929002398602.html#philips-929002398602)

markchalloner, 2024-04-17:
Just tested and this is also an issue adding a scene directly to a bulb too.

Koenkk, 2024-04-17:
Both scene and group ID `0` are not allowed, please pick another number.

markchalloner, 2024-04-17:
@Koenkk thanks. 

To be clear this bug report is only regarding scene ids _not_ group ids, I've updated the title and added a screenshot below:

[image]

Is there a reason for disallowing _scene_ id `0`?

As far as I can tell from [the Zigbee cluster spec](https://www.zigbeealliance.org/wp-content/uploads/2021/10/07-5123-07-ZigbeeClusterLibrary_Revision_7-1.pdf) under section `3.7.2.3 Scene Table` a scene id of `0` seems to be valid:

```
Scene ID | uint8 0x00 to 0xff (see 3.7.2.3.2) | The identifier, unique within this group, which is used to identify this scene.
```

Disallowing scene id `0` means the [Philips Hue Dimmer V2](https://www.zigbee2mqtt.io/devices/929002398602.html#philips-929002398602) Hue button which send alternating recalls with scene id `0` and `1` can't be fully bound:

```
2024-04-17 21:13:12Received Zigbee message from 'Landing Light Dimmer', type 'commandRecall', cluster 'genScenes', data '{"groupid":53372,"sceneid":0}' from endpoint 1 with groupID 53372
2024-04-17 21:13:13Received Zigbee message from 'Landing Light Dimmer', type 'commandRecall', cluster 'genScenes', data '{"groupid":53372,"sceneid":1}' from endpoint 1 with groupID 53372
```

Cheers

Koenkk, 2024-04-18:
@Nerivec from what I remember this was for the Ember adapter, but reading the spec `0x00` should be valid.

Nerivec, 2024-04-18:
I found this while hunting for a bug related to Ember, but it has no relation. It's ZCL spec.
ID zero for groups and scenes are of course valid, but reserved for the global scene.

> 3.7.2.4.1 Generic Usage Notes
Scene identifier 0x00, along with group identifier 0x0000, is reserved for the global scene used by the OnOff cluster.

[ZCL spec R8](https://zigbeealliance.org/wp-content/uploads/2021/10/07-5123-08-Zigbee-Cluster-Library.pdf)
A few more references can be found throughout the spec by searching for `global scene`.

Now, it's not very detailed as to how it should be reserved. Could be a global requirement, or they could have meant just for scene ID zero to not be used in conjunction with group ID zero (as in, scene ID zero can be used with any group ID that is NOT zero)... Left to interpretation...

_Will have to be careful as to what a change here could potentially break. Ikea seems to be using this scheme and I suspect was the cause for some weird issues people reported with the brand in the past. i.e. Any change should not only be tested against that Philips device, but also against several Ikea devices._

jaydeethree, 2024-05-16:
It looks like https://github.com/Koenkk/zigbee-herdsman-converters/pull/7185 is the reason this error gets thrown, and that was first included in Z2M version 1.36.1. The PR doesn't explain *why* the change was made though - was there a specific problem this was supposed to fix?

markchalloner, 2024-05-17:
@Nerivec

> Will have to be careful as to what a change here could potentially break. Ikea seems to be using this scheme and I suspect was the cause for some weird issues people reported with the brand in the [...]

Do you have any links to the issues you suspect might be caused by this so I can do some reading up?

Happy to help with testing, I have a few different IKEA bulbs and remotes too.

Thanks

markchalloner, 2024-06-08:
Thank you @Nerivec I tested it on 1.38.0 and setting scene 0 for a grous works fine. Appreciate the fix!

closed 2024-06-08 by markchalloner

kklepacz, 2024-06-13:
@markchalloner Have you managed to set scene 0 for single bulb/lamp or group? 
Trying to make Hue button working on my Hue Dimmer v2 which is bound to Philips Enrave lamp. 
 I'm getting:
 `Publish 'set' 'scene_store' to 'Living Room Left Ceiling Light' failed: 'Error: Scene ID 0 cannot be used with group ID 0 (reserved).'`
 which is not a surprise given that the code states: 
 `const groupid = isGroup ? entity.groupID : (value.group_id != undefined ? value.group_id : 0);`
 meaning for a single bulb/lamp groupId will be always 0 and therefore:
 `if (groupid === 0 && sceneid === 0) {`
 will be always true when setting Scene 0. 
 @Nerivec is it expected outcome?

Nerivec, 2024-06-13:
Indeed. Zigbee specification reserves the use of scene ID 0 with group ID 0 (it's used for the global scene mechanism).
See https://github.com/Koenkk/zigbee2mqtt/issues/22231#issuecomment-2065316124

@Koenkk Probably the fallback shouldn't be zero for group IDs to avoid this kind of trouble? Not sure what ramifications changing this might have though.

Koenkk, 2024-06-13:
Can you try to pass the group ID? add `group_id: 1` to the payload, although I'm not sure exactly what the behaviour will be:

[image]