Koenkk/zigbee2mqtt #22233: Current latest-dev crashes on startup

Gnarfoz, 2024-04-16:
### What happened?

After updating to 1.36.1-dev (commit #172aa74), Zigbee2MQTT crashes on startup.
Downgrading to docker.io/koenkk/zigbee2mqtt:latest-dev@sha256:f89ab688981b6e1f80d59a8218aa71fae5287f616e3751bb583bfb78ea4db3be fixes it.

### What did you expect to happen?

It to not crash on startup. ;-)

### How to reproduce it (minimal and precise)

Use latest-dev from from 2024-04-16 (1.36.1-dev (commit #172aa74), with ConBee II coordinator, I guess?

### Zigbee2MQTT version

1.36.1-dev (commit #172aa74)

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Setup

Docker container on aarch64 (Armbian on Odroid N2+)

### Debug log

```
Using '/app/data' as data directory
[2024-04-16 20:30:52] info: z2m: Logging to console, file (filename: log.log)
[2024-04-16 20:30:54] debug: z2m: Loaded state from file /app/data/state.json
[2024-04-16 20:30:54] info: z2m: Starting Zigbee2MQTT version 1.36.1-dev (commit #172aa74)
[2024-04-16 20:30:54] info: z2m: Starting zigbee-herdsman (0.42.2)
[2024-04-16 20:30:54] debug: z2m: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[xxx],"networkKey":"HIDDEN","panID":xxx},"serialPort":{"adapter":"deconz","path":"/dev/ttyACM0"}}'
[2024-04-16 20:30:55] debug: zh:adapter: Path '/dev/ttyACM0' is valid for 'DeconzAdapter'
[2024-04-16 20:30:55] debug: zh:deconz:driver: Set delay to 0
[2024-04-16 20:30:55] debug: zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[xxx],"panID":xxx,"extendedPanID":[xxx],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0","adapter":"deconz"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
[2024-04-16 20:30:55] debug: zh:deconz:driver: Opening with /dev/ttyACM0
[2024-04-16 20:30:55] debug: zh:deconz:driver: Serialport opened
[2024-04-16 20:30:55] debug: zh:controller: Started with result 'resumed'
[2024-04-16 20:30:55] debug: zh:controller: Injected database: true, adapter: true
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x00212effff086f66): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x04cd15fffe789738): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x5c0272fffe7e4c71): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0xec1bbdfffeb4ddb6): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x2c1165fffe6dc191): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0xcc86ecfffef3a7e8): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0xec1bbdfffea232d7): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x84b4dbfffecf288c): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0xa4c138fc3d73e19b): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0xa4c138c1187031e7): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x94deb8fffe8765b8): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x84b4dbfffecf2892): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x0c4314fffe67f3d7): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x2c1165fffe96b58e): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0xdc8e95fffe1087ed): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x84fd27fffe9c5f26): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x84b4dbfffebc8bbb): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x00158d000a98718e): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x00158d000a986fc7): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0xa4c138c09c49f19b): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0xa4c138a32b1162bb): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0xa4c1388a5be5f9b6): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x00158d000a986e0f): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x00158d000a98a827): default expiration timeout set to 0
[2024-04-16 20:30:55] debug: zh:controller:device: Request Queue (0x00158d000a98ad6b): default expiration timeout set to 0
[... 32 lines cut ...]
[2024-04-16 20:30:55] info: z2m: UP Aktor 2 (0xdc8e95fffe1087ed): ZG9101SAC-HP-Switch - Sunricher Zigbee AC in wall switch (Router)
[2024-04-16 20:30:55] info: z2m: Steckerleiste (0x84fd27fffe9c5f26): HG06338 - Lidl Silvercrest 3 gang switch, with 4 USB (EU, FR, CZ, BS) (Router)
[2024-04-16 20:30:55] info: z2m: Keller Licht (0x84b4dbfffebc8bbb): 399629_2110 - Lidl Livarno Lux Ceiling Panel RGB+CCT (Router)
[2024-04-16 20:30:55] info: z2m: Briefkasten (0x00158d000a98718e): MCCGQ11LM - Aqara Door and window sensor (EndDevice)
[2024-04-16 20:30:55] info: z2m: Fenster Bad (0x00158d000a986fc7): MCCGQ11LM - Aqara Door and window sensor (EndDevice)
[2024-04-16 20:30:55] info: z2m: Waschmaschine (0xa4c138c09c49f19b): HG08673 - Lidl Silvercrest smart plug with power monitoring (EU, FR) (Router)
[2024-04-16 20:30:55] info: z2m: Heizung (0xa4c138a32b1162bb): HG08673 - Lidl Silvercrest smart plug with power monitoring (EU, FR) (Router)
[2024-04-16 20:30:55] info: z2m: Kühlschrank (0xa4c1388a5be5f9b6): HG08673 - Lidl Silvercrest smart plug with power monitoring (EU, FR) (Router)
[2024-04-16 20:30:55] info: z2m: Terrassentür (0x00158d000a986e0f): MCCGQ11LM - Aqara Door and window sensor (EndDevice)
[2024-04-16 20:30:55] info: z2m: Fenster Gast (0x00158d000a98a827): MCCGQ11LM - Aqara Door and window sensor (EndDevice)
[2024-04-16 20:30:55] info: z2m: Fenster Jannik (0x00158d000a98ad6b): MCCGQ11LM - Aqara Door and window sensor (EndDevice)
[2024-04-16 20:30:55] info: z2m: Zigbee: disabling joining new devices.
[2024-04-16 20:30:55] debug: zh:controller: Disable joining
[2024-04-16 20:30:55] debug: zh:deconz: zclFrame to all - zclFrame.payload:
[2024-04-16 20:30:55] debug: zh:deconz: [object Object]
[2024-04-16 20:30:55] debug: zh:deconz: sendZclFrameToAll - message send
[2024-04-16 20:30:55] debug: zh:deconz:driver: DATA_REQUEST - destAddr: 0xfffd EP:242 SeqNr. 8 request id: 1
[2024-04-16 20:30:55] debug: zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 1 status: 6
[2024-04-16 20:30:55] debug: zh:deconz:driver: networkstate: 0 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 0
TypeError: Cannot read properties of null (reading 'asduPayload')
    at DeconzAdapter.checkReceivedDataPayload (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1104:41)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:76:34)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
```

fjumeaux, 2024-04-17:
Idem with conbee3

Ivano62, 2024-04-17:
Idem with raspbee

vezyr, 2024-04-18:
Same here, with Conbee3 running latest firmware (0x26510900).

SCiunczyk, 2024-04-18:
Same problem, Conbee3 running latest firmware as well.

Koenkk, 2024-04-18:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-04-18 by Koenkk

SCiunczyk, 2024-04-21:
Thanks! Works great.