Koenkk/zigbee2mqtt #22235: Zigbee Dongle E with 7.4.1 and EZSP - Green Power Devices are not running since last update

Burki24, 2024-04-17:
### What happened?

When I use zigbee dongle E with 7.4.1.0 and ember, my network (143 Devices) slows down and a device needs up to 5 Seconds to recognize the command. So I switched back tu EZSP and then zigbee2MQTT crashes, if Ih using a greenpower Device (https://www.zigbee2mqtt.io/devices/PTM_215Z.html#enocean-ptm%2520215z)

Inside the log I get the following messages:

```
Apr 17 05:41:59 Z2M-Erdgeschoss node[18007]: Unhandled MQTT topic
Apr 17 05:45:33 Z2M-Erdgeschoss node[18007]: Unhandled MQTT topic
Apr 17 05:45:39 Z2M-Erdgeschoss node[18007]: TypeError: this.buffer.readUInt8 is not a function
Apr 17 05:45:39 Z2M-Erdgeschoss node[18007]:     at BuffaloZcl.readUInt8 (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:46:35)
Apr 17 05:45:39 Z2M-Erdgeschoss node[18007]:     at Function.fromBuffer (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/zcl/zclHeader.ts:62:43)
Apr 17 05:45:39 Z2M-Erdgeschoss node[18007]:     at EZSPAdapter.processMessage (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:115:39)
Apr 17 05:45:39 Z2M-Erdgeschoss node[18007]:     at Driver.emit (node:events:518:28)
Apr 17 05:45:39 Z2M-Erdgeschoss node[18007]:     at Driver.handleGPMessage (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:877:18)
Apr 17 05:45:39 Z2M-Erdgeschoss node[18007]:     at Driver.handleFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:431:18)
Apr 17 05:45:39 Z2M-Erdgeschoss node[18007]:     at Ezsp.emit (node:events:518:28)
Apr 17 05:45:39 Z2M-Erdgeschoss node[18007]:     at Ezsp.onFrameReceived (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:456:18)
Apr 17 05:45:39 Z2M-Erdgeschoss node[18007]:     at SerialDriver.emit (node:events:518:28)
Apr 17 05:45:39 Z2M-Erdgeschoss node[18007]:     at SerialDriver.handleDATA (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:250:14)
Apr 17 05:45:39 Z2M-Erdgeschoss systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Apr 17 05:45:39 Z2M-Erdgeschoss systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Apr 17 05:45:39 Z2M-Erdgeschoss systemd[1]: zigbee2mqtt.service: Consumed 22.231s CPU time.
Apr 17 05:45:49 Z2M-Erdgeschoss systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 2.
Apr 17 05:45:49 Z2M-Erdgeschoss systemd[1]: Stopped zigbee2mqtt.service - zigbee2mqtt.
Apr 17 05:45:49 Z2M-Erdgeschoss systemd[1]: zigbee2mqtt.service: Consumed 22.231s CPU time.
Apr 17 05:45:49 Z2M-Erdgeschoss systemd[1]: Starting zigbee2mqtt.service - zigbee2mqtt...
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]: [2024-04-17 05:45:51] error:         zh:ezsp:uart: <-- CRC error: ffd6db1d1c96a5877e|a587|b287
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]: [2024-04-17 05:45:51] error:         zh:ezsp:uart: Error while parsing to NpiFrame 'Error: <-- CRC error: ffd6db1d1c96a5877e|a587|b287
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at Frame.checkCRC (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/frame.ts:87:19)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at SerialDriver.onParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:169:19)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at Parser.emit (node:events:518:28)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/parser.ts:49:26)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at Parser.Transform._write (node:internal/streams/transform:171:8)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at writeOrBuffer (node:internal/streams/writable:564:12)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at _write (node:internal/streams/writable:493:10)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at Parser.Writable.write (node:internal/streams/writable:502:10)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at SerialPort.ondata (node:internal/streams/readable:1007:22)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at SerialPort.emit (node:events:518:28)'
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]: [2024-04-17 05:45:51] error:         zh:ezsp:uart: <-- CRC error: 6a0bb1a9112a15b65894a524ab5593499c76e767724b9874f0cf048bfcd436a5eba9112a15b65894a524ab5593499c77e767724b9874f0cf048bfcd436a5ebc9de6f8ee1c3ba807e|ba80|ec40
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]: [2024-04-17 05:45:51] error:         zh:ezsp:uart: Error while parsing to NpiFrame 'Error: <-- CRC error: 6a0bb1a9112a15b65894a524ab5593499c76e767724b9874f0cf048bfcd436a5eba9112a15b65894a524ab5593499c77e767724b9874f0cf048bfcd436a5ebc9de6f8ee1c3ba807e|ba80|ec40
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at Frame.checkCRC (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/frame.ts:87:19)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at SerialDriver.onParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:169:19)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at Parser.emit (node:events:518:28)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/parser.ts:49:26)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at Parser.Transform._write (node:internal/streams/transform:171:8)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at writeOrBuffer (node:internal/streams/writable:564:12)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at _write (node:internal/streams/writable:493:10)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at Parser.Writable.write (node:internal/streams/writable:502:10)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at SerialPort.ondata (node:internal/streams/readable:1007:22)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at SerialPort.emit (node:events:518:28)'
Apr 17 05:46:08 Z2M-Erdgeschoss systemd[1]: Started zigbee2mqtt.service - zigbee2mqtt.
Apr 17 05:46:24 Z2M-Erdgeschoss node[18042]: Unhandled MQTT topic

```

### What did you expect to happen?

Before changing to zigbee2MQTT V.1.36.1-dev commit: [172aa746](https://github.com/Koenkk/zigbee2mqtt/commit/172aa746) all was running fine with Dongel E V 7.4.1.0 and zigbee2MQTT including the greenpower devices. I think, there is a problem in the last zigbee-herdsmann update.

### How to reproduce it (minimal and precise)

Update the Dongle to latest FW (7.4.1.0) Update zigbee to latest dev (1.36.1-dev commit: [172aa746](https://github.com/Koenkk/zigbee2mqtt/commit/172aa746))

Switch the adapter to ember (look if there are time delays in the execution of actions and then switch to EZSP in the configuration.yaml. Then use a greenpower Device (as the wallswitch, I use) 

### Zigbee2MQTT version

1.36.1-dev commit: 172aa746

[... 53 lines cut ...]
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at SerialPort.ondata (node:internal/streams/readable:1007:22)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at SerialPort.emit (node:events:518:28)'
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]: [2024-04-17 05:45:51] error:         zh:ezsp:uart: <-- CRC error: 6a0bb1a9112a15b65894a524ab5593499c76e767724b9874f0cf048bfcd436a5eba9112a15b65894a524ab5593499c77e767724b9874f0cf048bfcd436a5ebc9de6f8ee1c3ba807e|ba80|ec40
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]: [2024-04-17 05:45:51] error:         zh:ezsp:uart: Error while parsing to NpiFrame 'Error: <-- CRC error: 6a0bb1a9112a15b65894a524ab5593499c76e767724b9874f0cf048bfcd436a5eba9112a15b65894a524ab5593499c77e767724b9874f0cf048bfcd436a5ebc9de6f8ee1c3ba807e|ba80|ec40
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at Frame.checkCRC (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/frame.ts:87:19)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at SerialDriver.onParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:169:19)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at Parser.emit (node:events:518:28)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/parser.ts:49:26)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at Parser.Transform._write (node:internal/streams/transform:171:8)
Apr 17 05:45:51 Z2M-Erdgeschoss node[18042]:     at writeOrBuffer (node:internal/streams/writable:564:12)
[... cut ...]
Apr 17 05:46:24 Z2M-Erdgeschoss node[18042]: Unhandled MQTT topic
Apr 17 05:46:24 Z2M-Erdgeschoss node[18042]: Unhandled MQTT topic
Apr 17 05:46:32 Z2M-Erdgeschoss node[18042]: Unhandled MQTT topic
Apr 17 05:46:34 Z2M-Erdgeschoss node[18042]: Unhandled MQTT topic
Apr 17 05:46:45 Z2M-Erdgeschoss node[18042]: Unhandled MQTT topic
Apr 17 05:47:46 Z2M-Erdgeschoss node[18042]: [2024-04-17 05:47:46] error:         zh:controller:device: Handling of poll check-in from 0x0cae5ffffeb04205 failed
Apr 17 05:48:42 Z2M-Erdgeschoss node[18042]: Unhandled MQTT topic

```

Koenkk, 2024-04-19:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

Burki24, 2024-04-19:
off course:

Debug with ezsp:

If I use the Greenpower-Device, zigbee2mqtt with ezsp stops with:
```
pi@Z2M-Erdgeschoss:/opt/zigbee2mqtt $ DEBUG=zigbee-herdsman* > ezsp.txt npm start
TypeError: this.buffer.readUInt8 is not a function
    at BuffaloZcl.readUInt8 (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:46:35)
    at Function.fromBuffer (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/zcl/zclHeader.ts:62:43)
    at EZSPAdapter.processMessage (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:115:39)
    at Driver.emit (node:events:518:28)
    at Driver.handleGPMessage (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:877:18)
    at Driver.handleFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:431:18)
    at Ezsp.emit (node:events:518:28)
    at Ezsp.onFrameReceived (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:456:18)
    at SerialDriver.emit (node:events:518:28)
    at SerialDriver.handleDATA (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:250:14)
pi@Z2M-Erdgeschoss:/opt/zigbee2mqtt $
```
log: ezsp.txt

When I run with ember, first start it breaks (ember1.txt). Second run it runs well, but it's getting slower after 1-5 minutes (ember2.txt).

[ember_2.txt](https://github.com/Koenkk/zigbee2mqtt/files/15046044/ember_2.txt)
[ember_1.txt](https://github.com/Koenkk/zigbee2mqtt/files/15046045/ember_1.txt)
[ezsp.txt](https://github.com/Koenkk/zigbee2mqtt/files/15046046/ezsp.txt)

Burki24, 2024-04-19:
Sorry,
forgot the configuration.yaml.
[configuration.zip](https://github.com/Koenkk/zigbee2mqtt/files/15046088/configuration.zip)

Koenkk, 2024-04-20:
@Burki24 if you remove `log_syslog` from the configuration.yaml, is the ember performance issue solved?

Burki24, 2024-04-20:
> @Burki24 if you remove `log_syslog` from the configuration.yaml, is the ember performance issue solved?

I'll give it a try.

Burki24, 2024-04-21:
No difference between with/without `log_syslog`.

I looked into this a little more closely and the delays are rather inconsistent. So sometimes up to 3 seconds and sometimes the commands come straight through. I didn't have that with EZSP. When using EZSP it always worked immediately (refers to a wide variety of actions).

I don't use direct binding for the automations but work with the extension from here: [https://github.com/Anonym-tsk/zigbee2mqtt-extensions](). Could this be a reason?

By the way, I'm wondering why the following appears in the log (when I run `sudo journalctl -u zigbee2mqtt.service -f`) for every action,  but nothing appears in the frontend log:

`Apr 21 06:47:42 Z2M-Ground floor node[28704]: Unhandled MQTT topic
`
It doesn't matter whether I trigger the action directly in the frontend of zigbee2mqtt or through a payload sent from externally. And that also only comes up, when I run on ember.

Thanks for your support.

Koenkk, 2024-04-21:
- EZSP issue is fixed in latest dev now.
- `Apr 21 06:47:42 Z2M-Ground floor node[28704]: Unhandled MQTT topic` -> could you provide the debug log of this?
See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

Burki24, 2024-04-22:
@Koenkk 

First log (ember_new.txt) zigbee2mqtt wouldn't start. Second log (ember_new_2.txt) zigbee2mqtt starts completely.

[ember_new.txt](https://github.com/Koenkk/zigbee2mqtt/files/15057254/ember_new.txt)
[ember_new_2.txt](https://github.com/Koenkk/zigbee2mqtt/files/15057256/ember_new_2.txt)

closed 2024-04-22 by Burki24

Nerivec, 2024-04-22:
Can you confirm 2 things regarding the `ember` slowness:
1. Is it still present without any custom extension?
2. Are there any offline devices in the network? How many? What availability setting?

If you can provide a debug log of that "slowness" happening, that would be great (like trigger a couple of commands that you know will be "slow" then pull the logs).

Burki24, 2024-04-22:
Okay, I deleted two extension, I used. Restarted zigbee2mqtt with:

DEBUG=zigbee-herdsman* > ember_without_extension.txt npm start

The result is add as file.

I have at the moment 145 Devices. 13 are offline (most of them are end-devices. Only 2 are Router). Running without extension helps to improve the speed of reaction, but it runs again slower then ezsp. Hope, that helps.

Availability runs in simple-mode 'availability: true'

Greetz Burkhard
[ember_without_extension.txt](https://github.com/Koenkk/zigbee2mqtt/files/15057909/ember_without_extension.txt)