Koenkk/zigbee2mqtt #22236: SPM02V2 Spamming Data and maybe let the Gateway crash and Z2M

HobboRobin, 2024-04-17:
### What happened?

I got for Power Measurment 20 of those [SPM02V2] and they are sending uncontrolled data in the network. Five of them are working without any problem. But when i add moore the network getting slower and crash every few hours.

### What did you expect to happen?

Maybe some Kind of controll how often the data get send. Same like the [A1Z]

### How to reproduce it (minimal and precise)

Just add the SPM02v2

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20230507

### Adapter

UZG-01

### Setup

Unraid Docker

### Debug log

The Log is not showing any data except Timeout after 6000ms when adapter already crashed

mrmaximas, 2024-04-17:
Try to flash 20240316 CC1352P7_coordinator_20240316.hex from https://github.com/Koenkk/Z-Stack-firmware/discussions/496
It's works!

DanielNagy, 2024-07-07:
FYI, although i don't have zigbee network crashing, My energy metrics was going screwy. Upon further investigations I noted that this SPM02V2 was flooding requests of like 10 times a second, blowing out the homeassistant statistics database.

I ended up configuring the following debounce setting on this device so the mqtt updates are not as frequent.

I reduces mqtt updates to between 8 and 30 seconds. (I only care about energy for graphs. however i added power and ac_frequency as ignore debounces as well because without them, no values update unless the energy consumed updated, so power/voltage/current etc all flatline)

[image]

I still imagine this is flooding the physical zigbee network. Wish there was a way to reduce the zigbee reporting period of this device.

HobboRobin, 2024-08-10:
I managed to get it not crashing anymoore with moore than 5 Router also in the Network.
But it is still making the network from zigbee really slow and laggy. (+- 20 seconds)
@Koenkk do you have any solution for this ?
I tried also flashing the newest Firmware for my Coordinator but i cant stop the flooding.

DanielNagy, 2024-08-11:
> I managed to get it not crashing anymoore with moore than 5 Router also in the Network.
>
> But it is still making the network from zigbee really slow and laggy. (+- 20 seconds)
>
> @Koenkk do you have [...]

They have poor firmware in them. The SPM01V2 has a newer firmware which reduces the reporting rate to every 15 seconds, but you need a Tuya hub to update it. (It doesn't appear to follow zigbee OTA specs)

There is still no new firmware update for the SPM02V2 when I checked the other day to reduce the report rate from flooding the zigbee network. 

My SPM01 and SPM02 devices were randomly sending erroneous data. Is yours doing that? Ie something like 40000volts randomly?

[image]

I've ended initiating a return of all of mine for a refund.

Boertie310, 2024-08-30:
Same problem here with the SPM02V2 (and with TS0225 presence sensors, also tuya). You can see it when it updates at the last seen epoch that has waves of many updates in a very short time.

DanielNagy, 2024-08-30:
@Boertie310 does yours also report random values randomly? like my screenshot above?

Boertie310, 2024-08-30:
No i cant find it yet, only the spamming of data. As i take a simple plug i can set some COV Values and min/max sendtimes. But this device seems not to have that.

Koenkk, 2024-09-02:
AFAIK there is no way to reduce the spamming of this device.

DanielNagy, 2024-09-02:
> AFAIK there is no way to reduce the spamming of this device.

The SPM01 has a firmware update (via Tuya hub) which fixes the reporting to 15 seconds. When I last checked there was no update to the spm02. 

I did wireshark to see the OTA process, but it seemed to be using some proprietary update method, instead of the standard zigbee OTA I've seen before.  
I did manage to get the download the firmware, but do you know anything about the way the Tuya update works when it's updating the vendors mcu, rather than the Tuya mcu? I assume zigbee2mqtt does not support the non standard OTA Tuya used in this instance.

mrmaximas, 2024-09-02:
> I did manage to get the download the firmware, but do you know anything about the way the Tuya update works when it's updating the vendors mcu, rather than the Tuya mcu? I assume zigbee2mqtt does not support the non standard OTA Tuya used in this instance.

Have you tried mitm for the firmware url?

DanielNagy, 2024-09-03:
> Have you tried mitm for retrieve new SPM01 firmware url? 

mine was 3.1.1 prior to this update via tuya hub / tuya smartlife app. it then updated to the following

          "url": "https://images.tuyaeu.com/smart/firmware/upgrade/ay1553162414592ZG6Kk/17211854247e38bd26c81.bin",
          "version": "3.1.3"

> What appVersion do you see for your SPM01 in z2m database.db?

"appVersion":68,

Not sure thats relavant as both my spm02 and spm01 show the same version.. Seems like the version of the Tuya Module, not the MCU version.

DanielNagy, 2024-09-03:
Attached is wireshark of SPM01 joining / doing MCU OTA / leaving tuya hub.

set wireshark filter to `(wpan.dst16 == 0x720f ) || (wpan.src16 == 0x720f)`

@Koenkk, You can probably confirm this isnt a OTA method supported by Zigbee2mqtt. Around packet 1795 is where it starts.

[SPM01V2 join, ota, leave.zip](https://github.com/user-attachments/files/16842467/SPM01V2.join.ota.leave.zip)

DanielNagy, 2024-09-03:
Start of MCU firmware file
[image]

Wireshark pcap packet 1975 confirms the same first line of data
[image]

The immediate 2 or 3 (or 4) bytes before the data seem to be the offset address, as that increases by 0x30 bytes each message.

Anyways. I'm fairly confident this is tuya custom ota to upgrade the vendors mcu, rather than the tuya zigbee module itself. @Koenkk will likely confirm this also.

Means you'll need to source a cheap Tuya zigbee hub to firmware update the spm01 to latest firmware that doesnt flood the zigbee network. Unsure if the SPM02 has the same firmware update yet.

DanielNagy, 2024-09-03:
> After I read the above message about there is a new firmware that increases the reporting time to 15 seconds I went ahead and bought a cheap tuya gateway. I can confirm the new firmware let you set a custom reporting interval from 5s to 1hr.

Is this SPM02? Update wasnt available when I upgraded the SPM01's

Do you have random values reported by any of these? look here for examples i'm seeing. https://github.com/Koenkk/zigbee2mqtt/issues/23291

Zemismart/BituoTechnick are telling me they are unable to reproduce it.

DanielNagy, 2024-09-03:
> @DanielNagy Oh sorry, no they are SPM01V2.
> 
> I have seen spikes in the values on mines but when I investigated it seemed it was because they stopped reporting for a while and when they reported back they reported the accumulated value. I always assumed it was because the zigbee network was so flooded that packages were lost and values never reached z2m. Now that there is no flooding any more I haven’t seen any spike… but I’ll keep an eye and let you know.

Accumilated values on kWh Energy yes.. But, look at my example graphs here.. https://github.com/Koenkk/zigbee2mqtt/issues/23291#issuecomment-2267993147
[image]

~65000 Volts. Its not a accumilation error. Looking at the debug logs, its a parsing error.

But given you dont see this, Can i confirm what zigbee coordinator you are using? Is it EFR or TI/CC based?

DanielNagy, 2024-09-03:
> It’s a Zonoff Zigbee Dongle P (CC2652P).

Hrm. Are you able to confirm what coordinator revision you are using?
Zigbee2mqtt -> Settings (cog icon) -> about tab.
[image]

Really trying to work out why no one else is reporting issues like mine is in that other issue i posted

DanielNagy, 2024-09-03:
> @DanielNagy I forgot I changed my coordinator 3 months ago to a SMLIGHT SLZB-06 (same CC2652P chip) when trying to figure out why my zigbee network was crashing (which at the end it rurned out to be these energy meters flooding the network) and I never switched back to the Zonoff Zigbee Dongle P.
> 
> 
> 
> In case it's helpful here is the revision of this coordinator:
> 
> 
> 
> [image]
> 
> 

Thanks. I'm trying to eliminate potential issues as to why I'm seeing random values daily. I had hoped maybe it was an issue related to the cc2652p chip, but it doesn't appear to be as yours isn't reporting any random values. I had even tried that coord firmware too. I had ordered a efr based coord and hoped maybe that would solve my problems. Back to the drawing board it seems.

DanielNagy, 2024-09-05:
Ok update. In regard to my erratic values as shown by the screenshot, it was fixed by upgrading the esp32/esphome firmware for my tubesZB Ethernet coordinator. It's been rock solid for over 24 hours now with no erratic values.

DanielNagy, 2024-09-05:
> AFAIK there is no way to reduce the spamming of this device.

There is a firmware updating coming soon for the SPM02 which will fix the data reporting to 15 seconds. But it requires a Tuya hub to update. 

The firmware for the SPM01 has already been released to reduce the reporting spam. But again, you need a Tuya hub to firmware upgrade it.

DanielNagy, 2024-09-05:
I have official word from Bituo saying there will be NO OTA fix for the spm02v2 to reduce the flooding reporting period like the recent OTA fix for the spm01v2 due to complications with the Tuya DP point change, and needing some users to press the button to re-pair the device as they experienced two weeks ago with the Tuya wifi version OTA. 

They have also said there will be a SPM02V3 which will have the updated firmware.

closed 2024-10-22 by HobboRobin

vilaemail, 2025-02-19:
> EDIT: As of today, there is no option to change the reporting time interval in z2m as you would in the Tuya app.

[This PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8843) adds support for SPM01V2 to change reporting time interval. [Firmware updating however is not available in Z2M](https://github.com/Koenkk/zigbee2mqtt/issues/22236#issuecomment-2325474143), you would first need to do an OTA update with a Tuya gateway.

S474N, 2025-03-09:
Same problem with SPM02V2.5.

```
info 2025-03-09 18:28:20z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":49.98,"current_a":0,"current_b":0,"current_c":0.008,"elapsed":75,"energy":0,"last_seen":"2025-03-09T17:28:19.032Z","linkquality":69,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor":0,"produced_energy":0,"voltage_a":230.3,"voltage_b":231.6,"voltage_c":235.2}'
info 2025-03-09 18:28:20z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":49.98,"current_a":0,"current_b":0,"current_c":0.008,"elapsed":85,"energy":0,"last_seen":"2025-03-09T17:28:19.117Z","linkquality":72,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor":0,"produced_energy":0,"voltage_a":230.3,"voltage_b":231.6,"voltage_c":235.2}'
info 2025-03-09 18:28:20z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":49.98,"current_a":0,"current_b":0,"current_c":0.008,"energy":0,"last_seen":"2025-03-09T17:28:19.191Z","linkquality":72,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor":0,"produced_energy":0,"voltage_a":230.3,"voltage_b":231.6,"voltage_c":235.2}'
info 2025-03-09 18:28:20z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":49.98,"current_a":0,"current_b":0,"current_c":0.008,"elapsed":150,"energy":0,"last_seen":"2025-03-09T17:28:19.269Z","linkquality":61,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor":0,"produced_energy":0,"voltage_a":230.3,"voltage_b":231.6,"voltage_c":235.2}'
info 2025-03-09 18:28:20z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":49.98,"current_a":0,"current_b":0,"current_c":0.008,"elapsed":77,"energy":0,"last_seen":"2025-03-09T17:28:19.344Z","linkquality":72,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor":0,"produced_energy":0,"voltage_a":230.3,"voltage_b":231.6,"voltage_c":235.2}'
info 2025-03-09 18:28:20z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":49.98,"current_a":0,"current_b":0,"current_c":0.008,"elapsed":71,"energy":0,"last_seen":"2025-03-09T17:28:19.417Z","linkquality":72,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor":0,"produced_energy":0,"voltage_a":230.3,"voltage_b":231.6,"voltage_c":235.2}'
info 2025-03-09 18:28:20z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":49.98,"current_a":0,"current_b":0,"current_c":0.008,"elapsed":85,"energy":0,"last_seen":"2025-03-09T17:28:19.501Z","linkquality":72,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor":0,"produced_energy":0,"voltage_a":230.3,"voltage_b":232.6,"voltage_c":235.2}'
info 2025-03-09 18:28:20z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":49.98,"current_a":0,"current_b":0,"current_c":0.008,"elapsed":77,"energy":0,"last_seen":"2025-03-09T17:28:19.580Z","linkquality":72,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor":0,"produced_energy":0,"voltage_a":230.3,"voltage_b":232.6,"voltage_c":235.2}'
info 2025-03-09 18:28:20z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":49.98,"current_a":0,"current_b":0,"current_c":0.008,"elapsed":72,"energy":0,"last_seen":"2025-03-09T17:28:19.652Z","linkquality":72,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor":0,"produced_energy":0,"voltage_a":230.3,"voltage_b":232.6,"voltage_c":235.2}'
info 2025-03-09 18:28:20z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":49.98,"current_a":0,"current_b":0,"current_c":0.008,"elapsed":72,"energy":0,"last_seen":"2025-03-09T17:28:19.723Z","linkquality":72,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor":0,"produced_energy":0,"voltage_a":230.3,"voltage_b":232.6,"voltage_c":234.3}'
info 2025-03-09 18:28:21z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":49.98,"current_a":0,"current_b":0,"current_c":0.009,"elapsed":84,"energy":0,"last_seen":"2025-03-09T17:28:19.806Z","linkquality":72,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor":0,"produced_energy":0,"voltage_a":230.3,"voltage_b":232.6,"voltage_c":234.3}'
info 2025-03-09 18:28:21z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":49.98,"current_a":0,"current_b":0,"current_c":0.009,"elapsed":68,"energy":0,"last_seen":"2025-03-09T17:28:19.876Z","linkquality":72,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor":0,"produced_energy":0,"voltage_a":230.3,"voltage_b":232.6,"voltage_c":234.3}'
info 2025-03-09 18:28:50z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":49.98,"current_a":0,"current_b":0,"current_c":0.009,"elapsed":29162,"energy":0,"last_seen":"2025-03-09T17:28:49.034Z","linkquality":69,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor":0,"produced_energy":0,"voltage_a":230.3,"voltage_b":232.6,"voltage_c":234.3}'
info 2025-03-09 18:28:50z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":49.98,"current_a":0,"current_b":0,"current_c":0.009,"elapsed":121,"energy":0,"last_seen":"2025-03-09T17:28:49.157Z","linkquality":69,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor":0,"produced_energy":0,"voltage_a":230.3,"voltage_b":232.6,"voltage_c":234.3}'
info 2025-03-09 18:28:50z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":49.98,"current_a":0,"current_b":0,"current_c":0.009,"energy":0,"last_seen":"2025-03-09T17:28:49.228Z","linkquality":69,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor": [... cut ...] or":0,"produced_energy":0,"voltage_a":230.1,"voltage_b":232.6,"voltage_c":234.5}'
info 2025-03-09 18:28:55z2m:mqtt: MQTT publish: topic 'Z2M/SPOTR_indukce', payload '{"ac_frequency":50,"current_a":0,"current_b":0,"current_c":0.009,"energy":0,"last_seen":"2025-03-09T17:28:54.154Z","linkquality":69,"power":0,"power_a":0,"power_b":0,"power_c":0,"power_factor":0,"produced_energy":0,"voltage_a":230.1,"voltage_b":232.6,"voltage_c":234.5}'
```

Does anyone have an address for a firmware update, please?

alexanderek, 2025-03-14:
Same problem with SPM01V2.5, firmware V3.2.1

mrmaximas, 2025-04-01:
@slapstick92 what version of z2m and coordinator do you have?

slapstick92, 2025-04-01:
@mrmaximas I am using a ZigBee stick based on the Ebyte E72-2G4M20S1E module (CC2652P chip)

[image]

DanielNagy, 2026-01-15:
> I have official word from Bituo saying there will be NO OTA fix for the spm02v2 to reduce the flooding reporting period like the recent OTA fix for the spm01v2 due to complications with the Tuya DP [...]

FYI. I noticed this Firmware Update appeared via the SmartLife / Tuya hub for this device.

[image]

So i paired the spm02 to the tuya hub, and updated it. The unit certainly has stopped spamming the zigbee network now, as the report period is now 300 seconds (not configurable like the single phase SPM01's). 

However, with a large power change, the unit reports an update immediately. But on small power changes it does not. 
Unclear on what the threshold is to trigger this, but a 2kw power change triggers a reports, a 400watt power change does not trigger a report.

matlab22, 2026-01-30:
I have two of them. One was offering an update and the other did not. So I opened them and the SPM0202V3 (power board) / SPM0202V2 (base board) did not offer me an update. But the SPM0202V4 let me update from V1.1.1 to V1.1.12.

[image]

[image]

I am wondering if it would be possible switch the ZB module and update the other as well?

[image]

DanielNagy, 2026-02-08:
I've got quite a number of SPM02 V2 Hardware, and only total. 
I also have Bituo ESP32 Wifi version based SPM02's, without opening the hardware, i can only assume it is V3 hardware. it DOES have per phase energy values + reverse