Koenkk/zigbee2mqtt #22240: Problem connecting ZG-204ZM _TZE200_2aaelwxk

Koterak, 2024-04-17:
### What happened?

Hello!
I bought presence sensor from this seller: [CLICK](https://pl.aliexpress.com/item/1005006593707347.html?spm=a2g0o.order_list.order_list_main.5.507a1c24MZeVoM&gatewayAdapt=glo2pol)

Unfortunatly I see some strange problems.. 
When trying to pair this device it keeps blinking all the time. Batteries I'm sure are good. I tried also custom converter which was mentioned by seller. But it's not working with newest Zigbee - got error "z2m: Failed to load external converter file 'cntop_zigbee_sensor.js' (Cannot find module 'zigbee-herdsman-converters/lib/extend'".

I tried also to sniff network and I can see timeouts in logs:

[image]

Is there anything I can try? 

### What did you expect to happen?

To work

### How to reproduce it (minimal and precise)

Use mentioned device

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20221226

### Adapter

ZigStar Stick v4

### Setup

Add-on on Home Assistant

Koterak, 2024-04-17:
In addition, custom converter from seller below:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const tzDatapoints = {
    ...tuya.tz.datapoints,
    key: [...tuya.tz.datapoints.key,  'alarm', 'alarm_time', 'alarm_volume', 'type', 'volume', 'ringtone', 'duration', 'medium_motion_detection_distance',
            'large_motion_detection_distance', 'large_motion_detection_sensitivity', 'small_motion_detection_distance',
            'small_motion_detection_sensitivity', 'static_detection_distance', 'static_detection_sensitivity', 'keep_time', 'indicator',
            'motion_sensitivity', 'detection_distance_max', 'detection_distance_min', 'presence_sensitivity', 'sensitivity', 'illuminance_interval',
            'medium_motion_detection_sensitivity', 'small_detection_distance', 'small_detection_sensitivity',]
}

const definition = [

    {
        fingerprint: tuya.fingerprint('TS0225', ['_TZE200_hl0ss9oa']),
        model: 'ZG-205ZL',
        vendor: 'TuYa',
        description: '24Ghz human presence sensor',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tzDatapoints],
        exposes: [
            e.presence(),
            e.enum('motion_state', ea.STATE, ['none', 'large', 'small', 'static']).withDescription('Motion state'),
            e.illuminance_lux(),
            e.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(1).withUnit('s')
                .withDescription('Presence keep time'),
            e.numeric('large_motion_detection_distance', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.01).withUnit('m')
                .withDescription('Large motion detection distance'),
            e.numeric('large_motion_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
                .withDescription('Large motion detection sensitivity'),
            e.numeric('small_motion_detection_distance', ea.STATE_SET).withValueMin(0).withValueMax(6).withValueStep(0.01).withUnit('m')
                .withDescription('Small motion detection distance'),
            e.numeric('small_motion_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
                .withDescription('Small motion detection sensitivity'),
            e.numeric('static_detection_distance', ea.STATE_SET).withValueMin(0).withValueMax(6).withValueStep(0.01).withUnit('m')
                .withDescription('Static detection distance'),
            e.numeric('static_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
                .withDescription('Static detection sensitivity'),
            e.enum('mode', ea.STATE_SET, ['off', 'arm', 'alarm']).withDescription('Alarm mode'),
            e.enum('alarm_volume', ea.STATE_SET, ['mute', 'low', 'medium', 'high']).withDescription('Alarm volume'),
            e.numeric('alarm_time', ea.STATE_SET).withValueMin(1).withValueMax(60).withValueStep(1).withUnit('m').withDescription('Alarm time'),
            e.binary('indicator', ea.STATE_SET, 'ON', 'OFF').withDescription('LED indicator mode'),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'presence', tuya.valueConverter.trueFalse1],
                [20, 'illuminance_lux', tuya.valueConverter.raw],
                [11, 'motion_state', tuya.valueConverterBasic.lookup({
                    'none': tuya.enum(0), 'large': tuya.enum(1), 'small': tuya.enum(2), 'static': tuya.enum(3),
                })],
                [12, 'fading_time', tuya.valueConverter.raw],
                [13, 'large_motion_detection_distance', tuya.valueConverter.divideBy100],
                [15, 'large_motion_detection_sensitivity', tuya.valueConverter.raw],
                [14, 'small_motion_detection_distance', tuya.valueConverter.divideBy100],
                [16, 'small_motion_detection_sensitivity', tuya.valueConverter.raw],
                [103, 'static_detection_distance', tuya.valueConverter.divideBy100],
[... 180 lines cut ...]
                [2, 'battery', tuya.valueConverter.raw],
                [102, 'illuminance_interval', tuya.valueConverter.raw],
            ],
        },
    },
    {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_8isdky6j']),
        model: 'ZG-225Z',
        vendor: 'TuYa',
        description: 'Gas sensor',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
     
        exposes: [e.gas(), tuya.exposes.gasValue().withUnit('ppm')],
        meta: {
            tuyaDatapoints: [
                [1, 'gas', tuya.valueConverter.trueFalse0],
                [2, 'gas_value', tuya.valueConverter.raw],
            ],
        },
    },
];

module.exports = definition;
```

fatshotty, 2024-05-09:
i have same problem, too :/
I cannot pair device

Koenkk, 2024-05-10:
Try removing `const extend = require('zigbee-herdsman-converters/lib/extend');` from the converter and see if it works.

fatshotty, 2024-05-11:
thanks @Koenkk 
already tried; device has been correctly recognized but it has 2 sensors only. So, it doesn't work well :/

Micka41, 2024-06-02:
I have the same problem!

closed 2024-12-29 by github-actions[bot]

induna-crewneck, 2025-04-10:
Did you ever find a solution? Naively I ordered two before looking for testimonials with zigbee2mqtt and I can get neither to pair. They keep blinking but nothing shows up in zigbee2mqtt