Koenkk/zigbee2mqtt #22245: SIN-4-FP-21 - Different firmware updates for apparently same devices

nerthazrim, 2024-04-18:
### What happened?

I own 2 SIN-4-FP-21, ordered at the same time from the same vendor.
Both received a firmware update today, but they are not ending up with the same version after update.

### What did you expect to happen?

Both SIN-4-FP-21 to have the same version after having completed the update.

### How to reproduce it (minimal and precise)

I don't know... Except by buying a new one and updating it?

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

 EZSP v8

### Setup

Add-on on Home Assistant supervised in a Proxmox Debian 12 VM

n0n0-44, 2024-04-25:
Hi,

Same problem for me, with 3 SIN-4-FP-21 ...
[image]

nerthazrim, 2024-05-10:
Fun fact, I got an update available for the one with the "oldest" version, today.
But they're still not at the same one 😁 
[image]

n0n0-44, 2024-05-12:
The same here ...
[image]

Th-Ar, 2024-05-22:
Same here, 3 different versions :
[image]

nerthazrim, 2024-08-02:
Hi,

New update available on both my SIN-4-FP-21 but still a difference in versions:
3.3.0-1.4.4 for one
3.3.0-1.4.5 for the other

Najihel, 2024-09-04:
Hello,

Same issue here, sometimes "new version" is available but the upgrade failed eachtime

```
info 2024-09-04 12:50:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/buanderie-chauff', payload '{"energy":2.1,"linkquality":255,"pilot_wire_mode":"frost_protection","power":0,"state":"OFF","update":{"installed_version":768,"latest_version":771,"state":"available"},"update_available":true}'
info 2024-09-04 12:50:55z2m: Updating 'buanderie-chauff' to latest firmware
info 2024-09-04 12:50:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating 'buanderie-chauff' to latest firmware","meta":{"device":"buanderie-chauff","status":"update_in_progress"},"type":"ota_update"}'
info 2024-09-04 12:50:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/buanderie-chauff', payload '{"energy":2.1,"linkquality":255,"pilot_wire_mode":"frost_protection","power":0,"state":"OFF","update":{"installed_version":768,"latest_version":771,"state":"available"},"update_available":true}'
info 2024-09-04 12:50:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'buanderie-chauff' failed (No new image available)","meta":{"device":"buanderie-chauff","status":"update_failed"},"type":"ota_update"}'
info 2024-09-04 12:50:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"buanderie-chauff"},"error":"Update of 'buanderie-chauff' failed (No new image available)","status":"error","transaction":"va46c-3"}'
error 2024-09-04 12:50:58z2m: Update of 'buanderie-chauff' failed (No new image available)
```

[image]

Th-Ar, 2024-09-04:
Hello,
For the same devices as before, I now have not 3 different versions, but 5.
[image]

Najihel, 2024-11-15:
> Hello,
> 
> Same issue here, sometimes "new version" is available but the upgrade failed eachtime
> 
> ```
> info 2024-09-04 12:50:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/buanderie-chauff', payload '{"energy":2.1,"linkquality":255,"pilot_wire_mode":"frost_protection","power":0,"state":"OFF","update":{"installed_version":768,"latest_version":771,"state":"available"},"update_available":true}'
> info 2024-09-04 12:50:55z2m: Updating 'buanderie-chauff' to latest firmware
> info 2024-09-04 12:50:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating 'buanderie-chauff' to latest firmware","meta":{"device":"buanderie-chauff","status":"update_in_progress"},"type":"ota_update"}'
> info 2024-09-04 12:50:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/buanderie-chauff', payload '{"energy":2.1,"linkquality":255,"pilot_wire_mode":"frost_protection","power":0,"state":"OFF","update":{"installed_version":768,"latest_version":771,"state":"available"},"update_available":true}'
> info 2024-09-04 12:50:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'buanderie-chauff' failed (No new image available)","meta":{"device":"buanderie-chauff","status":"update_failed"},"type":"ota_update"}'
> info 2024-09-04 12:50:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"buanderie-chauff"},"error":"Update of 'buanderie-chauff' failed (No new image available)","status":"error","transaction":"va46c-3"}'
> error 2024-09-04 12:50:58z2m: Update of 'buanderie-chauff' failed (No new image available)
> ```
> 
> <img alt="image" width="1320" src="https://private-user-images.githubusercontent.com/1873677/364344231-5b607ef1-95a2-4b3f-8ee1-04743ddde5c9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE2Nzc1NzUsIm5iZiI6MTczMTY3NzI3NSwicGF0aCI6Ii8xODczNjc3LzM2NDM0NDIzMS01YjYwN2VmMS05NWEyLTRiM2YtOGVlMS0wNDc0M2RkZGU1YzkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTVUMTMyNzU1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NmU3YjkxZTAxNDE4MmNhNDI2OWUyYTQ5ZmUwM2M5YWI0YWQyNzNiNWY5NjVlMDgyYzU0MDMzOTYyYzUyMmE1MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.H7EM-wfi7z-iwkNhAz77gjJwSQeBvBIjTIzRsecy_JI">

Hello everyone,

I found a workaround :
* Download the Nodon Connect app : https://nodon.fr/nodon/application-nodon-connect/
* Create an account
* Associate all your modules (physical access is mandatory)
* Upgrade manually the firmware

Before

[image]

After

[image]

nerthazrim, 2024-11-19:
Well, even after updating the modules with the Nodon Connect app, I still get different versions displayed in Z2M.
I guess I probably have 2 slightly different models without me knowing it so it's not a Z2M issue, I'm closing this :)

Thanks!

closed 2024-11-19 by nerthazrim

Najihel, 2024-11-19:
> Well, even after updating the modules with the Nodon Connect app, I still get different versions displayed in Z2M.
> I guess I probably have 2 slightly different models without me knowing it so it's not a Z2M issue, I'm closing this :)
> 
> Thanks!

You have to re-interrogate the device (the "i" button) to show the new firmware version

nerthazrim, 2024-11-19:
Oh, indeed! Thanks!
Well, let's keep it closed anyway, problem solved as far as I'm concerned ;-)