Koenkk/zigbee2mqtt #22246: CC2652 issue reading payload: zigbee-herdsman keeps crashing

KKraljic, 2024-04-18:
### What happened?

Several things happened:

- Updated Ubuntu OS (and hence received a new version of the Docker engine)
- Pulled the latest zigbee2mqtt container
After restarting the container, I'm receiving the following error messages.
Extract from the logs:

```
$ cat log.log
[2024-04-18 14:23:59] info:     z2m: Logging to console, file (filename: log.log)
[2024-04-18 14:24:00] info:     z2m: Starting Zigbee2MQTT version 1.36.1-dev (commit #1544317)
[2024-04-18 14:24:00] info:     z2m: Starting zigbee-herdsman (0.42.2)
[2024-04-18 14:24:00] info:     zh:zstack:znp: Opening TCP socket with <FQDN-SCRAPPED>:6645
[2024-04-18 14:24:01] info:     zh:zstack:znp: Socket connected
[2024-04-18 14:24:01] info:     zh:zstack:znp: Socket ready
[2024-04-18 14:24:01] info:     zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-04-18 14:24:02] info:     zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-04-18 14:24:02] error:    z2m: Error while starting zigbee-herdsman
[2024-04-18 14:24:02] error:    z2m: Failed to start zigbee
[2024-04-18 14:24:02] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-04-18 14:24:02] error:    z2m: Exiting...
[2024-04-18 14:24:02] error:    z2m: TypeError: Cannot read properties of undefined (reading 'payload')
    at AdapterNvMemory.writeItem (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/adapter-nv-memory.ts:111:31)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at ZnpAdapterManager.clearAdapter (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:468:9)
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:325:9)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:86:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:108:27)
    at start (/app/index.js:107:5)
```

Out of the container I am getting the following error message:

```
Using '/app/data' as data directory
[2024-04-18 14:23:14] info:     z2m: Logging to console, file (filename: log.log)
[2024-04-18 14:23:15] info:     z2m: Starting Zigbee2MQTT version 1.36.1-dev (commit #1544317)
[2024-04-18 14:23:15] info:     z2m: Starting zigbee-herdsman (0.42.2)
[2024-04-18 14:23:16] info:     zh:zstack:znp: Opening TCP socket with <FQDN-SCRAPPED>:6645
[2024-04-18 14:23:16] info:     zh:zstack:znp: Socket connected
[2024-04-18 14:23:16] info:     zh:zstack:znp: Socket ready
[2024-04-18 14:23:16] info:     zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-04-18 14:23:17] info:     zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-04-18 14:23:17] error:    z2m: Error while starting zigbee-herdsman
[2024-04-18 14:23:17] error:    z2m: Failed to start zigbee
[2024-04-18 14:23:17] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-04-18 14:23:17] error:    z2m: Exiting...
[2024-04-18 14:23:17] error:    z2m: TypeError: Cannot read properties of undefined (reading 'payload')
    at AdapterNvMemory.writeItem (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/adapter-nv-memory.ts:111:31)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at ZnpAdapterManager.clearAdapter (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:468:9)
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:325:9)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:86:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:108:27)
    at start (/app/index.js:107:5)

```

On my SLZB-06, I am getting the following Debug information:

```
[724405] -> ef
[725411] -> fe 00 21 01 20
[725415] <- fe 02 61 01 59 06 3d
[725422] -> fe 00 21 02 23
[725426] <- fe 0a 61 02 02 01 02 07 01 6b b1 34 01 00 81
[725433] -> fe 00 21 04 25
[725437] <- fe 08 61 04 09 f2 d0 26 00 4b 12 00 39
[725445] -> fe 02 21 13 82 00 b2
[725454] <- fe 02 61 13 18 00 68
[725468] -> fe 04 21 1c 82 00 00 00 bb
[725484] <- fe 1a 61 1c 00 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7f
[725492] -> fe 02 21 13 60 00 50
[725503] <- fe 02 61 13 00 00 70
[725509] -> fe 02 21 13 21 00 11
[725520] <- fe 02 61 13 00 00 70
[725528] -> fe 02 21 13 62 00 52
[725535] <- fe 02 61 13 10 00 60
[725541] -> fe 04 21 1c 62 00 00 00 5b
[725554] <- fe 12 61 1c 00 10 74 30 60 fb 50 e8 ac e2 13 bb 10 4c b2 c8 5f 0d 8a
[725563] -> fe 02 21 13 3a 00 0a
[725572] <- fe 02 61 13 11 00 61
[725579] -> fe 04 21 1c 3a 00 00 00 03
[725594] <- fe 13 61 1c 00 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7f
[725601] -> fe 02 21 13 3b 00 0b
[725610] <- fe 02 61 13 11 00 61
[725617] -> fe 04 21 1c 3b 00 00 00 02
[725632] <- fe 13 61 1c 00 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7f
[725641] -> fe 02 21 13 21 00 11
[725652] <- fe 02 61 13 00 00 70
[725659] -> fe 02 21 13 03 00 33
[725666] <- fe 02 61 13 01 00 71
[725673] -> fe 07 21 1d 03 00 00 00 01 00 03 3a
[725685] <- fe 01 61 1d 0a 77
[725690] -> fe 07 21 1d 03 00 00 00 01 00 03 3a
[725702] <- fe 01 61 1d 0a 77
[725708] -> fe 07 21 1d 03 00 00 00 01 00 03 3a
[725720] <- fe 01 61 1d 0a 77
[735662] -> fe 00 21 01 20
[735666] <- fe 02 61 01 59 06 3d
[735671] -> fe 01 27 49 00 6f
[735678] <- fe 24 67 49 fe ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0b
[735684] -> fe 05 21 32 01 04 00 00 00 13
[735693] <- fe 04 61 32 14 00 00 00 43
[735697] -> fe 08 21 33 01 04 00 00 00 00 00 14 0b
[735708] <- fe 16 61 33 00 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff 00 00 00 af
[735713] -> fe 00 21 01 20
[735717] <- fe 02 61 01 59 06 3d
[735722] -> fe 02 21 13 21 00 11
[735733] <- fe 02 61 13 00 00 70
[735738] -> fe 02 21 13 00 0f 3f
[735742] <- fe 02 61 13 00 00 70
[735746] -> fe 02 21 13 60 00 50
[735757] <- fe 02 61 13 00 00 70
[735761] -> fe 02 21 13 5f 00 6f
[735772] <- fe 02 61 13 00 00 70
[735777] -> fe 02 21 13 55 00 65
[735784] <- fe 02 61 13 01 00 71
[735793] -> fe 04 21 12 55 00 01 00 63
[735811] <- fe 01 61 12 00 72
[735815] -> fe 02 21 13 03 00 33
[735822] <- fe 02 61 13 01 00 71
[735827] -> fe 07 21 1d 03 00 00 00 01 00 03 3a
[735957] <- fe 01 61 1d 00 7d
[735962] -> fe 01 41 00 01 41
[737961] <- fe 06 41 80 00 02 01 02 07 01 c0
[738050] -> fe 02 21 13 83 00 b3
[738058] <- fe 02 61 13 02 00 72
[738065] -> fe 08 21 1d 83 00 00 00 02 00 ff ff b5
[738077] <- fe 01 61 1d 0a 77
[738082] -> fe 01 41 00 01 41
[739562] <- fe 06 41 80 00 02 01 02 07 01 c0
```
I've already tried to flash my SLZB-06 several times after the error started (also trying downgrades), and have tried to use different versions of zigbee2mqtt (1.35, 1.36, ...) without success.

### What did you expect to happen?

Zigbee2mqtt starts successfully.

### Zigbee2MQTT version

1.36.1-dev

### Adapter firmware version

0.9.9 (Mar 3 2023), CC2652PP1FRGZR

### Adapter

SLZB-06

### Setup

Docker Container

Koenkk, 2024-04-20:
- Does this issue also happen when connecting over USB?
- Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

ALERTua, 2024-06-07:
@KKraljic have you found the solution? I have the same problem

feuerrot, 2024-10-08:
I've got the same problem with a SLZB-06 after I updated its Firmware to a newer version. Neither a downgrade back to 0.9.9 nor flashing the previously working coordinator firmware using `cc2538-bsl.py` does help.
I haven't tried connecting the adapter via USB (as this isn't supported in my current setup).

@Koenkk is this the correct debug log? Your link is 404, but the content is still on archive.org - I used both `log_level: debug` in the config file and `DEBUG=zigbee-herdsman*` in the env: [1728415379_z2m.log](https://github.com/user-attachments/files/17298350/1728415379_z2m.log).

Koenkk, 2024-10-10:
This looks like communication issues with the adapter, @feuerrot for the SLZB-06 I would try connecting it directly via USB (to be sure it's not a network issue)

closed 2025-01-06 by github-actions[bot]

coalfield, 2025-07-23:
same issue here on the SLZB-06P7