Koenkk/zigbee2mqtt #22247: [Feature request]: Please add RGB Effects to Aqara T1M Ceiling light (CL-L02D)

johnsturgeon, 2024-04-18:
### Is your feature request related to a problem? Please describe

Currently the support for the T1M is limited.

### Describe the solution you'd like

I would love to have access to all of the RGB effects that are in the ceiling light, and even (if possible) the ability to create new ones.

### Describe alternatives you've considered

I could use the Aqara hub, but that doesn't get the effects into Home Assistant

### Additional context

Here is a discussion of the issue
https://github.com/Koenkk/zigbee2mqtt/discussions/22215

LZXCorp, 2024-04-26:
Here is the Aqara T1M (CL-L02D) external definition if it is required.

```ts
const {deviceEndpoints, identify, light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['lumi.light.acn032'],
    model: 'lumi.light.acn032',
    vendor: 'Aqara',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), identify(), light({"colorTemp":{"range":[153,370]},"color":true}), light({"colorTemp":{"range":[153,370]}})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

schwagerik, 2024-05-12:
Hi guys, where can find External Definition and Database Entry? Can´t find, somebody me navigate? 

> Here is the Aqara T1M (CL-L02D) external definition if it is required.
> 
> ### External Definition
> ```ts
> const {deviceEndpoints, identify, light} = require('zigbee-herdsman-converters/lib/modernExtend');
> 
> const definition = {
>     zigbeeModel: ['lumi.light.acn032'],
>     model: 'lumi.light.acn032',
>     vendor: 'Aqara',
>     description: 'Automatically generated definition',
>     extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), identify(), light({"colorTemp":{"range":[153,370]},"color":true}), light({"colorTemp":{"range":[153,370]}})],
>     meta: {"multiEndpoint":true},
> };
> 
> module.exports = definition;
> ```
> 
> ### Database Entry
> ```json
> {
> 	"id": 41,
> 	"type": "Router",
> 	"manufName": "Aqara",
> 	"powerSource": "DC Source",
> 	"modelId": "lumi.light.acn032",
> 	"epList": [2, 1],
> 	"endpoints": {
> 		"1": {
> 			"profId": 260,
> 			"epId": 1,
> 			"devId": 258,
> 			"inClusterList": [5, 4, 3, 0, 768, 8, 6, 64704],
> 			"outClusterList": [25, 10],
> 			"clusters": {
> 				"genBasic": {
> 					"attributes": {
> 						"61440": 2,
> 						"modelId": "lumi.light.acn032",
> 						"manufacturerName": "Aqara",
> 						"powerSource": 4,
> 						"zclVersion": 3,
> 						"appVersion": 23,
> 						"stackVersion": 27,
> 						"hwVersion": 1,
> 						"dateCode": "20230525"
> 					}
> 				},
> 				"lightingColorCtrl": {
> 					"attributes": {
> 						"245": 10498122,
> 						"colorCapabilities": 16,
> 						"colorTempPhysicalMin": 153,
> 						"colorTempPhysicalMax": 370,
> 						"colorTemperature": 250
> 					}
> 				},
> 				"manuSpecificLumi": {
> 					"attributes": {
> 						"223": {
> 							"type": "Buffer",
> 							"data": [
> 								22, 33, 0, 0, 23, 33, 0, 0, 12, 33, 0, 0, 24, 33, 241, 12
> 							]
> 						},
> 						"229": {
> 							"type": "Buffer",
> 							"data": [
> 								9, 0, 9, 0, 0, 4, 155, 154, 28, 37, 61, 66, 64, 37, 132, 73, 65,
> 								37, 191, 33, 79, 37, 79, 24, 86, 37, 50, 166, 136, 37, 96, 154,
> 								181, 37, 135, 89, 225, 37, 130
> 							]
> 						},
> 						"247": {
> 							"type": "Buffer",
> 							"data": [117, 32, 254, 115, 35, 239, 135, 30, 69]
> 						}
> 					}
> 				},
> 				"genLevelCtrl": {
> 					"attributes": { "245": 10498121, "currentLevel": 132 }
> 				},
> 				"genOnOff": { "attributes": { "245": 10498246, "onOff": 0 } }
> 			},
> 			"binds": [],
> 			"configuredReportings": [],
> 			"meta": {}
> 		},
> 		"2": {
> 			"profId": 260,
> 			"epId": 2,
> 			"devId": 258,
> 			"inClusterList": [5, 4, 3, 768, 8, 6, 64704],
> 			"outClusterList": [],
> 			"clusters": {
> 				"lightingColorCtrl": {
> 					"attributes": {
> 						"245": 10498273,
> 						"currentX": 17694,
> 						"currentY": 34799,
> 						"colorCapabilities": 24,
> 						"colorTempPhysicalMin": 153,
> 						"colorTempPhysicalMax": 370,
> 						"colorMode": 1,
> 						"colorTemperature": 153
> 					}
> 				},
> 				"genOnOff": { "attributes": { "245": 10498275, "onOff": 0 } },
> 				"genLevelCtrl": {
> 					"attributes": { "245": 10498272, "currentLevel": 254 }
> 				}
> 			},
> 			"binds": [],
> 			"configuredReportings": [],
> 			"meta": {}
> 		}
> 	},
> 	"appVersion": 23,
> 	"stackVersion": 27,
> 	"hwVersion": 1,
> 	"dateCode": "20230525",
> 	"zclVersion": 3,
> }
> ```

steuerlexi, 2024-06-20:
any progress on this feature request?

LZXCorp, 2024-06-21:
> any progress on this feature request?

From what I can see, there is absolutely 0 progress. Aqara also didn't seem to be interested in sharing the technical aspects of the RGB animations too.

So another year of waiting ig.

fillwe, 2024-07-01:
Would really like to see this as well!

adamk125, 2024-07-14:
Would love for this to be improved. Currently can only set solid colours using the Z2M Integration for the Aqara T1M. Cant seem to use the flashing (blinking) function on the device nor any Effects etc for the device.

Hopefully some smarter people can help :)

jyenzen, 2024-07-17:
I would also really like for this to be added :D

saxxonian, 2024-08-29:
I would love this option in HA too!
Any progress or roadmap?
I wish I had the skill to help...

jncanches, 2024-09-02:
Is there some hopes about this feature ?

adamk125, 2024-09-03:
Havent heard anything about this feature yet from anyone on here - not sure how we get this some attention for those smarter than me :) @LZXCorp or @johnsturgeon any updates?

samzorio, 2024-09-05:
probably need someone with an aqara hub to sniff traffic to find out if they are using non ZCL clusters or weird attributes for the effects.

adamk125, 2024-09-05:
> probably need someone with an aqara hub to sniff traffic to find out if they are using non ZCL clusters or weird attributes for the effects.

I have a Camera Hub G2H and the T1M. If you can tell me how to sniff the traffic I can pair the T1M to the Aqara system and do it

samzorio, 2024-09-05:
> > probably need someone with an aqara hub to sniff traffic to find out if they are using non ZCL clusters or weird attributes for the effects.
> 
> I have a Camera Hub G2H and the T1M. If you can tell me how to sniff the traffic I can pair the T1M to the Aqara system and do it

Forgot to mention you will also need a CC2531 USB stick (5-10$ on ebay) You will also need a programmer to flash the needed firmware can also be done with a raspberrypi/arduino
see: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#_1-flashing-the-cc2531-adapter
and: https://www.zigbee2mqtt.io/guide/adapters/flashing/alternative_flashing_methods.html

For instructions on how to sniff traffic
see: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

You will need to extract the network key from your Camera Hub G2H ( I started exploring how to do it, found this) https://github.com/Koenkk/zigbee2mqtt/discussions/13976

I may get an aqara hub to try that in the future but would be kind of a waste as im never going to need it after that. Would be awesome if someone else can do it :)

samzorio, 2024-09-09:
So i ended up buying a E1 aqara hub to sniff traffic to t1m.

Here is what i found so far looks like to control the effects its done via the following endpoint/cluster (One would assume it would be done via endpoint 2 but no)

  Destination Endpoint: 1
    Cluster: Manufacturer Specific (0xfcc0)
    Profile: Home Automation (0x0104)

The build in effects are the following:

`DINNER: 
    Command: Write Attributes (0x02)
    Attribute Field, Octets: 47:41:02:01:01:01:01
        Attribute: 0x050a
        Data Type: Octet String (0x41)
        Octet String: 47:41:02:01:01:01:01
		
SUNSET:	
    Command: Write Attributes (0x02)
    Attribute Field, Octets: 48:42:02:01:01:01:01
        Attribute: 0x050a
        Data Type: Octet String (0x41)
        Octet String: 48:42:02:01:01:01:01
		
AUTUMN:
    Command: Write Attributes (0x02)
    Attribute Field, Octets: 49:43:02:01:01:01:01
        Attribute: 0x050a
        Data Type: Octet String (0x41)
        Octet String: 49:43:02:01:01:01:01
		
GALAXY:
    Command: Write Attributes (0x02)
    Attribute Field, Octets: 4a:44:02:01:01:01:01
        Attribute: 0x050a
        Data Type: Octet String (0x41)
        Octet String: 4a:44:02:01:01:01:01
		
DAYDREAM:
    Command: Write Attributes (0x02)
    Attribute Field, Octets: 4b:45:02:01:01:01:01
        Attribute: 0x050a
        Data Type: Octet String (0x41)
        Octet String: 4b:45:02:01:01:01:01
		
HOLIDAY: 
    Command: Write Attributes (0x02)
    Attribute Field, Octets: 4c:46:02:01:01:01:01
        Attribute: 0x050a
        Data Type: Octet String (0x41)
        Octet String: 4c:46:02:01:01:01:01

PARTY:
    Command: Write Attributes (0x02)
    Attribute Field, Octets: 4d:47:02:01:01:01:01
        Attribute: 0x050a
        Data Type: Octet String (0x41)
        Octet String: 4d:47:02:01:01:01:01
		
METEOR:
    Command: Write Attributes (0x02)
    Attribute Field, Octets: 4e:48:02:01:01:01:01
        Attribute: 0x050a
        Data Type: Octet String (0x41)
        Octet String: 4e:48:02:01:01:01:01
		
ALERT:
    Command: Write Attributes (0x02)
    Attribute Field, Octets: 51:4b:02:01:01:01:01
        Attribute: 0x050a
        Data Type: Octet String (0x41)
        Octet String: 51:4b:02:01:01:01:01`

There is also the option to add custom effects. Have not played with that yet will try to capture it later this week and try and figure it out. 

As for the built-in effects this is all new to me will try to create an extended definition to get this working or at least try to get it working via the dev console of zigbee2mqtt (Devices --> dev console tab) 

If anyone with more experience would like to jump in or help me with this I would appreciate it. 

Thanks all.

adamk125, 2024-09-09:
> So i ended up buying a E1 aqara hub to sniff traffic of t1m.
> 
> Here is what i found so far looks like to control the effects its done via the following endpoint/cluster (One would assume it would be done via endpoint 2 but no)
> 
> ```
> Destination Endpoint: 1
>   Cluster: Manufacturer Specific (0xfcc0)
>   Profile: Home Automation (0x0104)
> ```
> 
> The build in effects are the following:
> 
> ```
> DINNER: 
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 47:41:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 47:41:02:01:01:01:01
> 		
> SUNSET:	
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 48:42:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 48:42:02:01:01:01:01
> 		
> AUTUMN:
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 49:43:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 49:43:02:01:01:01:01
> 		
> GALAXY:
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 4a:44:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 4a:44:02:01:01:01:01
> 		
> DAYDREAM:
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 4b:45:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 4b:45:02:01:01:01:01
> 		
> HOLIDAY: 
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 4c:46:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 4c:46:02:01:01:01:01
> 
> PARTY:
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 4d:47:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 4d:47:02:01:01:01:01
> 		
> METEOR:
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 4e:48:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 4e:48:02:01:01:01:01
> 		
> ALERT:
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 51:4b:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 51:4b:02:01:01:01:01
> ```
> 
> There is also the option to add custom effects. Have not played with that yet will try to capture it later this week and try and figure it out.
> 
> As for getting the built-in effects working in z2m, this is all new to me will try to create an extended definition to get this working or at least try to get it working via the dev console of zigbee2mqtt (Devices --> dev console tab)
> 
> If anyone with more experience would like to jump in or help me with this I would appreciate it.
> 
> Thanks all.

I wish I could help but most of what you have just stated is a bit out out of my depth. 

Looks like a step in the right direction though and I’m sure heaps of us would be very grateful for some extra functionality of the T1M.

SixFive7, 2024-09-09:
Great effort @samzorio!
I understand you didn't get the custom command to work (yet). But did you get the build-in command to work without a hub (using the data you provided)? Or is there also something missing for those?
Also, let me know if there is anything I can do or test for you on my end. I have a light, but no hub.

Caine72, 2024-09-09:
@samzorio 
I tried to use the devconsole to write to the specific cluster, but was stopped more or less at the gate. 

[image]

Tried with 0x00 and 01 as well. No go.

samzorio, 2024-09-09:
Again, 
I'm new to all this trying to figure things out(I have no idea what im doing really). But i think for dev console payload to work attribute/command would need to be added here

https://github.com/Koenkk/zigbee-herdsman/blob/2e948a96c311d9ad27670e8a6ee11c1d34f4598a/src/zspec/zcl/definition/cluster.ts#L4273

I'm running ha OS so i don't have direct access to the z2m docker. 

And unless i'm wrong payload should look something like this

{"attrId":"0x050a", "dataType":"0x41", "attrData":"47410201010101"}
or
{"attrId":1290, "dataType":65, "attrData":47410201010101}

Caine72, 2024-09-09:
> Again, I'm new to all this trying to figure things out(I have no idea what im doing really). But i think for dev console payload to work attribute/command would need to be added here [...]

Ah makes sense that it needs to be added. I am also running z2m on HassOS so i do not have the possibility to change it. 
Or .. does the code need to compile? It's Node.js right?
So .. I think we can jump directly into the docker container and change the file in runtime.

Seem to remember me doing this some time ago. Use the "Advanced SSH & Web Terminal" Add-on. 

Will investigate more tomorrow. :)

Caine72, 2024-09-09:
Was a bit curious and could not drop it. ( well now I am)
So yes I, was right. At least with what i discovered so far. :) 

_* note these instructions are not for setting this up securely, only for a fast test if it works or not. I to set it up in a more secure manner if you the reader intend to keep running the Advanced SSH- add-on._

1. Download and install the add-on: Advanced SSH & Web Terminal
2. Disable ( or change the port of Advanced SSH) the official SSH add-on if you have it. 
3. In it's Main page, i.e. where you start it, turn off Protected mode (required to access docker), and log in via the Web UI-link.
4. Enter the command: `docker ps `
5. Copy the container id ( far left column) for `zigbee2mqtt/zigbee2mqtt`
6. Enter the command (added my container id): `docker exec -it 16bc9cc4e813 /bin/bash`
7. The file is under this path: `/app/node_modules/zigbee-herdsman/dist/zspec/zcl/definition`
8. Should be possible to edit with with. e.g. vi. So command:` vi cluster.ts` ( if you never used the vi-editor before, please do a fast read-up with the basics, else you will have a miserable experience. :))

samzorio, 2024-09-09:
> Was a bit curious and could not drop it. ( well now I am) So yes I, was right. At least with what i discovered so far. :)
>
> _* note these instructions are not for setting this up securely, only for a [...]

i'll try setting up something in a vm to play with it can't really mess around too much with my z2m setup atm as it's used for the whole house. Some ppl might get angry :rofl:

Patresss, 2024-09-20:
@samzorio Hey, did you manage to get anything done?

samzorio, 2024-09-20:
> @samzorio Hey, did you manage to get anything done?

did not have much time to play with it i did try and modify the cluster defintion to add the attribute. 

I'm able to read it now and returns 
zhc:tz: Read result of 'manuSpecificLumi': {"t1meffects":{"type":"Buffer","data":[]}}

But can't figure out how to send the payload i'm ussing it has to be done via command:{}

to be honnest i havent had much time to spend on this will circle back to it eventually

geduxas, 2024-10-16:
Hi @Patresss, @samzorio  nice work you had done! i am almost on same way for now, have sniffed traffic and by searching for data, i stepped in on this thread :D 
could you share your progress right now?

adamk125, 2024-10-16:
> A litle bit of play with static effect, Static effect is directly transmitted while creating it: Attribute: 0x0527 Type: 0x42 Octet sample: 01:01:01:0f:ff:ff:ff:ff:c0:00:00:00:00:00:00:00:00:02:bc
> 
> First bytes alway 01:01:01:0F Next byte is brightnest of whole string: 03-FF (1-100%) Second goes MASK to which element is making changes (brigthenst makes changes to all elements) 80:00:00:00 for First LED element 00:00:00:40 for last LED element FF:FF:FF:C0 Mask for all LED's
> 
> Next goes spacing 00:00:00:00 And Y X for color White: 50:0d:54:39 Red: a3:db:54:78 Green: 4c:cc:99:99 Blue: 26:67:0f:5c
> 
> Everything ends: 02:bc
> 
> To save and run static preset Attribute: 0x050A Type: 0x41 Octet: 06:00:02:01:01:01:01
> 
> Every save increases first byte. Save/run2: 07:01:02:01:01:01:01 Save/run3: 08:02:02:01:01:01:01
> 
> To delete 3rd saved effect: 08:01:03:01:01:01:01
> 
> Hope it help's

Looks like a step in the right direction. Hopefully this is helpful to those that can - help!

luke5stevens, 2024-12-14:
Is there already more progress in that regard?

Some new instruction for less in-depth tech savvy people?

adamk125, 2024-12-14:
> Is there already more progress in that regard?
> 
> Some new instruction for less in-depth tech savvy people? @adamk125 @geduxas @samzorio
Havent heard or seem progress with the latest Z2M updates :( I'm not one of the smart ones unfortunately

luke5stevens, 2024-12-14:
@adamk125 okay sad, don't want to buy and run an additional Aqara hub just to use these effects. 

If there's a way to support this (maybe also financially) to make this happen, just let me know.

adamk125, 2024-12-14:
> @adamk125 okay sad, don't want to buy and run an additional Aqara hub just to use these effects.
> 
> If there's a way to support this (maybe also financially) to make this full integration happen, just let me know.

You dont need a hub to run the effects, you just need to use their App. Hopefully someone can make progress on the additional effects in HomeAssistant but for now, I've been using the solid colours that are exposed in HA to use the light in automations etc.

Maybe one day we'll get the full functionality in HA. Especially as Aqara are trying to be more HA friendly/certified.

luke5stevens, 2024-12-14:
@adamk125 how to do that? When I use the Aqara app and click on add the T1M device, I have to select one of their own hubs first. Is there a way to circumvent that? Thanks a lot!

phips2996, 2025-01-03:
Hello,
I'm also really interesstend in using the effects which are offered via Aqara app in HA.

BernardoEstrada, 2025-01-16:
Also very interested! Wish I could help but I'm pretty new to zigbee z2m

rider87, 2025-02-09:
Are there any news for it?

irekzielinski, 2025-02-09:
I think critical technical data is here, but we seem to lack connecting final dots. Maybe @Koenkk could help?

YvesZH, 2025-02-09:
I'm also waiting for a solution. As soon as it works, I'll buy 3 more Aqara T1M. (Actually, I don't understand Aqara, that they don't support and don't want to increase sales)

honzis1364, 2025-03-14:
Hey everyone, just wanted to hit you up with a temp solution. Aqara supports Scenes via Matter. So you we can control effects via HomeAssistant (and other platforms). 
1. Delete light in Z2M
2. Reset it (on/off 5 times)
3. Connect it to Aqara
4. If you don't have it, add Aqara Hub to HA via Matter 
5. Create scenes in the Aqara app with effects of the light(s)
6. Go to "Connect to other ecosystems" in the Aqara App
7. Scene sync and expose via Matter 
8. You can now turn on the effect via HA and use it in automations etc. 

And if you didn't know, you can expose Syncs as well (G3 Gestures, G4 doorbell press, camera notifications,...) 

https://github.com/user-attachments/assets/d116f04b-f1fa-469c-8b93-67f0da4867a3

michalg-, 2025-03-14:
In this case you need aqara hub device or not? I'm pretty sure that I tried this way before w/o hub and light was not listed.

honzis1364, 2025-03-14:
> What happens if you add the light directly to HA via Matter/Thread using something like a Sonoff or SLZB-06M stick or similar (without going via aqara)?
> 
> Are there any effects options that we can see/use or no?

My guess is NO. Since this is bound to Aqara app, everything is set via Aqara. Matter is only for the communication in this usage. So you communicate to Aqara via Matter to do something with the scene. If you'd create the scene, expose it via Matter, and delete it inside Aqara, it will disappear inside HA. 

**I don't wanna shine over the REAL PROBLEM -> Z2M support. Guys upthere worked their magic to read the data and create it via Z2M. This is just a workaround for those with Aqara Matter Hubs, the issue is not resolved by this.**

geduxas, 2025-03-14:
Maybe @banister4728 interested in extending support for this light and add RGB effect's to it i can confirm same information  for effect's also find out that first byte is just cheskum of rest packet..

```[checksum]:4[1-8,b]:02:01:01:01:01```
checksum is Sum of Bytes % 256  (CheckSum8 Modulo 256), 

> So i ended up buying a E1 aqara hub to sniff traffic of t1m.
> 
> Here is what i found so far looks like to control the effects its done via the following endpoint/cluster (One would assume it would be done via endpoint 2 but no)
> 
> ```
> Destination Endpoint: 1
>   Cluster: Manufacturer Specific (0xfcc0)
>   Profile: Home Automation (0x0104)
> ```
> 
> The build in effects are the following:
> 
> ```
> DINNER: 
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 47:41:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 47:41:02:01:01:01:01
> 		
> SUNSET:	
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 48:42:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 48:42:02:01:01:01:01
> 		
> AUTUMN:
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 49:43:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 49:43:02:01:01:01:01
> 		
> GALAXY:
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 4a:44:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 4a:44:02:01:01:01:01
> 		
> DAYDREAM:
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 4b:45:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 4b:45:02:01:01:01:01
> 		
> HOLIDAY: 
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 4c:46:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 4c:46:02:01:01:01:01
> 
> PARTY:
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 4d:47:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 4d:47:02:01:01:01:01
> 		
> METEOR:
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 4e:48:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 4e:48:02:01:01:01:01
> 		
> ALERT:
>     Command: Write Attributes (0x02)
>     Attribute Field, Octets: 51:4b:02:01:01:01:01
>         Attribute: 0x050a
>         Data Type: Octet String (0x41)
>         Octet String: 51:4b:02:01:01:01:01
> ```
> 
> There is also the option to add custom effects. Have not played with that yet will try to capture it later this week and try and figure it out.
> 
> ~As for getting the built-in effects working in z2m, this is all new to me will try to create an extended definition to get this working or at least try to get it working via the dev console of zigbee2mqtt (Devices --> dev console tab)~
> 
> Well tried a bunch of stuff nothing works guess what needs to be done to get this working is add the missing attribute (0x050a) to cluster 0xfcc0 (manuSpecificLumi)
> 
> If anyone with more experience would like to jump in or help me with this I would appreciate it.
> 
> Thanks all.

geduxas, 2025-03-19:
ok, i have tried to build external converter, and it doesn't work.. truly dont know why, sniffed protocol from aqara, and from z2m it looks same for effect's changing.. but in my build command's it responses with 0x83 instead of 0x80 in third byte of payload.. 

also i have noticed something what i didn't like at all, while pairing lamp with aqara hub, it perfoms some OTA updates (send's 13 image files with same version number (256)) (unable to find in aqara cloud, also made OTA dumps)
also i retrived OTA update history from aqara cloud, and here one of updates:

    {
      "releaseTime": "1691464255000",
      "necessary": 0,
      "fileSize": 386106,
      "updateLog": "1. Added "Preview function"\n2. Added the function of saving the newly created dynamic scene to the local device through OTA", [[Translated]]
      "state": 2,
      "firmwareVersion": "0.0.0_0023"
    },

my lamp is on  0.0.0_0026 version 

here is  file i have used to test:

```js
const lumi = require ('zigbee-herdsman-converters/lib/lumi');

const {Buffer} = require('node:buffer');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const {
    lumiAction, lumiOperationMode, lumiPowerOnBehavior, lumiZigbeeOTA,
    lumiSwitchType, lumiAirQuality, lumiVoc, lumiDisplayUnit, lumiLight,
    lumiOutageCountRestoreBindReporting, lumiElectricityMeter, lumiPower,
    lumiOverloadProtection, lumiLedIndicator, lumiButtonLock, lumiMotorSpeed,
    lumiOnOff, lumiLedDisabledNight, lumiFlipIndicatorLight, lumiPreventReset,
    lumiClickMode, lumiSlider, lumiSetEventMode, lumiSwitchMode, lumiVibration,
    lumiKnobRotation, lumiCommandMode, lumiBattery,
} = lumi.modernExtend;

const NS = 'zhc:lumi';
const {manufacturerCode} = lumi;

const {deviceEndpoints, identify, light, numeric, enumLookup, custom} = require('zigbee-herdsman-converters/lib/modernExtend');

const effects = {
    dinner: 0x41,
    sunset: 0x42,
    autumn: 0x43,
    galaxy: 0x44,
    daydream: 0x45,
    holiday: 0x46,
    party: 0x47,
    meteor: 0x48,
    alert: 0x4B,
};

const effectExtension = () => {
    const fromZigbee = [
        {
            cluster: 'manuSpecificLumi',
            type: ['attributeReport', 'readResponse'],
            convert: (model, msg, publish, options, meta) => {
                if (msg.data[0x050a] !== undefined) {
                    const value = [...msg.data[0x050a]];
                    if (value.length === 7 && value[2] === 0x80) {
                        const effect = Object.entries(effects).find(([, v]) => v === value[1])?.[0];
                        return effect ? {effect} : {};
                    }
                }
            },
        },
    ];

    const toZigbee = [
        {
            key: ['effect'],
            convertSet: async (entity, key, value, meta) => {
                const effectCode = effects[value];
                const checksum = (effectCode + 0x02 + 0x01 * 4) & 0xFF;
                const data = Buffer.from([checksum, effectCode, 0x02, 0x01, 0x01, 0x01, 0x01]);

                await entity.write('manuSpecificLumi',
                    {0x050a: {value: data, type: 0x41}},
                    {manufacturerCode, disableDefaultResponse: true}
                );

                return {state: {effect: value}};
            },
        },
    ];

    const exposes = [
        e.enum('effect', ea.STATE_SET, Object.keys(effects))
            .withDescription('Dynamic lighting effect')
    ];

    const configure = [
        // Set device to event reporting mode
        async (device, coordinatorEndpoint, logger) => {
            await device.getEndpoint(1).write('manuSpecificLumi',
                {mode: 1},
                {manufacturerCode, disableResponse: true}
            );
        },
        // Configure attribute reporting
        async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.configureReporting('manuSpecificLumi', [{
                attribute: 0x050a,
                minimumReportInterval: 0,
                maximumReportInterval: 3600,
                reportableChange: 1,
            }], {manufacturerCode});
        }
    ];

    return {
        fromZigbee,
        toZigbee,
        exposes,
        configure,
        isModernExtend: true
    };
};

const definition = {
        zigbeeModel: ['lumi.light.acn032'],
        model: 'CL-L02D (test)',
        vendor: 'Aqara (test)',
        description: 'Ceiling light T1M (test)',
        extend: [
            deviceEndpoints({endpoints: {white: 1, rgb: 2}}),
            lumiLight({colorTemp: true, endpointNames: ["white"]}),
            lumiLight({colorTemp: true, deviceTemperature: false, powerOutageCount: false, color: {modes: ["xy", "hs"]}, endpointNames: ["rgb"]}),
            lumiZigbeeOTA(),
            enumLookup({
                name: "power_on_behaviour",
                lookup: {on: 0, previous: 1, off: 2},
                cluster: "manuSpecificLumi",
                attribute: {ID: 0x0517, type: 0x20},
                description: "Controls the behavior when the device is powered on after power loss",
                zigbeeCommandOptions: {manufacturerCode},
            }),
            effectExtension(),
        ],
};

module.exports = definition;
```

jath58, 2025-03-19:
Same experience here.

I have also sniffed the zigbee communication from the Aqara hub M2 to the T1M light while setting the scenes predefined in the Aqara app (like "Dinner", "Sunset" etc.).

And I have made a Z2M converter that sends exactly the same zigbee commands, ie. the sniffs from/to the T1M are identical when paired with the Aqara hub and when paired with Z2M.
But still the T1M does not respond with lighting up with the expected scenes when connected to Z2M (as it does when connected to the Aqara Hub).

This leads me to suspect that the T1M somehow knows wheter its connected to a Aqara hub or not, either through some special commands received during/just after paring, or by knowing that its communicating with a aqara coordinator zigbee IEEE address (or some other mechanism). If this is true it becomes more than complicated to make the T1M scenes work with anything else than an Aqara hub.

johnsturgeon, 2025-03-19:
Wow, fantastic work even though the results are unfortunate.  I wonder if the ultimate solution might be to rip out the strip that's around the rim and just drop in my own with an ESP32 / WLED.

jath58, 2025-03-19:
geduxas, I think you are on the right track in assuming that the data for the scenes are pre-loaded by the Aqara hub just after the T1M are paired to the hub.
In my sniffing I have seen rather large data transfers from the hub to T1M, happening after the pairing proces have finished. This could very well be the scene pre-loads.

So it seems that the reason why T1M does not work with Z2M, when trying to turn on the scenes with the exact same zigbee commands, are that the scene setup data are missing in the T1M after the reset that are part of the re-pairing proces.

Or - Aqara have intentionally put in a blocking mechanism in the T1M firmware that does not allow the scene funtionality to be set by any means other than through an Aqara hub.

geduxas, 2025-04-13:
I think it's easier to run some wled with esp...

geduxas, 2025-04-13:
Truly didn't tried to investigate, but seams to me simple addressable rgb led strip.. 12V or even 5V..

geduxas, 2025-04-28:
60W for led stripe? It's overkill :D but not problems if you gave spare laying around :) i think it's enough to use some 5V usb adapter's.. 10-15w should be enough for this. Or use some sort of in pcb power supplies. But yes nice workaround! And i think it's only one reasonable Way to Walk away from tuya.. also! I think it's possible to change controllers inside of original power supply unit, there should be some sort of standard Tuya adapter's, and i think it's possible to interchangea to esp chip..

modem7, 2025-04-28:
It's unfortunately that this is what we have to do, especially with Aqara's "Home Assistant pledge". 

If we're going this route, we may as well go Govee at half the price, and install our LED's that way. Ah wait......Govee integration barely works (not the fault of the dev, but due to Govee themselves).

colinnewman2000, 2025-11-19:
What an amazing step forward. Nice !!

absent42, 2025-11-19:
> yes, i restarted the z2mqtt addon but i can not find it on any of my 4 T1M

I don't use Zigbee2MQTT as an addon but in a seperate LXC, maybe the directory for external converters is different, check the Zigbee2MQTT documentation. The device should have a little cyan external converter icon next to it if one's active. 

[image]

absent42, 2025-11-21:
> Just downloaded and tested the built in effects, fantastic! Thank you so much! I haven't had a chance to run through all the blueprint settings, but so far this is amazing.
> 
> Question [@absent42](https://github.com/absent42) -- what is the process for getting this into the next release?

Thanks. I'm not sure if I'll have everything quite ready to my satisfaction for the next release on the 1st December but it'll definitely be ready for the one after that. I'll have to refactor the code into the zigbee-herdsman repository, some of it split out to also be used in the new Aqara T2 bulb converter I've been doing.

johnsturgeon, 2025-11-21:
> Thanks. I'm not sure if I'll have everything quite ready to my satisfaction for the next release on the 1st December but it'll definitely be ready for the one after that. I'll have to refactor the [...]

I'll admit, I don't use blueprints, but after installing yours I was not able to use it to change / set the segment colors.  That might very well be a case of me not doing it right, I tried to create a script that used the blueprint to set the segments to a few colors, but it had no effect.

Thoughts?

geduxas, 2025-11-22:
> It fails when trying to activate a preset from the light's memory. I haven't actually tried it with the T1M but that's what happens with the T2 which shares the same effect mechanisms with the T1M, I've only had the light for about 5 days.

did you succeed uploading presset? I had sniffed whole traffic with zigbee sniffer, and just failed to do same for uploading pressets from z2m

geduxas, 2025-11-22:
How tou trigger action if preset are removed from lamp? :) o think it wasn't there if it's not uploaded.. i think lamp not stores them in memory.. 

But indeed, if it's possible accomplish without upload, why not...

geduxas, 2025-11-27:
In latest z2m, added external converter, restarted z2m, removed device in aqara, enabled join in z2m. Lamp automatically paired to z2m,a and everything was worked, so it's not required to turn off lights, unless they are not pired anywhere.

absent42, 2025-12-07:
@rbalaev Hi, do you know how many ring light segments the smaller version has please? I need to know this in the hope of getting the segment control to work properly on the smaller version without having one to physically test it. You can find out in the Aqara app or by counting black chips/copper cut marks on the LED strip going around the light. Thanks.

Jazkaloo, 2025-12-09:
Just tried it on my brand new light. It took a bit of tinkering, before I understood where to save the blueprints to. Imho the instructions could be a bit better.

I got it all working now. Love that you guys put all the work in it. 

Thanks.

geduxas, 2025-12-10:
@absent42 it's just simple addressable led, eatch segment's (chip) have 3 leds, which count's as one. 

@rbalaev I thin you can cut last segment and no one will notice :D

absent42, 2025-12-10:
@geduxas I'm well aware of how the light and LED strips work.

absent42, 2025-12-29:
@xLerate Thanks I'll put it in the README, I've only tested it with 0.0.0_0027 firmware. If it's the same as the T2 bulbs it might be possible to update the firmware via the Aqara Home App without a hub, just via Bluetooth, but I haven't tried.

Also I've just put up V0.1.0 of the Home Assistant integration for easier control of the light features. It's early days but the basic features are there.

domotick, 2026-01-08:
Hi everyone!

Thx you for all the work done here.

However, I'm having a problem with Home Assistant Scene. It seems to activate default effect first and I can't set a stable color. I have a similar device (an RGB bulb) that doesn't support scenes, but I can still set a fixed RGB color on it.

I don't know if the problem is with Z2M or something else, but here's my feedback.

My solution is a external definition of your integration but with effect disable.

[1 more reply]