Koenkk/zigbee2mqtt #22248: [Feature request]: Improvements to PJ-1203A - Bidirectional energy meter

schauveau, 2024-04-18:
### Is your feature request related to a problem? Please describe

The PJ-1203A is a Bidirectional dual channel energy meter. The current converter suffers from a few issues ; the most annoying one being that the energy flow direction is reported during the next update.  This is highly problematic because that means that incoherent data is published when the direction changes. 

For example, suppose that solar panels are producing an excess of 100W on channel X and that a device that consumes 500W is turned on. The following states will be published by Z2M:
  
      ...
      { "energy_flow_x":"producing" , "power_x":100 , ... }
      ...  
      { "energy_flow_x":"producing" , "power_x":400 , ... }
      ...  
      { "energy_flow_x":"consuming" , "power_x":400 , ... } 
      ...

So during a full update cycle (10s by default), the device is seen as producing 400W which is absolutely wrong. 
 

### Describe the solution you'd like

Using the information provided in https://github.com/Koenkk/zigbee2mqtt/issues/18419 , I have been working on an improved external converter that caches the channels datapoints and emit them together once a full and hopefully coherent set is collected.

So the new features are:
- Collect datapoints for each channel and emit them together once a full and hopefully coherent data set is collected.
- Per channel option to select when the channel data should be published:
    - as soon as possible but potentially with an incorrect energy flow direction.
    - or delayed with, hopefully, a correct energy flow direction.
- Detect missing datapoints using TUYA seq to invalidate ambiguous data sets. 
- Per channel option to indicate the energy flow direction using signed power values instead of a using a dedicated attribute (`energy_flow_x = consuming|producing`).
- Option to configure the update frequency from 3 to 60s.
- Two new attributes `timestamp_a|b` to indicate when `power_a|b` was received.

### Describe alternatives you've considered

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

// The PJ1203A is sending quick sequence of messages containing a single datapoint.
// A sequence occurs every `update_frequency` seconds (10s by default) 
//
// A typical sequence is composed of two identical groups for channel a and b. 
//
//     102 energy_flow_a
//     112 voltage
//     113 current_a
//     101 power_a
//     110 power_factor_a
//     111 ac_frequency
//     115 power_ab
//     ---
//     104 energy_flow_b
//     112 voltage
//     114 current_b
//     105 power_b
//     121 power_factor_b
//     111 ac_frequency
//     115 power_ab
//
// It should be noted that when no current is detected on channel x then
// energy_flow_x is not emited and current_x==0, power_x==0 and power_factor_x==100.
//
// The other datapoints are emitted every few minutes.  
//
// There is a known issue on the _TZE204_81yrt3lo with appVersion 74, stackVersion 0 and hwVersion 1. 
[... 456 lines cut ...]
      power_ab:
        device_class: power
        entity_category: diagnostic
        state_class: measurement
      current_a:
        device_class: current
        entity_category: diagnostic
        state_class: measurement
        enabled_by_default: false
      current_b:
        device_class: current
        entity_category: diagnostic
        state_class: measurement
        enabled_by_default: false
      timestamp_a:
        icon: mdi:clock
        device_class: timestamp
        entity_category: diagnostic
        enabled_by_default: false
      timestamp_b:
        icon: mdi:clock
        device_class: timestamp
        entity_category: diagnostic
        enabled_by_default: false
```

Koenkk, 2024-04-20:
Could you make a pull request for this?

schauveau, 2024-04-24:
I am working on a PR but I may not be able to pass the tests on my raspberry Pi 2 . The last time I tried, the processes were killed before the end.

Koenkk, 2024-04-24:
Just submit the PR, the pipeline will run the tests

schauveau, 2024-05-03:
This is now available in zigbee2mqtt master with the following features:
  - `update_frequency` can now be set between 3 and 60 seconds
  - new options to solve the incorrect energy flow at the expense of a delayed publication.
  - new options to publish signed power values instead of  'consuming/producing' in `energy_flow`  
  - new timestamp attributes to know when the power value was actually received from the device
     
Unfortunately, the Home Assistant integration cannot be improved from within the converter (see https://github.com/Koenkk/zigbee2mqtt/issues/22098 ) so I have the following in my `zigbee2mqtt/data/configuration.yaml` to have HA recognize `timestamp_a` and `timestamp_b` as proper timestamp instead of strings. 

```
devices:
  0xa4c138078e510fdc:
    ...
    homeassistant:
      timestamp_a:
        icon: mdi:clock
        device_class: timestamp
        entity_category: diagnostic
        enabled_by_default: false
      timestamp_b:
        icon: mdi:clock
        device_class: timestamp
        entity_category: diagnostic
        enabled_by_default: false
```

closed 2024-05-03 by schauveau

thisIsLiving, 2024-10-08:
@schauveau, are you able to assist with #24259? I think it's the same meter as PJ-1203A but with different datapointids. I'm not sure how to adapt the code in tuya.ts into a .js to test if your improvements work on this device.