Koenkk/zigbee2mqtt #22256: UNSUPPORTED_ATTRIBUTE reading attributes on HS2AQ-EF-3.0

HyperBCS, 2024-04-19:
### What happened?

I am attempting to read values from heimanSpecificAirQuality[Cluster 0xFC81] attribute Language[AttrID 0xF000] and I am receiving a status 134[UNSUPPORTED_ATTRIBUTE] as the response. I understand this device is not technically supported, however I contacted HEIMAN support to send me the API spec and it seems to match except the command IDs noted [here](https://github.com/Koenkk/zigbee-herdsman/blob/d855b3bf85a066cb7c325fe3ef0006873c735add/src/zcl/definition/cluster.ts#L4814).

This issue occurs with any attributes under this cluster ID. I've attached the API spec.
[f559af52f9ad5fe36cf0bf6dc17a4764.pdf](https://github.com/Koenkk/zigbee2mqtt/files/15033066/f559af52f9ad5fe36cf0bf6dc17a4764.pdf)

### What did you expect to happen?

I expect to receive a language string back based on the attribute I read with no error

### How to reproduce it (minimal and precise)

Go to dev console, select cluster heimanSpecificAirQuality and choose any attribute to read.

### Zigbee2MQTT version

1.36.1 commit: ffc2ff1

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus Gateway

### Setup

Pi

### Debug log

```
zigbee2mqtt  | Zigbee2MQTT:info  2024-04-19 00:13:25: MQTT publish: topic 'zigbee2mqtt/Living Room', payload '{"aqi":4,"battery":100,"battery_state":"charged","hcho":0.024,"humidity":31,"last_seen":"2024-04-19T04:13:25.431Z","linkquality":123,"pm10":3,"pm25":4,"temperature":24.88,"voltage":0}'
zigbee2mqtt  | Zigbee2MQTT:debug 2024-04-19 00:13:25: Received Zigbee message from 'Bedroom', type 'attributeReport', cluster 'heimanSpecificAirQuality', data '{"pm10measuredValue":3}' from endpoint 1 with groupID 0
zigbee2mqtt  | 2024-04-19T04:13:25.928Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,129,252,30,44,1,1,0,114,0,98,32,65,0,0,8,24,53,10,3,240,33,3,0,30,44,11,32]
zigbee2mqtt  | 2024-04-19T04:13:25.928Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,129,252,30,44,1,1,0,114,0,98,32,65,0,0,8,24,53,10,3,240,33,3,0,30,44,11,32]
zigbee2mqtt  | 2024-04-19T04:13:25.928Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,129,252,30,44,1,1,0,114,0,98,32,65,0,0,8,24,53,10,3,240,33,3,0,30,44,11] - 32
zigbee2mqtt  | 2024-04-19T04:13:25.928Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":64641,"srcaddr":11294,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":114,"securityuse":0,"timestamp":4268130,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,53,10,3,240,33,3,0]}}
zigbee2mqtt  | 2024-04-19T04:13:25.929Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":53,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":61443,"dataType":33,"attrData":3}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":11294,"endpoint":1,"linkquality":114,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
zigbee2mqtt  | 2024-04-19T04:13:25.930Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt  | Zigbee2MQTT:info  2024-04-19 00:13:25: MQTT publish: topic 'zigbee2mqtt/Bedroom', payload '{"aqi":4,"battery":100,"battery_low":false,"battery_state":"charged","hcho":0.02,"humidity":31,"last_seen":"2024-04-19T04:13:25.929Z","linkquality":114,"pm10":3,"pm25":3,"temperature":26.46}'
zigbee2mqtt  | 2024-04-19T04:13:36.138Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,129,252,30,44,1,1,0,99,0,245,220,74,0,0,8,24,54,10,4,240,33,0,0,30,44,11,86]
zigbee2mqtt  | 2024-04-19T04:13:36.138Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,129,252,30,44,1,1,0,99,0,245,220,74,0,0,8,24,54,10,4,240,33,0,0,30,44,11,86]
zigbee2mqtt  | 2024-04-19T04:13:36.138Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,129,252,30,44,1,1,0,99,0,245,220,74,0,0,8,24,54,10,4,240,33,0,0,30,44,11] - 86
zigbee2mqtt  | 2024-04-19T04:13:36.139Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":64641,"srcaddr":11294,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":99,"securityuse":0,"timestamp":4906229,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,54,10,4,240,33,0,0]}}
zigbee2mqtt  | 2024-04-19T04:13:36.141Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":54,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":61444,"dataType":33,"attrData":0}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":11294,"endpoint":1,"linkquality":99,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
zigbee2mqtt  | 2024-04-19T04:13:36.144Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt  | Zigbee2MQTT:debug 2024-04-19 00:13:36: Received Zigbee message from 'Bedroom', type 'attributeReport', cluster 'heimanSpecificAirQuality', data '{"tvocMeasuredValue":0}' from endpoint 1 with groupID 0
zigbee2mqtt  | Zigbee2MQTT:info  2024-04-19 00:13:36: MQTT publish: topic 'zigbee2mqtt/Bedroom', payload '{"aqi":4,"battery":100,"battery_low":false,"battery_state":"charged","hcho":0.02,"humidity":31,"last_seen":"2024-04-19T04:13:36.141Z","linkquality":99,"pm10":3,"pm25":3,"temperature":26.46}'
zigbee2mqtt  | Zigbee2MQTT:debug 2024-04-19 00:13:48: Received MQTT message on 'zigbee2mqtt/Bedroom/1/set' with data '{"read":{"attributes":["language"],"cluster":"heimanSpecificAirQuality","options":{}}}'
zigbee2mqtt  | Zigbee2MQTT:debug 2024-04-19 00:13:48: Publishing 'set' 'read' to 'Bedroom'
zigbee2mqtt  | 2024-04-19T04:13:48.448Z zigbee-herdsman:controller:endpoint ZCL command 0x84fd27fffe8b627f/1 heimanSpecificAirQuality.read(["language"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4619,"transactionSequenceNumber":null,"writeUndiv":false})
zigbee2mqtt  | 2024-04-19T04:13:48.450Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x84fd27fffe8b627f:11294/1 (0,0,1)
zigbee2mqtt  | 2024-04-19T04:13:48.451Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":11294,"destendpoint":1,"srcendpoint":1,"clusterid":64641,"transid":61,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[20,11,18,41,0,0,240]}}
zigbee2mqtt  | 2024-04-19T04:13:48.452Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,17,36,1,30,44,1,1,129,252,61,0,30,7,20,11,18,41,0,0,240,139]
zigbee2mqtt  | 2024-04-19T04:13:48.467Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt  | 2024-04-19T04:13:48.467Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt  | 2024-04-19T04:13:48.468Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt  | 2024-04-19T04:13:48.468Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt  | 2024-04-19T04:13:48.469Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt  | 2024-04-19T04:13:48.472Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,61,251]
zigbee2mqtt  | 2024-04-19T04:13:48.472Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,61,251]
zigbee2mqtt  | 2024-04-19T04:13:48.472Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,61] - 251
zigbee2mqtt  | 2024-04-19T04:13:48.473Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":61}
zigbee2mqtt  | 2024-04-19T04:13:48.473Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt  | 2024-04-19T04:13:48.499Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,129,252,30,44,1,1,0,99,0,211,166,86,0,0,8,12,11,18,41,1,0,240,134,30,44,11,172]
[... cut ...]
zigbee2mqtt  | 2024-04-19T04:13:48.545Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,62,248]
zigbee2mqtt  | 2024-04-19T04:13:48.545Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,62,248]
zigbee2mqtt  | 2024-04-19T04:13:48.545Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,62] - 248
zigbee2mqtt  | 2024-04-19T04:13:48.546Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":62}
zigbee2mqtt  | 2024-04-19T04:13:48.546Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

```

RobLeighton67, 2024-04-19:
I'd suggest updating the firmware on the Sonoff P.
I've run the unit and found that the stock z2m code generates an UNSUPPORTED_ATTRIBUTE error when configuring reporting for the Heiman Specific HCHO measurement - do you see that error (or any other) in your log when the device joins?

HyperBCS, 2024-04-19:
Yes I saw the UNSUPPORTED_ATTRIBUTE when configuring the device. I commented it out in my device configuration. I've just updated the firmware to 20230507 and still see the issue. This is what shows up in the debug log. Were you able to figure out any solution?

```
zigbee2mqtt  | Zigbee2MQTT:error 2024-04-19 11:58:22: Failed to configure 'Living Room', attempt 3 (Error: ZCL command 0x84fd27fffe8b625d/1 heimanSpecificFormaldehydeMeasurement.configReport([{"attribute":"measuredValue","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4619,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
zigbee2mqtt  |     at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28)
zigbee2mqtt  |     at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:762:22)
zigbee2mqtt  |     at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:561:9)
zigbee2mqtt  |     at Object.formAldehydeMeasuredValue (/app/data/extension/externally-loaded.js:24:25)
zigbee2mqtt  |     at Object.configure (/app/data/extension/externally-loaded.js:64:13)
zigbee2mqtt  |     at Configure.configure (/app/lib/extension/configure.ts:122:13)
zigbee2mqtt  |     at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

HyperBCS, 2024-04-19:
Just an update. If you remove the manufacturerCode from the cluster definition things work fine.

closed 2024-11-16 by github-actions[bot]