Koenkk/zigbee2mqtt #22257: [Feature request]: Philips 929002398602, modify recall_0 and recall_1 behaviour

splump, 2024-04-19:
### Is your feature request related to a problem? Please describe

The "Hue" button on the Philips 929002398602 remote assumes that scene 0 and scene 1 is available to use, and will cycle through those scenes. Zigbee2mqtt does not allow saving scenes to scene 0 and thus the "Hue" button becomes not very usable. 

### Describe the solution you'd like

Preferably, if possible it would be great if the "Hue" button could cycle all available scenes in the group it is currently bound to. 

### Describe alternatives you've considered

If not possible to cycle all scenes, then is it possible to shift the scenes it cycles to 1 and 2, since those can actually be saved to groups and used?

### Additional context

No context, but will happily answer any potential questions

thk-socal, 2024-07-11:
@splump you can save scene 0 in a "group" you cannot save scene 0 at the "device" level as the "group" is then group 0. However, if you bind the Hue dimmer to a "group" with the Scenes attribute, then you get commands that have "action":"recall_0" or "action":"recall_1" which include the group ID of the bound group "action_group": 51. Where "51" is the Zigbee group number. 

IMHO, the better feature would be for the recall_0 and recall_1 to be actioned against the bound group so that you can bind these to any group and then get the 0/1 scenes stored at the group level.

splump, 2024-07-28:
@thk-socal at the time when i created this ticket, it wasn’t possible to save scene with id0 in a group. It’s since been changed though.

51 isn’t the assigned group for all dimmers, they all have different groups assigned to them. Mine are all different. You can just check whatever group your particular dimmer is assigned to out of the box, then create a group with that same particular group id and bind the dimmer to it and it will behave the way you suggest it should.

closed 2025-01-06 by github-actions[bot]