Koenkk/zigbee2mqtt #22258: Entity 'homeassistant/sensor' is unknown

BWFCGUIXT, 2024-04-19:
### What happened?

Not sure when this started but for a while I keep getting:

`Entity 'homeassistant/sensor' is unknown`

Multiple times on startup of Zigbee2Mqtt

### Zigbee2MQTT version

1.36.1-dev commit: d9628df

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

Docker

moster11, 2024-04-19:
I can confirm these errors happening in my instance, too. No idea what this means, though.

[image]

Koenkk, 2024-04-20:
Could you provide your z2m `configuration.yaml`?

Koenkk, 2024-04-25:
I cannot reproduce it, debug logging when I do it:

```
[2024-04-25 21:56:46] debug: 	z2m: Received MQTT message on 'homeassistant/sensor/bla-bla-bla' with data ''
```

Can you provide your debug log?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

timmo65, 2024-05-02:
I have also had this for some time - keep trying to ignore it but it bugs me ....

[image]

aperezva, 2024-05-03:
The same error, and persits

Koenkk, 2024-05-04:
Please provide the info as requested in https://github.com/Koenkk/zigbee2mqtt/issues/22258#issuecomment-2078070547 , otherwise I cannot help

BWFCGUIXT, 2024-05-06:
@Koenkk  Hope this helps :) 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/15217334/log.txt)

theDiverDK, 2024-05-10:
I see the same error in 1.37.1

sebastienserre, 2024-05-12:
> Please provide the info as requested in [#22258 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/22258#issuecomment-2078070547) , otherwise I cannot help

Same issue here but adding 
```
advanced:
  log_debug_to_mqtt_frontend: true
  ```
  issue this in HA:
  Integration error: advanced - Integration 'advanced' not found.

theDiverDK, 2024-05-12:
Here is my log.txt fil, hope it helps.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/15287456/log.txt)

dingausmwald, 2024-05-12:
same for me with switch, not sensor

ytugarev, 2024-05-13:
In my setup this error appears to be related to thermostat valves (both Bosch and Danfoss) , after homeassistant/sensor/*valve address*/setpoint_change_source/config'.
Attaching the log
[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15292912/log.log)

BWFCGUIXT, 2024-05-15:
@ytugarev  can confirm that I also have Danfoss in my home, so maybe it's somehow related.

theDiverDK, 2024-05-15:
@BWFCGUIXT and @ytugarev 

I also have Danfoss Ally thermostats.

I have 8 of them.

And i get 8 times the error within same second, and 2 seconds later i get another 8 times the same error.

So you might be on to something.

sebastienserre, 2024-05-15:
No Danfoss but Aqara [SRTS-A01](https://www.zigbee2mqtt.io/devices/SRTS-A01.html#aqara-srts-a01) on my side.

aperezva, 2024-05-15:
The same, Aqara on my side

timmo65, 2024-05-15:
I posted earlier and have 6x that error message and also have 6x Trvs - 3 are Hive (rebranded Danfoss) and 3 Sonoff  TRVSBs - so looks like that might be where to look.

mainmind83, 2024-05-16:
Same problem after restore full backup HassOS:

`[2024-05-16 15:16:25] error: 	z2m: Entity 'homeassistant/number' is unknown
[2024-05-16 15:16:25] error: 	z2m: Entity 'homeassistant/number' is unknown`

Core 2024.5.3
Supervisor 2024.05.1
Operating System 12.3
Frontend 20240501.1
Zigbee2MQTT 1.37.1-1

difrost, 2024-05-19:
Same issue here with Aqara SRTS-A01, four errors returned for two devices:
```
z2m: Received MQTT message on 'homeassistant/binary_sensor/0x54ef4410007f622a/setup/config' with data '{"availability":[
{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x54ef4410007f622a"],"manufacturer":"Aqara","model
":"Smart radiator thermostat E1 (SRTS-A01)","name":"SalonTRV","sw_version":"0.0.0_1030","via_device":"zigbee2mqtt_bridge_0xe0798dfffeeafb62"},"device_class":"running","
entity_category":"diagnostic","object_id":"salontrv_setup","origin":{"name":"Zigbee2MQTT","sw":"1.37.0","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload
_on":true,"state_topic":"zigbee2mqtt/SalonTRV","unique_id":"0x54ef4410007f622a_setup_zigbee2mqtt","value_template":"{{ value_json.setup }}"}'
z2m: Entity 'homeassistant/binary_sensor' is unknown
```
```
z2m: Received MQTT message on 'homeassistant/binary_sensor/0x54ef4410007f300c/setup/config' with data '{"availability":[
{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x54ef4410007f300c"],"manufacturer":"Aqara","model
":"Smart radiator thermostat E1 (SRTS-A01)","name":"AntresolaTRV","sw_version":"0.0.0_1030","via_device":"zigbee2mqtt_bridge_0xe0798dfffeeafb62"},"device_class":"runnin
g","entity_category":"diagnostic","object_id":"antresolatrv_setup","origin":{"name":"Zigbee2MQTT","sw":"1.37.0","url":"https://www.zigbee2mqtt.io"},"payload_off":false,
"payload_on":true,"state_topic":"zigbee2mqtt/AntresolaTRV","unique_id":"0x54ef4410007f300c_setup_zigbee2mqtt","value_template":"{{ value_json.setup }}"}'
z2m: Entity 'homeassistant/binary_sensor' is unknown
```
```
z2m: Received MQTT message on 'homeassistant/binary_sensor/0x54ef4410007f622a/setup/config' with data '{"availability":[
{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x54ef4410007f622a"],"manufacturer":"Aqara","model
":"Smart radiator thermostat E1 (SRTS-A01)","name":"SalonTRV","sw_version":"0.0.0_1030","via_device":"zigbee2mqtt_bridge_0xe0798dfffeeafb62"},"device_class":"running","
entity_category":"diagnostic","object_id":"salontrv_setup","origin":{"name":"Zigbee2MQTT","sw":"1.37.0","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload
_on":true,"state_topic":"zigbee2mqtt/SalonTRV","unique_id":"0x54ef4410007f622a_setup_zigbee2mqtt","value_template":"{{ value_json.setup }}"}'
z2m: Entity 'homeassistant/binary_sensor' is unknown
```
```
z2m: Received MQTT message on 'homeassistant/binary_sensor/0x54ef4410007f300c/setup/config' with data '{"availability":[
{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x54ef4410007f300c"],"manufacturer":"Aqara","model
":"Smart radiator thermostat E1 (SRTS-A01)","name":"AntresolaTRV","sw_version":"0.0.0_1030","via_device":"zigbee2mqtt_bridge_0xe0798dfffeeafb62"},"device_class":"runnin
g","entity_category":"diagnostic","object_id":"antresolatrv_setup","origin":{"name":"Zigbee2MQTT","sw":"1.37.0","url":"https://www.zigbee2mqtt.io"},"payload_off":false,
"payload_on":true,"state_topic":"zigbee2mqtt/AntresolaTRV","unique_id":"0x54ef4410007f300c_setup_zigbee2mqtt","value_template":"{{ value_json.setup }}"}'
z2m: Entity 'homeassistant/binary_sensor' is unknown
```

jeleniain, 2024-05-19:
Having the same errors with my Aqara TRVs:
`[2024-05-19 20:13:35] error:    z2m: Entity 'homeassistant/binary_sensor' is unknown
[2024-05-19 20:13:35] error:    z2m: Entity 'homeassistant/binary_sensor' is unknown`

and

`[2024-05-19 20:13:51] error:    z2m: Publish 'set' 'sensor' to 'Office_Radiator' failed: 'Error: ZCL command 0x54ef44100067368c/1 manuSpecificLumi.write({"65522":{"value":[170,113,52,68,19,90,2,65,49,102,74,65,79,61,5,84,239,68,16,0,103,54,140,0,21,141,0,1,157,27,152,8,0,7,253,22,10,2,10,201,232,177,184,212,218,207,223,192,235,0,0,0,0,0,1,61,4,101],"type":65}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'`

Please let me know if more information is needed.
Thanks!

Koenkk, 2024-05-19:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-19 by Koenkk