Koenkk/zigbee2mqtt #22263: [New device support]: Zigbee Model: TS0001_power Not supported

Destroyer061090, 2024-04-20:
### Link

https://a.aliexpress.com/_EwCpvtV

### Database entry

{"id":18,"type":"Router","ieeeAddr":"0xa4c138fc65500a2f","nwkAddr":39377,"manufId":4417,"manufName":"_TZ3000_kqvb5akv","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"+9ï¿½-\u0012ï¿½\u0000\u0000\u0000\u0012","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_kqvb5akv","zclVersion":3,"appVersion":72,"modelId":"TS0001","powerSource":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":241,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,9]}},"manuSpecificBosch":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffef8b317","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffef8b317","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffef8b317","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-892627439},"lastSeen":1713597413370}

### Comments

Hello,

Z2MQTT is connecting my device TuYa | TS0001_power to the following link product
https://www.zigbee2mqtt.io/devices/TS0001_power.html#tuya-ts0001_power but it is as the picture
image
[image]

### External definition

```shell
None
```

mrmaximas, 2024-04-20:
In the TUYA zoo there are many identical devices in different closures, there is no possibility to add a picture for each one.

Destroyer061090, 2024-04-20:
Important is working good!😅

closed 2024-11-17 by github-actions[bot]