Koenkk/zigbee2mqtt #22271: ROUTE_ERROR_ADDRESS_CONFLICT

fred-gb, 2024-04-20:
### What happened?

I get lots of error logs lines=
```
Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
```

And many devices are unusable. Pass to timeout state and back. 

These modules, no communication anymore.
https://www.zigbee2mqtt.io/devices/PIR313-E.html

Network is slow, lot of timeouts.

And when I get back to driver `eszp` and my second backup Sonoff ZBDongle-E (under 7.3 firmware), my zigbee network is working normally again.

I know in life we as to do somes choices, but between switches PTM216 and multisensor PIR313-E, it's difficult to decide! 

### What did you expect to happen?

For operating PTM216 switches, I follow this tutorial: https://github.com/Koenkk/zigbee2mqtt/discussions/21462

### How to reproduce it (minimal and precise)

- Upgrade Sonoff ZBDongle-E 7.4.1
- Update Z2M to stable 1.36.1
- Change driver adapter to `ember`

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

7.4.1 [GA]

### Adapter

Sonoff ZBDongle-E

### Setup

LXC container Ubuntu, Nomad as orchestrator, Z2M Docker version

### Debug log

```
Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
```

When I try to wakeup device on battery:
```
024-04-20 17:12:08Failed to configure '0x3c6a2cfffed22b52', attempt 1 (Error: Bind 0x3c6a2cfffed22b52/2 msIlluminanceMeasurement from '0xe8e07efffeefa989/1' failed (Delivery failed for {"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4160,"groupId":0,"sequence":126}) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:94:31) at EmberAdapter.onMessageSentDeliveryFailed (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:556:30) at Ezsp.emit (node:events:517:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:3960:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:804:18) at Ezsp.tick (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:461:18) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))
```

When I activate "Permit join"
```
[ZDO] Failed permit joining request with status=NETWORK_BUSY.
[ZDO] Failed permit joining request with status=NETWORK_BUSY.
[ZDO] Failed permit joining request with status=NETWORK_BUSY.
[ZDO] Failed permit joining request with status=NETWORK_BUSY.
[ZDO] Failed permit joining request with status=NETWORK_BUSY.
~x~> [ZCL BROADCAST] Failed to send with status=NETWORK_BUSY.
~x~> [ZCL BROADCAST] Failed to send with status=NETWORK_BUSY.
~x~> [ZCL BROADCAST] Failed to send with status=NETWORK_BUSY.
~x~> [ZCL BROADCAST] Failed to send with status=NETWORK_BUSY.
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'motDEtat'
~x~> [ZCL BROADCAST] Failed to send with status=NETWORK_BUSY.
~x~> [ZCL BROADCAST] Failed to send with status=NETWORK_BUSY.
~x~> [ZCL BROADCAST] Failed to send with status=NETWORK_BUSY.
~x~> [ZCL BROADCAST] Failed to send with status=NETWORK_BUSY.
```

Outsidewall, 2024-04-25:
Hello, Have you tried updating the dongle's firmware to 7.4.2, I had many errors with 7.4.1 the uplift fixed them.

fred-gb, 2024-04-26:
😞 

With `latest-dev`same errors

```
error 2024-04-26 17:59:27Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":251} messageTag=255]
warning 2024-04-26 17:59:29Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
```

and docker logs
```
[2024-04-26 19:00:07] debug:    zh:ember:ezsp: ezspIncomingNetworkStatusHandler(): callback called with: [errorCode=ROUTE_ERROR_ADDRESS_CONFLICT], [target=0]
[2024-04-26 19:00:07] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-04-26 19:00:07] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-04-26 19:00:07] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4]
[2024-04-26 19:00:07] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=0] Added to rxQueue
[2024-04-26 19:00:07] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1]
[2024-04-26 19:00:07] debug:    zh:ember:ezsp: <=== [FRAME: ID=196:"INCOMING_NETWORK_STATUS_HANDLER" Seq=51 Len=8]
[2024-04-26 19:00:07] debug:    zh:ember:ezsp: ezspIncomingNetworkStatusHandler(): callback called with: [errorCode=ROUTE_ERROR_ADDRESS_CONFLICT], [target=0]
[2024-04-26 19:00:07] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-04-26 19:00:07] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-04-26 19:00:07] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4]
[2024-04-26 19:00:07] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=1] Added to rxQueue
[2024-04-26 19:00:07] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2]
[2024-04-26 19:00:07] debug:    zh:ember:ezsp: <=== [FRAME: ID=196:"INCOMING_NETWORK_STATUS_HANDLER" Seq=51 Len=8]
[2024-04-26 19:00:07] debug:    zh:ember:ezsp: ezspIncomingNetworkStatusHandler(): callback called with: [errorCode=ROUTE_ERROR_ADDRESS_CONFLICT], [target=0]
[2024-04-26 19:00:07] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-04-26 19:00:07] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-04-26 19:00:07] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4]
[2024-04-26 19:00:07] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=2] Added to rxQueue
[2024-04-26 19:00:07] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3]
[2024-04-26 19:00:07] debug:    zh:ember:ezsp: <=== [FRAME: ID=196:"INCOMING_NETWORK_STATUS_HANDLER" Seq=51 Len=8]
[2024-04-26 19:00:07] debug:    zh:ember:ezsp: ezspIncomingNetworkStatusHandler(): callback called with: [errorCode=ROUTE_ERROR_ADDRESS_CONFLICT], [target=0]
[2024-04-26 19:00:07] warning:  zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-04-26 19:00:10] debug:    z2m: Saving state to file /app/data/state.json
[2024-04-26 19:00:12] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-04-26 19:00:12] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4]
[2024-04-26 19:00:12] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=3] Added to rxQueue
[2024-04-26 19:00:12] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4]
[2024-04-26 19:00:12] debug:    zh:ember:ezsp: <=== [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=51 Len=40]
[2024-04-26 19:00:12] debug:    zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":222}], [lastHopLqi=76], [lastHopRssi=175], [sender=4754], [bindingIndex=255], [addressIndex=255], [messageContents=081b0a01002045e2ff201fe4ff2000]
[2024-04-26 19:00:12] debug:    zh:controller: Data is from unknown device with address '4754', skipping...
[2024-04-26 19:00:18] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-04-26 19:00:18] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4]
[2024-04-26 19:00:18] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=4] Added to rxQueue
[2024-04-26 19:00:18] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5]
[2024-04-26 19:00:18] debug:    zh:ember:ezsp: <=== [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=51 Len=39]
[2024-04-26 19:00:18] debug:    zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":197}], [lastHopLqi=52], [lastHopRssi=169], [sender=22388], [bindingIndex=255], [addressIndex=255], [messageContents=08120a010020c0e2ff2036e4ff2000]
[2024-04-26 19:00:18] debug:    zh:controller: Data is from unknown device with address '22388', skipping...
```

😭

fred-gb, 2024-04-27:
I tried today on my "production" site to switch with dongle with 7.4.2 firmware and the latest-dev docker images. 

And I have lot of errors, I can't use devices on battery.

When I try to open network to new device:

```
error 2024-04-27 09:02:42[ZDO] Failed permit joining request with status=NETWORK_BUSY.
error 2024-04-27 09:02:42~x~> [ZCL BROADCAST] Failed to send with status=NETWORK_BUSY.
```

And conflict errors.

Unusable... 😭

Outsidewall, 2024-04-27:
@fred-gb You might have some sort of network storm going on, I wonder if removing the devices from power/unpluging them, restart the docker container, and carefully put them back one at a time, maybe one of them is causing the issue.

fred-gb, 2024-05-02:
Thanks @Nerivec 

I restart one dev plateform.

This is a docker-compose:
```yaml
version: '3.8'
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:1.37.0
      # image: koenkk/zigbee2mqtt:latest-dev
    restart: unless-stopped
    volumes:
      - ./z2m:/app/data
      - /run/udev:/run/udev:ro
    ports:
      # Frontend port
      - 8080:8080
    environment:
      - TZ=Europe/Paris
    devices:
      # Make sure this matched your adapter location
      - /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240219194401-if00:/dev/ttyACM0

  mosquitto:
    image: eclipse-mosquitto
    hostname: mosquitto
    container_name: mosquitto
    restart: unless-stopped
    ports:
      - "1883:1883"
      - "9001:9001"
    volumes:
      - ./mosquitto:/etc/mosquitto
      - ./mosquitto/mosquitto.conf:/mosquitto/config/mosquitto.conf
```

Z2M configuration:
```yaml
homeassistant: false
availability: true
advanced:
  log_level: debug
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
permit_join: false
frontend:
  port: 8080
  host: 0.0.0.0
  auth_token: zigbee2mqtt
  url: http://192.168.10.116:8080
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.10.116
  user: admin
  password: admin
serial:
  port: /dev/ttyACM0
  adapter: ember
device_options:
  legacy: false
devices:
  '0xa49e69fffe051947':
    friendly_name: '0xa49e69fffe051947'
```

My Z2M frontend:
[image]
[... 116 lines cut ...]
```
warning 2024-05-02 17:33:49Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
error 2024-05-02 17:33:56Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":38} messageTag=255]
error 2024-05-02 17:33:56Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":48} messageTag=255]
warning 2024-05-02 17:34:03Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
warning 2024-05-02 17:34:17Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "12040".
warning 2024-05-02 17:34:17Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
error 2024-05-02 17:35:31Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":219} messageTag=255]
error 2024-05-02 17:35:31Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":221} messageTag=255]
error 2024-05-02 17:35:31Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":217} messageTag=255]
warning 2024-05-02 17:35:48Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
error 2024-05-02 17:35:56Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":253} messageTag=255]
error 2024-05-02 17:36:01Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":3} messageTag=255]
error 2024-05-02 17:36:06Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":129} messageTag=255]
error 2024-05-02 17:36:06Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":7} messageTag=255]
error 2024-05-02 17:36:11Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":93} messageTag=255]
error 2024-05-02 17:36:16Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":21} messageTag=255]
warning 2024-05-02 17:36:18Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
error 2024-05-02 17:36:21Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":182} messageTag=255]
error 2024-05-02 17:36:21Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":97} messageTag=255]
error 2024-05-02 17:36:46Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":142} messageTag=255]
warning 2024-05-02 17:36:48Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
```

Thanks if you can help me!

Nerivec, 2024-05-02:
This is likely all linked to the 0 conflict. You will have to figure out which device is causing this strange behavior. I'm guessing the rest of your troubles will disappear after that.

I see you mentioned a "backup" adapter. If you have two adapters with the exact same network configuration, running at the same time, around the same location, that will definitely create issues. Unplug one when you run the other.
Similarly, if you have two adapters made to run on the same network configuration, but they have different IEEE address, that can create issues if you swap back and forth.

If the above doesn't fix it, can you post the log file it created when you ran that last test?

Last resort, like @Outsidewall said, you may have to remove all devices and add them back one by one, until you find the culprit.
I see you have a lixee device, try that one first; the zlinky_tic has caused plenty of troubles in the past.

---

Did you use the firmware from darkxst? If not, try that one [ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl)
_I have yet to have time to test 7.4.2, though the changelog from silabs seems minimal. Still, you could try 7.4.1 instead if nothing else seems to work._

lettore, 2024-05-03:
I have the same problem with EMBER driver, I thought something happened to my board while switching to 7.4.2 firmware but returning back to EZSP adapter mode in 1.37.0 version fixed all errors.
I got the same `Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".` error and plus a `[ZDO clusterId=32799 sender=37662] Support not implemented upstream.`
In the meanwhile IKEA remote stopped working, but from the dashboard I was able to turn on or off devices.
I also had some CRC errors at the start of the container, something I never seen before.

LDprg, 2024-05-03:
Having same issue with the ember driver.

phenix1990, 2024-05-04:
Same here. I have a docker installation on a Pi5 with Sonof zigbee dongle-e.

I moved all files and devices from my previous pi4 and other zigbee stick to the new setup.
Z2m is on version 1.37.0
Sonof firmware was first 7.4.1, same issues on 7.4.2
Moved same time from ezsp to ember

short time after booting i get 5 times per second:
`warning zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".`

After a hour or two the network is dead with error
`error: 	zh:ember:uart:ash: Received ERROR from NCP, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
error: 	zh:ember:ezsp: [SEND COMMAND] Cannot send second one before processing response from first one.`

Then i have to restart the container two make it work for another period.

Nerivec, 2024-05-04:
@LDprg That's likely a battery-powered device that needs to wake up before it can configure properly. Try pushing a button on it before doing the configure. (You may need to push the button at 2-3 seconds intervals until it's done to keep it awake.)

Can someone with the `Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".` issue provide a full `debug`-level log from z2m start to after that error starts occurring?

---

PS: Also, anyone not using a firmware from darkxst, please switch to it ([Sonoff Dongle E 7.4.1](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl)). There has been issues with some others. https://github.com/Koenkk/zigbee2mqtt/discussions/21462#discussioncomment-8536401

phenix1990, 2024-05-04:
So i went back to ember driver with firmware 7.4.1 and debug level.

I dont know why but z2m crashed also with ezsp this evening with 7.4.2.

@Nerivec here are the loggings from container restart.
Sorry for the long list, can I make this code fence scrollable in vertical?

```[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: <-- [0112a1a96c2a15093dad7e]
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: <-- DATA (0,1,0): 0112a1a96c2a15093dad7e
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> ACK  (1)
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> [8160597e]
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: <-- ACK (1): 0112a1a96c2a15093dad7e
[2024-05-04 20:28:42] debug: 	zh:ezsp:ezsp: <== Frame: 508001380000bb
[2024-05-04 20:28:42] debug: 	zh:ezsp:ezsp: <== 0x38: {"_cls_":"sendMulticast","_id_":56,"_isRequest_":false,"status":0,"sequence":187}
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: -+- waiting (1) success
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: <-- [7d3112b1a97d312a1652f8b54ad7555593b463f5d8abedce9874fbdf6e8bf67d5e3f647c7e]
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: <-- DATA (1,1,0): 1112b1a9112a1652f8b54ad7555593b463f5d8abedce9874fbdf6e8bf67e3f647c7e
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> ACK  (2)
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> [82503a7e]
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: <-- ACK (1): 1112b1a9112a1652f8b54ad7555593b463f5d8abedce9874fbdf6e8bf67e3f647c7e
[2024-05-04 20:28:42] debug: 	zh:ezsp:ezsp: <== Frame: 509001450003e0a12100f2ff0001fdffbbff000000ffff06190d020a0000
[2024-05-04 20:28:42] debug: 	zh:ezsp:ezsp: <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":3,"apsFrame":{"profileId":41440,"sequence":187,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":255,"groupId":65533,"options":256},"lastHopLqi":255,"lastHopRssi":0,"sender":0,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[25,13,2,10,0,0]}}
[2024-05-04 20:28:42] debug: 	zh:ezsp: processMessage: {"messageType":3,"apsFrame":{"profileId":41440,"sequence":187,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":255,"groupId":65533,"options":256},"lqi":255,"rssi":0,"sender":0,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[25,13,2,10,0,0]}}
[2024-05-04 20:28:42] debug: 	zh:ezsp:ezsp: ==> clearTransientLinkKeys: null
[2024-05-04 20:28:42] debug: 	zh:ezsp:ezsp: ==> {"_cls_":"clearTransientLinkKeys","_id_":107,"_isRequest_":true}
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> DATA (1,2,0): 5100016b00
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> [127d3321a93f2a87287e]
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: -?- waiting (2)
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: <-- [227d33a1a93f2a759c7e]
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: <-- DATA (2,2,0): 2213a1a93f2a759c7e
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> ACK  (3)
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> [83401b7e]
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: <-- ACK (2): 2213a1a93f2a759c7e
[2024-05-04 20:28:42] debug: 	zh:ezsp:ezsp: <== Frame: 5180016b00
[2024-05-04 20:28:42] debug: 	zh:ezsp:ezsp: <== 0x6b: {"_cls_":"clearTransientLinkKeys","_id_":107,"_isRequest_":false}
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: -+- waiting (2) success
[2024-05-04 20:28:42] debug: 	zh:ezsp:ezsp: ==> permitJoining: {"duration":0}
[2024-05-04 20:28:42] debug: 	zh:ezsp:ezsp: ==> {"_cls_":"permitJoining","_id_":34,"_isRequest_":true,"duration":0}
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> DATA (2,3,0): 520001220000
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> [231021a9762a1527cb7e]
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: -?- waiting (3)
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: <-- [3310a1a9762a1532607e]
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: <-- DATA (3,3,0): 3310a1a9762a1532607e
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> ACK  (4)
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> [8430fc7e]
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: <-- ACK (3): 3310a1a9762a1532607e
[2024-05-04 20:28:42] debug: 	zh:ezsp:ezsp: <== Frame: 528001220000
[2024-05-04 20:28:42] debug: 	zh:ezsp:ezsp: <== 0x22: {"_cls_":"permitJoining","_id_":34,"_isRequest_":false,"status":0}
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: -+- waiting (3) success
[2024-05-04 20:28:42] debug: 	zh:controller:database: Writing database to '/app/data/database.db'
[2024-05-04 20:28:42] debug: 	zh:controller: Creating coordinator backup
[2024-05-04 20:28:42] debug: 	zh:ezsp:backup: creating backup
[2024-05-04 20:28:42] debug: 	zh:ezsp:ezsp: ==> version: {"desiredProtocolVersion":13}
[2024-05-04 20:28:42] debug: 	zh:ezsp:ezsp: ==> {"_cls_":"version","_id_":0,"_isRequest_":true,"desiredProtocolVersion":13}
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> DATA (3,4,0): 53000100000d
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> [347d3121a9542a7d38755f7e]
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: -?- waiting (4)
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: <-- [447d31a1a9542a7d38b049e0bbfc7e]
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: <-- DATA (4,4,0): 4411a1a9542a18b049e0bbfc7e
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> ACK  (5)
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: --> [8520dd7e]
[2024-05-04 20:28:42] debug: 	zh:ezsp:uart: <-- ACK (4): 4411a1a9542a18b049e0bbfc7e
[2024-05-04 20:28:42] debug: 	zh:ezsp:ezsp: <== Frame: 53800100000d021074
[... cut ...]
[2024-05-04 20:28:57] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0 frmNum=3] Added to rxQueue
[2024-05-04 20:28:57] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4]
[2024-05-04 20:28:57] debug: 	zh:ember:ezsp: <=== [FRAME: ID=196:"INCOMING_NETWORK_STATUS_HANDLER" Seq=55 Len=8]
[2024-05-04 20:28:57] debug: 	zh:ember:ezsp: ezspIncomingNetworkStatusHandler(): callback called with: [errorCode=ROUTE_ERROR_ADDRESS_CONFLICT], [target=0]
[2024-05-04 20:28:57] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
```

lettore, 2024-05-04:
> @LDprg That's likely a battery-powered device that needs to wake up before it can configure properly. Try pushing a button on it before doing the configure. (You may need to push the button at 2-3 seconds intervals until it's done to keep it awake.)
> 
> Can someone with the `Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".` issue provide a full `debug`-level log from z2m start to after that error starts occurring?
> 
> PS: Also, anyone not using a firmware from darkxst, please switch to it ([Sonoff Dongle E 7.4.1](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl)). There has been issues with some others. [#21462 (reply in thread)](https://github.com/Koenkk/zigbee2mqtt/discussions/21462#discussioncomment-8536401)

Mine is not battery powered and without any changes works fine with EZSP, just when turning to EMBER driver after initialization it reports `Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".` more time per second, it doesn't receive remote commands , and the same as @phenix1990 after some minutes it hangs up as the board is not responding anymore. In fact to have it working again I have to remove it from USB port and replug. The only problem I have with 1.37.0 with EZSP and Sonoff Dongle E 7.4.2 is that when I start pairing it throws another error, ERROR : Received undefined command from '0'. This don't affect anything, pairing works as expected. I think ember driver is not mature enough for production.

Nerivec, 2024-05-04:
@phenix1990 You should be able to upload the file directly next time with `Attach files` 😉

Can you please check with `ezsp`, if you see the same messages in the logs? In `ember` this is a `warning`, but in `ezsp`, it is a `debug` so you would not see it unless you are looking for it in the logs directly (you will need log level set to `debug` of course).
Look for lines that say
```logs
handleNetworkStatus: nwk="0", errorCode="13"
```
which is the equivalent of the ones you are receiving with `ember`.

phenix1990, 2024-05-04:
@Nerivec yep, found this error with ezsp
```
handleNetworkStatus: nwk=0, errorCode=13
```

so its not related to ember? 
I also updated z2m from 1.32.2 to 1.37 due to some interesting changes.
As mentioned other change was switching from Slaesh's CC2652RB stick to sonof dongle-e.
And pi4 to pi5.

A lot of changes but i was optimistic. Now its dark and i cant control some lights from time to time. Good my wife already sleeping today :innocent:

Nerivec, 2024-05-04:
When you switched from zstack to ezsp/ember, you re-paired everything, correct? _Just making sure that by some weird coincidence you didn't manage to start ezsp/ember with a zstack-based network..._ 😅

You may have to re-pair devices one by one, until you start seeing the error again (make sure to run the network with zero devices for a few minutes first, just in case the error appears right away for some reason...). You don't have many, so it should be pretty quick. Easiest way to get a "clean slate" (remove all devices and ensure the creation of a new network), is to use `GENERATE` for network settings (also pick a good channel while you're at it, [see docs](https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html#network-config))
Basically something like this:
```yaml
advanced:
  channel: 20
  network_key: GENERATE
  pan_id: GENERATE
  ext_pan_id: GENERATE
```

phenix1990, 2024-05-04:
Infact i didnt re-paired all devices because it seem to work for some people without. 

For most of my devices re-pairing is no problem, just 3 hard wired ones will be a bit painfull to process.
But then i have a task for the rest of the weekend. Thanks for the help so far!

fred-gb, 2024-05-08:
Bonjour, 👋🏻 

After somes days. It's a very good surprise for me.

First step: `upgrade to zigbee2mqtt, firmware 7.4.2, drivers EZSP`

First good surprise, my Button PTM216 finally works!

I read in the last few days, that switching to ember drivers will cause re-inclusion of modules. So this morning I'm getting ready to do this.

https://github.com/Koenkk/zigbee2mqtt/assets/17061623/126a71f6-6c6e-4a9b-bb85-5328a240fef3

😅 

I switched to EMBER drivers in debug mode and... ALL WORKS FINE WITHOUT NEED REINCLUDED ANY DEVICE! 🥳 

Very nice surprise to start this day!

Now I'm waiting for next z2m release after https://github.com/Koenkk/zigbee-herdsman/pull/1052

I let open this issue for @phenix1990 .

Nerivec, 2024-05-09:
@phenix1990 I included a few fixes for the next release (hotfix coming out soon). Let me know how it goes after that.

phenix1990, 2024-05-10:
> @phenix1990 I included a few fixes for the next release (hotfix coming out soon). Let me know how it goes after that.

I already updated z2m but with older adapter version and ezsp driver. 
Tomorrow evening I hopefully will have time to test with ember and 7.4.1 Firmware.

Nerivec, 2024-05-16:
Closing this to avoid mixing issues any further.
@phenix1990 If you encounter any trouble, please open a new issue.

closed 2024-05-16 by Nerivec

zerpex, 2024-06-01:
Hi there,

Same issue with SonOff Dongle E 7.4.2 firmware even with z2m 1.38.0-dev.
Only way to recover : switch back to ezsp.

LDprg, 2024-06-02:
> Hi there,
> 
> Same issue with SonOff Dongle E 7.4.2 firmware even with z2m 1.38.0-dev.
> Only way to recover : switch back to ezsp.

Try 7.4.3 firmware downloaded from https://github.com/darkxst/silabs-firmware-builder/blob/4.4.3/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl

zerpex, 2024-06-03:
Hi @LDprg . I just updated to 7.4.3.
This is better (Z2M is usable) and seems to work. Anyway, I still have the ROUTE_ERROR_ADDRESS_CONFLICT for "0".

How can I change this network ID ? Had a look around, be I was unable to find where...

phenix1990, 2024-06-03:
> Hi @LDprg . I just updated to 7.4.3. This is better (Z2M is usable) and seems to work. Anyway, I still have the ROUTE_ERROR_ADDRESS_CONFLICT for "0".
> 
> How can I change this network ID ? Had a look around, be I was unable to find where...

its written some posts above. Just enter the keyword Generate in your z2m config and repair your devices. also was necessary for me.
```
advanced:
  channel: 20
  network_key: GENERATE
  pan_id: GENERATE
  ext_pan_id: GENERATE
```

LDprg, 2024-06-03:
@zerpex You will need to set following flags in the config to GENERATE. Z2M will automatically fill them with new values

```yaml
advanced:
  pan_id: GENERATE
  ext_pan_id: GENERATE
  network_key: GENERATE
```

You might also want to delete the mqtt cache since it might cause weird malformed data in mqtt. This shouldn't affect zigbee2mqtt directly.

zerpex, 2024-06-03:
Thanks a lot.
I confiirm that with firmware 7.4.3 and these parameters, everything works with ember !

AlexMKX, 2025-10-19:
encountered the same Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0". while moving from cc to ember.
Fixed without repairing all devices. The procedure follows:
1. disable pairing and remove from the network UPS-powered mains devices. 
2. turn off z2m
3. turn off stick - literally, remove from the port, or from the outlet.
4. turn off all mains-powered devices for 1 minute (everything which could act as a router must be either turned off or removed from the network - see pt.1 )
5. turn on mains-powered devices.
6. Wait for 20 minutes - this should put network in "recovery" mode (AFAIK)
7. turn on coordinator, z2m and pairing. The pairing should be turned on for 1 hour or so (it could be done by HA automation). 
8. Wait 30-240 minutes to settle the network. It will start working right away, but errors could happen for next 4 hours (in my ~180 device network it took couple hours).

IgnacioHR, 2026-01-28:
Hi,

I landed here this morning after 5 days having all sort of issues with my Zigbee network at home. A few facts:

1. I've 103 devices in my network.
2. Everything was running fine since a couple of months.
3. I purchased a new Sonoff Dongle-M because I wanted to connect Zigbee2Mqtt via tcp rather than a serial port (z2m is installed on a VM and I wanted to achieve HA for that VM)
4. The new dongle has Firmware version 7.4.5
5. In the process to migrate the coordinator I assumed the backup/restore process will be writing the coordinator ieee address but THAT'S NOT TRUE so I made the mistake of replacing old coordinador with a new coordinator with different ieee address.
6. Coordinator version was 7.4.5 and I previously was running on 8.0.2

It is 5 days now and the network is a total disaster.
The process to turn everything off for 1 minute is not possible at home.
There should exists tools to determine what is causing the issue and why the network is not converging to stability
Without that tools the entire Zigbee specification is in danger of adoption.
The problem and lack of clear solutions have convinced me to stop recommending Zigbee to my friends.

My wife wants to go back to the old hand actionable switches and I'm starting to think she is right. Hopefully I've no small kids at home.

Due to the reason there are no clean solutions and tools I'm just testing random things. I've disconnected from my network almost all devices I can. No solution yet.
There are devices inside boxes that can't be opened without breaking the paints in the walls.

Hopefully this message can reach the right persons that can develop the right tools to evaluate a network and provide clear guidance of what to do.