Koenkk/zigbee2mqtt #22274: ubisys H1 not working

jobabi, 2024-04-20:
### What happened?

I´m using FHEM and zigbee2mqtt on a Linux computer.
The ubisys H1 can be coupled without problems to zigbee2mqtt, but if I send a "Occupied heating setpoint" from FHEM to zigbee2mqtt I receive this error:
Publish `set` ´occupied_heating_setpoint` to "device name" failed: `Error: ´occupied_heating_setpoint` is not a number, got string (14.0)`
However, If I use the same setup on a raspberry (different building) it is working without problems.
I would be happy for a hint what is running wrong.

I first thought that there a problem with 1.0 and 1,0 , but an other device (Siterwell Modell GS361A-H04) is working in both buildings / setups.

### Zigbee2MQTT version

    1.33.1 commit: 51f15bd3

### Adapter firmware version

20210708

### Adapter

slaesh

### Setup

apple notebook with Linux

Sailing74, 2024-04-30:
Hi, did you try sending without the decimal point ("14" instead of "14.0")?

jobabi, 2024-06-06:
sorry for the delay. Also without decimal it gets the same error

jobabi, 2024-06-06:
Zigbee version 1.35.1 commit 3c96204
Coordinator zStack3x0
zigbee-herdsman-converters version 18.9.0
zigbee herdman version 0.30.0

sjorge, 2024-06-07:
That's an older version of z2m, I'm pretty sure I fixed that issue... twice already. Try on 1.37.0 or newer

jobabi, 2024-06-07:
I updated to 1.38.0 but still not working. same error message

I tried the H1 with an other setup Zigbee 1.33.1. Here is working (but this setup is running in a different building....

sjorge, 2024-06-07:
Working fine here, ...

```
mosquitto_pub -t zigbee2mqtt/trv/kitchen/get/occupied_heating_setpoint -m ''
```

```
[2024-06-07 19:53:51] debug:    z2m: Publishing get 'get' 'occupied_heating_setpoint' to 'trv/kitchen'
[2024-06-07 19:53:53] debug:    z2m: Received Zigbee message from 'trv/kitchen', type 'readResponse', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":1800}' from endpoint 1 with groupID 0
```

```
mosquitto_pub -t zigbee2mqtt/trv/kitchen/set/occupied_heating_setpoint -m 18.5
```

```
[2024-06-07 19:54:51] debug:    z2m: Publishing 'set' 'occupied_heating_setpoint' to 'trv/kitchen'
[2024-06-07 19:54:53] debug:    z2m: Received Zigbee message from 'trv/kitchen', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":1850}' from endpoint 1 with groupID 0
```

```
mosquitto_pub -t zigbee2mqtt/trv/kitchen/set/occupied_heating_setpoint -m 18
```

```
[2024-06-07 19:55:05] debug:    z2m: Publishing 'set' 'occupied_heating_setpoint' to 'trv/kitchen'
[2024-06-07 19:55:06] debug:    z2m: Received Zigbee message from 'trv/kitchen', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":1800}' from endpoint 1 with groupID 0
```

Both float and int work. Also works fine via the z2m frontend and node-red here...

z2m version: 1.38.0-dev commit: [fd3d9a31](https://github.com/Koenkk/zigbee2mqtt/commit/fd3d9a31)
zhc version: 19.51.0
zh verison: 0.49.3

jobabi, 2024-06-07:
Well my problem seems more the communication between FHEM and zigbee2mqtt.
z2m works fine with the H1
if FHEM sends the temperature settings then z2m creates the mentioned error.

sjorge, 2024-06-07:
Sound more like an issue in FHEM (not sure what that even is) not sending a correct payload then.

Sailing74, 2024-06-08:
Jochen, there appears to be a related thread in the fhem forum with some hints on , vs . and when to use quotes. Maybe there are some helpful hints in there?

https://forum.fhem.de/index.php?topic=137326

jobabi, 2024-06-09:
Thank you for your comments:
here is the link o FHEM https://fhem.de/ (FHEM replaces Mosquitto or simliar software and has a nice frontend) 
Well, in the FHEM Forum, they suggested that it is a z2m problem.

I´m puzzled, by the observation, that other heating devices like this one:
https://www.zigbee2mqtt.io/devices/SMART-HEAT10.html
is working with the same commands sent from FHEM.
So it seems for me that in z2m for this ubisys device is something different ??

sjorge, 2024-06-09:
I tried with node-red-contrib-aedes, another MQTT broker to rule out it only working on moquitto. And it is working fine too. This time I was injecting with a mqtt-out node in node-red.

I'm still inclined to say it's a FHEM issue.

Do note that your SMART-HEAT10 uses the Tuya abstraction layer on top of zigbee and the H1 uses the standard ZCL hvacThermostat. The don't share much device specific code at all.

I'm fairly confident the code for the ZCL hvacThermostat is correct and doesn't do anything weird. I'm not sure on the Tuya side as I tend to be very picky on what devices I get/work so I always have the ZCL spec to fall back on.

Sailing74, 2024-06-09:
Does changing the setting in z2m‘s webif work? If so, but not in FHEM, some tweaking of the device definition in FHEM might be required. If it doesn’t work from z2m‘s webif, either, the problem is probably elsewhere.

jobabi, 2024-06-09:
you mean webif = frontend ?
Yes, with the z2m frontend I can change all settings without problems.

Sailing74, 2024-06-09:
In such situations, for me some changes to the z2m device configuration in FHEM solved the problem. There is a MQTT subsection in the forum.

closed 2024-12-29 by github-actions[bot]