Koenkk/zigbee2mqtt #22275: [Feature request]: Search function in Map view

peter1rhodes, 2024-04-20:
### Is your feature request related to a problem? Please describe

I often want to check how a particular device connects into the network, i.e. what devices it speaks to and how clear the signal is. Although marvellous in many ways, the Z2M Map is a bit useless for this, with hundreds of lines criss-crossing everywhere, and the text all jumbled up and hard to read.

### Describe the solution you'd like

It would be great to able to find particular nodes in the map by having them highlighted when I enter a search term in a text box. For example, if I want to see the kitchen light, typing "kitchen" might make all devices containing "kitchen" to change text colour and be brought to the front

### Describe alternatives you've considered

Honestly, I can't think of another way. Right now I just have to do without, and this seems like a natural solution

### Additional context

[image]

timdonovanuk, 2024-04-22:
Control F in your browser...

[image]

And then if you click on it, it singles it out:

[image]

peter1rhodes, 2024-05-23:
Thank you, Ctrl-F is a really good suggestion. I think it would still be a useful feature for the map itself, but this will help me for now

> Control F in your browser... [...]

closed 2024-12-29 by github-actions[bot]