Koenkk/zigbee2mqtt #22277: Z-Station error- Failure send version

michapr, 2024-04-21:
### What happened?

after installation Zigbee2MQTT does not start, get error
`Error: Failure send version:{"type":"Buffer","data":[0,0,0,4]}`

No idea what I can try - maybe you can help?
(I know that adapter is "Experimental" ;) ...)

### How to reproduce it (minimal and precise)

my configuration:
```
homeassistant: false
frontend: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.178.92
serial:
  port: /dev/serial/by-id/usb-Silicon_Labs_CP2105_Dual_USB_to_UART_Bridge_Controller_012BCB52-if01-port0
  adapter: ezsp
  baudrate: 115200
advanced:
  network_key:
    - 233
...
    - 229
  pan_id: 55246
  ext_pan_id:
    - 126
...
    - 58
```

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

unknown, latest (new adapter)

### Adapter

Z-Wave.Me Z-Station dual Zigbee & Z-Wave 

### Setup

plain installatoion, Pi3, bookworm

### Debug log

```
> zigbee2mqtt@1.36.1 start
> node index.js

Zigbee2MQTT:debug 2024-04-21 10:56:45: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
Zigbee2MQTT:info  2024-04-21 10:56:45: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-04-21.10-56-44' filename: log.txt
Zigbee2MQTT:info  2024-04-21 10:56:45: Starting Zigbee2MQTT version 1.36.1 (commit #ffc2ff1)
Zigbee2MQTT:info  2024-04-21 10:56:45: Starting zigbee-herdsman (0.40.3)
Zigbee2MQTT:debug 2024-04-21 10:56:45: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[126,16,86,28,148,93,99,58],"networkKey":"HIDDEN","panID":55246},"serialPort":{"adapter":"ezsp","baudRate":115200,"path":"/dev/serial/by-id/usb-Silicon_Labs_CP2105_Dual_USB_to_UART_Bridge_Controller_012BCB52-if01-port0"}}'
  zigbee-herdsman:adapter:ezsp:debg Adapter concurrent: 8 +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[233,133,127,233,136,36,218,9,183,204,96,126,232,92,139,229],"panID":55246,"extendedPanID":[126,16,86,28,148,93,99,58],"channelList":[11]},"serialPort":{"baudRate":115200,"path":"/dev/serial/by-id/usb-Silicon_Labs_CP2105_Dual_USB_to_UART_Bridge_Controller_012BCB52-if01-port0","adapter":"ezsp"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:ezsp:uart RTS/CTS config is off, enabling software flow control. +0ms
  zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/serial/by-id/usb-Silicon_Labs_CP2105_Dual_USB_to_UART_Bridge_Controller_012BCB52-if01-port0","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true} +1ms
  zigbee-herdsman:adapter:ezsp:uart Serialport opened +17ms
  zigbee-herdsman:adapter:ezsp:uart Uart reseting +1ms
  zigbee-herdsman:adapter:ezsp:uart --> Write reset +2ms
  zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e] +0ms
  zigbee-herdsman:adapter:ezsp:uart -?- waiting reset +9ms
  zigbee-herdsman:adapter:ezsp:uart <-- [c1020b0a527e] +0ms
  zigbee-herdsman:adapter:ezsp:uart <-- RSTACK c1020b0a527e +1s
  zigbee-herdsman:adapter:ezsp:uart RSTACK Version: 2 Reason: RESET_SOFTWARE frame: c1020b0a527e +1ms
  zigbee-herdsman:adapter:ezsp:uart -+- waiting reset success +3ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> version: {"desiredProtocolVersion":4} +0ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"version","_id_":0,"_isRequest_":true,"desiredProtocolVersion":4} +4ms
  zigbee-herdsman:adapter:ezsp:uart --> DATA (0,0,0): 00000004 +2s
  zigbee-herdsman:adapter:ezsp:uart --> [004221a850ed2c7e] +3s
  zigbee-herdsman:adapter:ezsp:uart -?- waiting (1) +3ms
  zigbee-herdsman:adapter:ezsp:uart <-- [a0547d3a7e] +2s
  zigbee-herdsman:adapter:ezsp:uart <-- NAK (0): a0541a7e +6ms
  zigbee-herdsman:adapter:ezsp:uart NAK Unexpected packet sequence 0 +1ms
  zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":1} after 4000ms +4s
  zigbee-herdsman:adapter:ezsp:uart -!- break waiting (1) +0ms
  zigbee-herdsman:adapter:ezsp:uart Can't send DATA frame (0,0,0): 00000004 +0ms
  zigbee-herdsman:adapter:ezsp:uart ->> DATA (0,0,1): 00000004 +502ms
  zigbee-herdsman:adapter:ezsp:uart --> [084221a850ef017e] +5s
  zigbee-herdsman:adapter:ezsp:uart -?- rewaiting (1) +1ms
  zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":1} after 4000ms +4s
  zigbee-herdsman:adapter:ezsp:uart -!- break rewaiting (1) +1ms
  zigbee-herdsman:adapter:ezsp:uart Can't resend DATA frame (0,0,1): 00000004 +0ms
  zigbee-herdsman:adapter:ezsp:ezsp onSerialReset() +9s
  zigbee-herdsman:adapter:ezsp:driv onEzspReset() +0ms
  zigbee-herdsman:adapter:ezsp:driv Reset connection. +2ms
  zigbee-herdsman:adapter:ezsp:driv Stopping driver +1ms
  zigbee-herdsman:adapter:ezsp:ezsp Closing Ezsp +5ms
  zigbee-herdsman:adapter:ezsp:uart Closing UART +7ms
Zigbee2MQTT:error 2024-04-21 10:56:58: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-04-21 10:56:58: Failed to start zigbee
Zigbee2MQTT:error 2024-04-21 10:56:58: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-04-21 10:56:58: Exiting...
Zigbee2MQTT:error 2024-04-21 10:56:58: Error: Failure send version:{"type":"Buffer","data":[0,0,0,4]}
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:664:23
    at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Ezsp.version (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:469:24)
    at Driver.startup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:180:9)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:62:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:109:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)
```

Botched1, 2024-04-24:
Same issue here.

Any guidance @PoltoS ?

michapr, 2024-04-24:
I have found out that there was already an issue here:
https://github.com/Koenkk/zigbee2mqtt/issues/21973  ("closed")

Will be solved by Z-Wave.Me with new firmware in next days (I think so).

Botched1, 2024-04-24:
I wouldn't hold my breath on that... 

I have no doubt a new firmware will come out, but whether it's actually posted on their portal or it's done soon is dubious. They are known for super timely response on firmware or software, although they make nice products.

michapr, 2024-04-24:
I'm sure, because already made a test of new FW ;)

Botched1, 2024-04-24:
You aren't reading what I wrote... Made for himself/created doesn't equal released or posted. :)

But to stay positive, I hope it gets released soon.

PoltoS, 2024-04-24:
A new firmware will be published within a few days. Please stay tuned

Botched1, 2024-04-25:
That's what I get for being a naysayer. My apologies, and thank you.

PoltoS, 2024-04-26:
You can find the new firmware here: https://z-wave.me/support/tools-for-zigbee-openthread-ble/

Botched1, 2024-04-26:
> You can find the new firmware here: https://z-wave.me/support/tools-for-zigbee-openthread-ble/

Thank you! Now I just have to figure out how to get zme_zigbee to run on my raspberry pi 5... I must be doing something as it won't run.

PoltoS, 2024-04-26:
Please show:
`file /bin/ls; file ./zme_zigbee`

Botched1, 2024-04-26:
> Please show: `file /bin/ls; file ./zme_zigbee`

/bin/ls: ELF 64-bit LSB pie executable, ARM aarch64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1, BuildID[sha1]=9f127c37a4c459cf01639f6ded2fcf11a49d3da9, for GNU/Linux 3.7.0, stripped

./zme_zigbee: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, BuildID[sha1]=11de62005d720c912422caf775e5ee4f661ba1f1, for GNU/Linux 3.2.0, stripped

PoltoS, 2024-04-26:
I think you need to install 32 bit libraries

Botched1, 2024-04-26:
> I think you need to install 32 bit libraries

Thank you for the pointer. It is not that simple though. After installing 32bit libraries then you get this error:
./zme_zigbee: error while loading shared libraries: libdl.so.2: ELF load command address/offset not page-aligned

So then one has to change kernel images to one that supports 4k pages...

Probably easier for me to just find a standard ubuntu system or a Raspberry Pi 4 and flash the firmware there.

PoltoS, 2024-04-26:
Doesn't this help?

`dpkg --add-architecture armhf`

Otherwise use Ubuntu or Windows

Botched1, 2024-04-26:
> Doesn't this help?
> 
> `dpkg --add-architecture armhf`
> 
> Otherwise use Ubuntu or Windows

No, I tried that but it gives the same error. Based on other posts online it is well known with the raspberry pi that you have to change kernel entry in config.txt to get the pages to align.

Not a problem, I'll do it on ubuntu or windows. I have both available.

Botched1, 2024-04-26:
I flashed the firmware in Windows.

After putting it back on my rpi5, now zigbee2mqtt starts with no issues.

Nice work, and thank you!

closed 2024-04-26 by michapr

SmartHomeForDummies, 2024-04-26:
Can also confirm that with the new Firmware Z-Station is working with Z2M. Both eszp and ember adapter typ.
But with ZHA still no luck.

PoltoS, 2024-04-27:
Please share info on the ZHA issue. ZHA was working even before the update.