Koenkk/zigbee2mqtt #22278: NodOn SIN-4-2-20 input wired switch 1 controls output 2 and vise-versa

fkhn, 2024-04-21:
### What happened?

After firmware upgrade to 3.1.0-1.0.7 by zigbee2mqtt the input switch 1  controls output 2 and input switch 2 controls output 1.
Prior to the upgrade input 1 was controlling output 1 as expected (same for 2). 

### What did you expect to happen?

no crossing.

### Zigbee2MQTT version

1.36.1 commit: ffc2ff1

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Debian : 6.1.0-20-arm64 #1 SMP Debian 6.1.85-1 (2024-04-11) / with homebridge and home bridge-z2m 1.9.3

titou4307, 2024-04-21:
Same problem....

titou4307, 2024-04-26:
I had the Nodon support today : 

A developper look at the problem ( they have the info....)

Bonus : App Nodon Connect wil be updated too

titou4307, 2024-05-01:
New App is come out
After install and update of my two Nodon, the two modules are KO

I have send a message to Nodon Support

azzieg, 2024-05-21:
Same here. It's a mess :-)

I have two SIN-4-2-20 relays. After updating them to the latest version using Zigbee2MQTT they switched outputs (or inputs). After further updates using the Nodon Connect app (Zigbee2MQTT was not willing to update beyond the broken version) and reinitializing this got fixed, but one relay is at 3.1.0-1.0.7 (same version as before the update but works?) and the other at 3.2.0-1.0.7 (it updated twice in a row in the app). For both the app reports this is the latest version.

I also have SIN-4-1-20 at this got updated to 3.1.0-1.1.2 using the app (Zigbee2MQTT reported that there was an update, but could not install it). Unfortunately, this completely broke (disconnected) the wired up switch. I can turn on the light using Zigbee2MQTT, but not the wall switch. Not a Zigbee2MQTT fault of course, but you might want to avoid buying, or at least updating Nodon relays, if the company creates such a mess.

titou4307, 2024-05-25:
Hello
After reinit all the SIN 4.2.20....I succed to update my 3 devices : 

One is at firmware 3.2.0-1.4.5
One is at firmware 3.1.0-1.4.5
One is at firmware 3.2.0-1.0.7

All the device are in the App at the last firmware...

Falconpage, 2024-07-27:
Hello,

Same issue here since today, Home Assistant aske me again to update the firmware of my NodOns modules (it always aske me since last year) but this time the firmware was upgraded sucessfully.
Yes but now all my switchs are inverted, not only on Zigbee2mqtt but physically too ! 

Modules : SIN-4-2-20
Firmware : 3.1.0-1.1.1

I have tried reset it and setup it again, the same, powered cycle the modules, etc.
The NodOn app say I'm up to date.

Falconpage, 2024-07-27:
@titou4307 how had you successfully updated your modules to a new firmware ?

RBI42, 2024-07-31:
I have talked with the support and an update come tomorrow :

Sur la version dans laquelle vous avez mis à jour le produit, il y a effectivement un dysfonctionnement. 

Nous avons poussé une nouvelle version sur Z2M qui corrige cela.
Elle devrait être disponible d'ici demain.

RBI42, 2024-08-14:
> > I have talked with the support and an update come tomorrow :
> > Sur la version dans laquelle vous avez mis à jour le produit, il y a effectivement un dysfonctionnement.
> > Nous avons poussé une nouvelle version sur Z2M qui corrige cela. Elle devrait être disponible d'ici demain.
> 
> @RBI42 Have you had any news? I haven't seen any new version available yet.

Yes patched firmware can be updated with zigbee2mqtt.
Same version shown in HA, but OTA update detect a new version if you ask to update.
Everything is fixed now.

zaxxon72, 2024-08-18:
Hi, I had modules with the crossed inputs as well. Nodon support told me to to a downgrade in the App. iPhone app does not allow that, and on a borrowed Android phone, the app does not want to scan for modules like on the iPhone, but wants to pair by reading QR on the module (which are screwed into the walls, yeah, I'm gonna rip them all out...)

Nevertheless, following the hint from Nodon support, I updated to the 'newest' version in Z2M because according to them the crossed input was fixed in the newest version which internally seems to be a downgrade.

now I have six modules on 3.3.0-1.0.7 which fixes the crossed inputs and seems to be fault free otherwise. five other modules are on 3.3.0-1.1.1, no crossed inputs, but when I turn them off via HA/Z2M, this action seems to be lost to the module because I when I press the hardware switch that is wired to the input, nothing happens (even in HA). I have to press the switch again to turn the light off/on. The input and the output seem to be decoupled (which would be a great feature, if it could be used on purpose.....)

The iPhone app reports all those modules to be on 3.3.0 and up to date, and Z2M also does not want to upgrade anything further.

does anyone else have such behaviour? (everything is fixed now sounds like a stretch...)

egretzbergerj, 2024-11-30:
Hi  zaxxon72, I've such devices too. Updated to firmware 3.5.0 1.0.7 last week with the same issue.
"when I press the hardware switch that is wired to the input, nothing happens (even in HA). I have to press the switch again to turn the light off/on."
Sent message to nodon support but no answer yet. Did you find a solution?
thx

zaxxon72, 2024-11-30:
Hi,
Yes, nodon support tried to help, but did not understand it fully. But while I tried to do some A/B tests with different modules in different switches, I found out that after a delete/rejoin, they would be cured of the condition. So problem solved after opening a few outlets.... since then no problems, even after power cycles of the modules. 

It seems to be pure software, and these modules could perfectly work in decoupled mode. If Nodon would get their act together....

egretzbergerj, 2024-12-01:
Tank‘s a lot – deleted and joined them again, afterwards little work on some automations and everything is fine again.

closed 2025-02-07 by github-actions[bot]