Koenkk/zigbee2mqtt #22287: GL-C-006P unresponsive after Z2M OTA

marce14-git, 2024-04-22:
### What happened?

Updated GL-C-006P(Hw-rev:2) to Version: 10651203 Build Date:20220627 via OTA 3days ago.

Both Controllers can be paired and removed from the network and are online (last seen ok) but they are completly unresponsive to every other command, no state change no ota request....

i tried to revert the Firmware but i cant --> OTA request result in a timeout

### Zigbee2MQTT version

 1.36.1 commit: ffc2ff1

### Adapter firmware version

0x26720700

### Adapter

Conbee2

### Setup

Docker

marce14-git, 2024-04-22:
`Failed to configure 'Bad LED', attempt 3 (Error: ZCL command 0xa4c1384e49a34ef2/1 genBasic.read(["modelId"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (109)) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23) at runNextTicks (node:internal/process/task_queues:60:5) at processTimers (node:internal/timers:509:9) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:404:24) at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/gledopto.ts:155:27) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:162:21) at Configure.configure (/app/lib/extension/configure.ts:122:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))`

marce14-git, 2024-04-22:
> Try this:
> 
> 1. Delete the zigbee2mqtt device, force its deletion
> 2. Reset the device (turn off 5 times waiting 2 seconds)
> 3. Restart zigbee2mqtt
> 4. Pair again

Wow thanks i am a step closer... the device responds again to all state changes and OTA request !

BUT the light strip stays OFF although the device state is on... but maybe i can revert the firmware now ;)

mrmaximas, 2024-04-22:
I have issue with 009P ;) Now all of my gledopto dimmers works fine. 
https://github.com/Koenkk/zigbee-OTA/issues/463

marce14-git, 2024-04-22:
Okay after repairing the device has ANOTHER Firmware Version: 20451203 Date: 20240227 --> this is the version of the MINI Version but i have the NORMAL GL-C-006P Version...

marce14-git, 2024-04-22:
i cannot get the device to update to the right firmware:

    {
        "fileVersion": 13,
        "fileSize": 267446,
        "manufacturerCode": 4687,
        "imageType": 0,
        "sha512": "28240ef0eb6b7beea8797b4d3e2acd02e3fb8967083f47a7a3597aa48306c66dcc8ad47a3a8a91189bb556445232d19f03ca3f77777421b251cf36dcf1b6b28e",
        "modelId": "GL-C-006P",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Gledopto/GL-C-006P_10576802_20240412.ota",
        "path": "images/Gledopto/GL-C-006P_10576802_20240412.ota",
        "force": true
    }

the device says no new firmware.... if i change the image type to 5138(like the mini firmware) it says update available but fails bc of imagetype missmatch...

l0g1c5, 2024-04-22:
I have the same issue after latest OTA update.. I can pair it but it does not react on any input over MQTT.

GoaMind, 2024-04-23:
Same here.
After reset and reconnect, the only attributes available are:
```json
{
    "color_mode": "xy",
    "linkquality": 255
}
```

With a few more resets, I am able to get:
```json
{
    "color_mode": "xy",
    "linkquality": 255,
    "state": null,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null
}
```

But after a few seconds it goes to the first state

GoaMind, 2024-04-26:
Same here.
After reset and reconnect, the only attributes available are:
```json
{
    "color_mode": "xy",
    "linkquality": 255
}
```

With a few more resets, I am able to get:
```json
{
    "color_mode": "xy",
    "linkquality": 255,
    "state": null,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null
}
```

But after a few seconds it goes to the first state.

After a day I see, this state:
```json
{
    "color_mode": "xy",
    "linkquality": 255,
    "update": {
        "installed_version": 604057601,
        "latest_version": 604057601,
        "state": "idle"
    },
    "update_available": false
}
```

However, if I try to check for OTA updates, it replies
```
Failed to check if update available for 'XXX' (OTA: Device didn't respond to OTA request)
```

marce14-git, 2024-04-27:
i was able to resolve the issue:

I emailed gledopto with my problem and they provided me with a new firmware which i OTA to the device and it works again!

1. Just add to your zigbee2mqtt config:
```
ota:
    zigbee_ota_override_index_location: my_index.json
```

2. create a new my_index.json in the root z2m folder and add:
```
[
    {
        "url": "Name_of_new_gledopto_FW.ota",
        "force": true
    }
]
```
3. Update over OTA

Link to the new Gledopto FW: https://drive.google.com/file/d/1Bn-vlj8rArY0YQu6Izj0kHLz4-t6uqRB/view?usp=drive_link

GoaMind, 2024-05-01:
Hey @marce14-git ,

Thank you for the detailed instruction. Configured my Zigbee2mqtt same way.

 After reset, force removal and then joining, got these messages:
```2024-05-01 20:45:01 Failed to configure '0xa4c138791xxx', attempt 5 (Error: ZCL command 0xa4c138791xxx/1 genBasic.read(["modelId"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (151)) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:404:24) at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/gledopto.ts:155:27) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:162:21) at Configure.configure (/app/lib/extension/configure.ts:122:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))```
```2024-05-01 20:44:42 MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0xa4c138791xxx"},"error":"Failed to configure (ZCL command 0xa4c138791xxx/1 genBasic.read([\"modelId\"], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (151)))","status":"error","transaction":"z91y2-1"}'```

Tried several times with button reset and with power cycle reset, same result all the time

I see firmware data available on `About` page:
[image]

But it is not available on `State` page
[image]
```json
{
    "color_mode": "xy",
    "linkquality": 255
}
```

And OTA shows this state:
[image]

Clicking on `Check for new updates` give me:
```2024-05-01 21:07:37 Failed to check if update available for '0xa4c138791xxx' (OTA: Device didn't respond to OTA request)```
```2024-05-01 21:07:37 Error: OTA: Device didn't respond to OTA request at requestOTA (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:312:15) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:390:25) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:209:45) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)```

And after restarting zigbee2mqtt, and waiting for 3-5 minutes state became as I mentioned before:
```json
{
    "color_mode": "xy",
    "linkquality": 255,
    "update": {
        "installed_version": 604057601,
        "latest_version": 620834817,
        "state": "available"
    },
    "update_available": true
}
```
Then upgrade is available as well:
[image]

But clicking on `Update device to lates` give you same error:
```2024-05-01 21:27:13 Update of '0xa4c138791xxx' failed (OTA: Device didn't respond to OTA request)```
```2024-05-01 21:27:13 Error: OTA: Device didn't respond to OTA request at requestOTA (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:312:15) at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:436:21) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:273:41) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)```

Tried several times. But no luck =(

Does anyone have any ideas what else I can try to rebirth the device? 🙏

marce14-git, 2024-05-01:
Hey @GoaMind,

luckily my device was responsive again after:
1. Force Remove
2. Restart Z2M
3. Pair Device

if this does not work the only thing that comes to mind is repeatedly pushing the button (like once every second) on the device while pairing and maybe even pushing the button while requesting the OTA update - if this does not work maybe start from step 1. again until it works... 
i cannot think of something else

good luck!

GoaMind, 2024-05-02:
Just to let you know, I was able to update the firmware after several retries by following steps you have proposed.

[image]

Update took approximately 17 minutes.

Right after update params like:
- brightness
- color_mode
- color_temp
- color_temp_startup
- state

were not available, but after clicking on `Refresh` button in front of each parameter in `Expose` tab everything has appeared.

all good now:
```json
{
    "brightness": 15,
    "color_mode": "color_temp",
    "color_temp": 495,
    "color_temp_startup": 65535,
    "linkquality": 255,
    "state": "OFF",
    "update": {
        "installed_version": 604057601,
        "latest_version": 604057601,
        "state": "idle"
    },
    "update_available": false
}
```

Thank you so much for your time and support!

l0g1c5, 2024-05-04:
> i was able to resolve the issue:
> 
> I emailed gledopto with my problem and they provided me with a new firmware which i OTA to the device and it works again!
> 
> 1. Just add to your zigbee2mqtt config:
> 
> ```
> ota:
>     zigbee_ota_override_index_location: my_index.json
> ```
> 
>  2. create a new my_index.json in the root z2m folder and add:
> 
> ```
> [
>     {
>         "url": "Name_of_new_gledopto_FW.ota",
>         "force": true
>     }
> ]
> ```
> 
>  3. Save the new Firmware (GL-C-006P[NORMAL Version]) in the root folder https://drive.google.com/file/d/1Bn-vlj8rArY0YQu6Izj0kHLz4-t6uqRB/view?usp=drive_link
> 4. Update over OTA

This solved the issue for me too! Thank you very much!!

robertoclg, 2024-06-22:
Hello, yesterday I updated my two GL-C-006P(Hw-rev:2) mini version Zigbee LED Controller WW/CW (pro) from installed_version":604057601,"latest_version":620834817
The update was correctly installed on both devices, but since then they do not respond to status changes (on/off), color change or brightness intensity change.
Now they have this version installed_version":620834817,"latest_version":620834817
Zigbee2mqtt versión: [1.38.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.38.0) commit: [fe048e6]
Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20230507,"transportrev":2},"type":"zStack3x0"}'
Both devices were working correctly until the last update.
I have Try this but not 

1. Delete the zigbee2mqtt device, force its deletion
2. Reset the device (turn off 5 times waiting 2 seconds). In my case the devices has reset button (I press 5 seconds an try pairing again)
3. Restart zigbee2mqtt
4. Pair again
This is the actual state:
{
    "color_mode": "xy",
    "linkquality": 87,
    "state": "ON",
    "update": {
        "installed_version": 604057601,
        "latest_version": 604057601,
        "state": "idle"
    },
    "update_available": null
}
What can I do to solve the problem?
Thankx in advance

webns, 2024-06-22:
Hey @robertoclg 

I have the same problem with the mini controller, I guess we need a fresh version of firmware for the mini version but don't know where to find it. I've created a dedicated issue for the problem — #23128

robertoclg, 2024-06-22:
> Hey @robertoclg
> 
> I have the same problem with the mini controller, I guess we need a fresh version of firmware for the mini version but don't know where to find it. I've created a dedicated issue for the problem — #23128

thank you very much ;-) I hope a solution comes soon

CoolID2, 2024-06-22:
Same problem with two "normal" controller.
Forcing firmware downgrade won't work.

`Failed to check if update available for 'XXXXXXXXX' (OTA: Device didn't respond to OTA request)`

kebabas911, 2024-06-25:
Hey guys,

I've received ota file from gledopto regarding mini version of GL-C-006P WW/CW (for CCT lights only).
[image]

I could not manage to downgrade the file locally, as i get 'no image found error'. Anyway, maybe this file is useful for someone
https://drive.google.com/file/d/1eMZJbDIJPxhWYHjbutycgwbDr_m_jlGj/view?usp=drive_link

webns, 2024-06-25:
@kebabas911 check solution in this topic https://github.com/Koenkk/zigbee2mqtt/issues/23128 by @Andrik45719

They sent that firmware to everyone I see, I've received it yesterday