Koenkk/zigbee2mqtt #22291: [New device support]:  innr smart string light - 8 meter OLS210

danielrudolf, 2024-04-22:
### Link

https://www.innr.com/de/produkt/innr-smart-outdoor-light-string/

### Database entry

{"id":53,"type":"Router","ieeeAddr":"0xa4c138165499405a","nwkAddr":1569,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"OLS 210","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[10,25],"clusters":{"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xdc8e95fffe08bc00","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":21,"stackVersion":54,"hwVersion":1,"dateCode":"20231130","swBuildId":"1.5.12","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1713813328336}

### Comments

Please help me to add this device. Thanks

### External definition

```shell
const {identify, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['OLS 210'],
    model: 'OLS 210',
    vendor: 'innr',
    description: 'Automatically generated definition',
    extend: [identify(), onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-04-23:
Does it work with the external definition?

danielrudolf, 2024-04-26:
I´m a happy User but i have no idea what i have to do. I don't understand the description of what I'm supposed to do.

Koenkk, 2024-04-28:
If I understand correctly, this device only support on/off and not changing the brightness?

danielrudolf, 2024-04-28:
https://www.innr.com/de/produkt/innr-smart-outdoor-light-string/

At the moment, zigbee2mqtt only recognizes off and on. But I think it should also be possible to dim. There is a button on the cable itself with which you can switch the dimming levels. But I didn't manage to find anything in the debug mode and the description how to add a device. 
In principle, I think it is very similar in function to this Innr OGL 130 lamp. This is already supported. https://www.zigbee2mqtt.io/devices/OGL_130_C.html

Koenkk, 2024-04-28:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-04-28 by Koenkk