Koenkk/zigbee2mqtt #22294: Z2M not pulling in Light Level / Illuminance / LUX from Bosch RFDL-ZB-MS

Ltek, 2024-04-22:
### What happened?

The devices page says Z2M doesnt read LUX from the device so I'm assuming this is a bug on Z2m since all of my my RFDL-ZB-MS in ConbeeDeconz read this perfectly fine. See screenshot. 

https://www.zigbee2mqtt.io/devices/RADON_TriTech_ZB.html

I moved 5  of these sensors from Conbee/Deconz to Z2M and Z2M is not reading Light Level / Illuminance / Lux but has the placeholders for them...

[image]

here's the debug log...

Debug 2024-04-22 17:53:15Received Zigbee message from 'Sensor Patio Bosch Motion', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":33}' from endpoint 1 with groupID 0
Info 2024-04-22 17:53:15MQTT publish: topic 'zigbee2mqtt/Sensor Patio Bosch Motion', payload '{"battery":100,"battery_low":false,"linkquality":200,"occupancy":true,"tamper":false,"temperature":22.49,"voltage":3000}'
Debug 2024-04-22 17:53:19Received Zigbee message from 'Sensor Patio Bosch Motion', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
Info 2024-04-22 17:53:19MQTT publish: topic 'zigbee2mqtt/Sensor Patio Bosch Motion', payload '{"battery":100,"battery_low":false,"linkquality":204,"occupancy":false,"tamper":false,"temperature":22.49,"voltage":3000}'

looks like Z2M is not publishing the a complete definition of what the payload should be. 

### What did you expect to happen?

I expect to see the light level like I can when using Deconz/Conbee...

[image]

### How to reproduce it (minimal and precise)

pair Z2M to a Bosch RFDL-ZB-MS 

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E (ezsp v8)

### Setup

Homeseer on Windows

Ltek, 2024-04-23:
Z2M definition has Illuminance but Z2M doesnt seem to be processing it correctly, or at all...

[image]

Here's the info from Deconz...

[image]

Ltek, 2024-04-27:
FOUND the BUG....

During joining the network Z2M is not Binding  Illuminance. I fumbled and looked around, since troubleshooting is not documented anywhere I could find, I thought it was odd this was not checked but other parameters/attributes were.

FIX: 

1. pull the battery
2. check the box for Illuminance
3. put battery back in
4. click "Bind"

You will see several confirmation messages show on the screen... red if they fail, blue if they success

[image]

Koenkk, 2024-04-28:
Thanks @Ltek, integrated the fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-04-28 by Koenkk

Ltek, 2024-04-28:
@Koenkk related bug.. https://github.com/Koenkk/zigbee2mqtt/issues/22336