Koenkk/zigbee2mqtt #22297: [New device support]: Develco alarm siren SIRZB-111

MarleneSallat, 2024-04-23:
### Link

https://frient.com/products/smart-siren/

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x0015bc00410059ab","nwkAddr":7687,"manufId":4117,"manufName":"frient A/S","powerSource":"Mains (single phase)","modelId":"SIRZB-111","epList":[1,43,242],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[5,6],"outClusterList":[],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x048727fffe1c9730","endpointID":1}],"configuredReportings":[],"meta":{}},"43":{"profId":260,"epId":43,"devId":1027,"inClusterList":[0,3,1,4,1280,1282],"outClusterList":[3,10,25],"clusters":{"genBasic":{"attributes":{"modelId":"SIRZB-111","manufacturerName":"frient A/S","powerSource":129,"zclVersion":7,"dateCode":"2023-09-05 20:59"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x048727fffe1c9730","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"dateCode":"2023-09-05 20:59","zclVersion":7,"interviewCompleted":true,"meta":{},"lastSeen":1713865199401}

### Comments

i followed the instruction but cannot get to work this siren

### External definition

```shell
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
     {
        zigbeeModel: ['SIRZB-111'],
        model: 'SIRZB-111',
        vendor: 'Develco',
        description: 'Customizable siren',
        fromZigbee: [develco.fz.temperature, fz.battery, fz.ias_enroll, fz.ias_wd, fz.ias_siren],
        toZigbee: [tz.warning, tz.warning_simple, tz.ias_max_duration, tz.squawk],
        meta: {battery: {voltageToPercentage: '3V_2500'}},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(43);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'ssIasZone', 'ssIasWd', 'genBasic']);
            await reporting.batteryVoltage(endpoint);
            await endpoint.read('ssIasZone', ['iasCieAddr', 'zoneState', 'zoneId']);
            await endpoint.read('ssIasWd', ['maxDuration']);

            const endpoint2 = device.getEndpoint(1);
            await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);

            await develco.configure.read_sw_hw_version(device);
        },
        endpoint: (device) => {
            return {default: 43};
        },
        exposes: [e.battery(), e.battery_low(), e.test(), e.warning(), e.squawk(),
            e.numeric('max_duration', ea.ALL).withUnit('s').withValueMin(0).withValueMax(900)
                .withDescription('Max duration of the siren'),
            e.binary('alarm', ea.SET, 'START', 'OFF').withDescription('Manual start of the siren')],
    },

module.exports = definition;
```

wachtell, 2024-05-23:
I concur and hope that someone have the time to look at this.

wachtell, 2024-05-23:
I also see that there many others who have tried to build the external converters without any luck:
https://community.simon42.com/t/zigbee2mqtt-develco-frient-alarmsirene-sirzb-111/23759
https://github.com/Koenkk/zigbee2mqtt/discussions/21069

ruant, 2024-12-03:
Support is added in #22999 
This issue should be closed @Koenkk

closed 2024-12-03 by Koenkk