Koenkk/zigbee2mqtt #22298: Lidl 14156408L reporting wrong brightness

foltymat, 2024-04-23:
### What happened?

The device [Lidl 14156408L](https://www.zigbee2mqtt.io/devices/14156408L.html) is reporting a wrong brightness value at least since ver 1.35.

After changing the brightness, it changes to a value beyond 254. Typically in the neighborhood of 800-1000.

This results is the light is flashing back to the maximum value and is unable to report the correct value.

No other light in my home does this. I restored the backup of ver 1.33.2 and the issue is non-existent.

### What did you expect to happen?

Theight should be reporting the correct value in the 0-254 range.

### How to reproduce it (minimal and precise)

Not sure as this seems to be model specific issue.

Turn on the light.
Change brightness.
Change brightness rapidly.

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

20220219

### Adapter

Sonoff ZigBee 3.0 USB dongle

### Setup

Home Assistant OS addon. Odroid N2+

Koenkk, 2024-04-24:
Will be fixed with the next z2m release!

closed 2024-04-24 by Koenkk

dankarization, 2024-05-01:
Same for [Miboxer LC2-ZR](https://miboxer.com/product/2-wires-2-in-1-led-controller-zigbee-3-0-2-4g) working under [TuYa TS0502B](https://www.zigbee2mqtt.io/devices/TS0502B.html#tuya-ts0502b) converter. After turn on or any brightness change i get a lot of mqtt messages with brightness value 1000