Koenkk/zigbee2mqtt #22299: Aqara ZNJLBL01LM error after latest dev update

mamrai1, 2024-04-23:
### What happened?

[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15082261/log.log)
Error with Aqara blind driver

### What did you expect to happen?

Device works OK.

### Zigbee2MQTT version

1.36.1-dev commit: e5d632b

### Adapter firmware version

7.4.2

### Adapter

Ezsp or ember

### Setup

Plain z2m in docker

### Debug log

Tried both ezsp and ember. Same error in logs

galant-ho, 2024-04-24:
I just installed one more blind and successfully updated the fw to 09-12-2023.
So far no problem running it.

mamrai1, 2024-04-24:
Are you sure is the same device? My firmware is 10-09-2022

mamrai1, 2024-04-24:
It's very weird, I have 2 blinds and none of them shows any firmware update

galant-ho, 2024-04-24:
Where did you order yours?
Mine directly purchased in China has manufactured date as 2023 Oct.

mamrai1, 2024-04-24:
It's the lumi.curtain.acn002

galant-ho, 2024-04-24:
same here too
lumi.curtain.acn002

mamrai1, 2024-04-24:
OK. So you updated the device with the Aqara hub?  Because z2m firmware is older

galant-ho, 2024-04-24:
I was able to use OTA update it on Z2M.

fir3drag0n, 2024-04-24:
I also have only 10-09-2022 available, no newer updates available with zigbee2mqtt / docker.

fir3drag0n, 2024-04-28:
I can confirm the problem, even with ember and stable branch

fir3drag0n, 2024-04-29:
`2024-04-29 08:34:47Publish 'set' 'position' to 'Rollo-Schlafzimmer' failed: 'Error: ZCL command 0x54ef4410004d26f3/1 genAnalogOutput.write({"presentValue":83}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":13,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":42})'
Error 2024-04-29 08:35:02Publish 'set' 'position' to 'Rollo-Schlafzimmer' failed: 'Error: ZCL command 0x54ef4410004d26f3/1 genAnalogOutput.write({"presentValue":52}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":13,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":59})'
Error 2024-04-29 08:35:18Publish 'set' 'position' to 'Rollo-Schlafzimmer' failed: 'Error: ZCL command 0x54ef4410004d26f3/1 genAnalogOutput.write({"presentValue":73}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":13,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":78})'`

10Thirty, 2024-05-02:
> I also have only 10-09-2022 available, no newer updates available with zigbee2mqtt / docker. Device lumi.curtain.acn002. I have 4 devices and no one has a newer update
> 
> I also have the same problem after dev update, blinds cannot be controlled anymore by z2m

I also only have 10-09-2022 available for my lumi.curtain.acn002. Tried connecting it to an aqara hub as well and it shows that there is no further updates available so looks like they might be different revisions or something. My motor was bought in Jul 2022 from Aqara Official UK on amazon.co.uk.

closed 2024-05-04 by mamrai1