Koenkk/zigbee2mqtt #22308: general state of power readouts from various plugs is generally broken.

tomtom13, 2024-04-24:
### What happened?

So, I've been trying to find a good power plug (UK) that can report power reliably and hopefully updating rapidly. 
1. I've started with Samsung ones, however those stoped reporting after week to few weeks, so I decided to ditch them
2. I moved to Frient ones (SPLZB-134) and those work like a charm ... to some extent. Those are reliable, and never drop out however power readout jump to lot higher values from time to time (if reporting every 2 seconds is enabled it can happen as often aas every 10 second power reported jumping from 200W to 800W ... sometimes even to 14kW) (screenshot below hopefully if I find some nice graphs) 
3. hoping I can improve my statistics I moved to Tuya plugs (BSD29_1) but those keep completely disconnecting from the zigbee network. I've noticed that it I change any setting from "setting specific" tab the disconnection happens within day or few, without any changes it tends to last longer (one plug still is connected since fresh). What's worse is that if I decide to delete setting, like reporting interval, GUI just leaves the value blank ( as reporting_interval: "" in configuration.yaml) and that causes Z2M to coming back with bunch of problems and other devices on the network become very sluggish

So in total, there are quite few issues.
- I've got quite few devices 
- I've got quite big network that can stress test the setup
- I've got some skills with software
how can I help fixing this issue ? 

### Zigbee2MQTT version

asdf

### Adapter firmware version

asdf

### Adapter

asdf

### Setup

asdf

### Debug log

asdf

Outsidewall, 2024-04-25:
@tomtom13 Hello what adapter are you using, and which firmware is it using?

tomtom13, 2024-04-25:
Zigbee2MQTT version
1.36.1 commit: unknown
Coordinator type
zStack3x0
Coordinator revision
20210708
Coordinator IEEE Address
0x00124b0024c0ba36
Frontend version
0.6.161
zigbee_herdsman_converters_version
19.11.2
zigbee_herdsman_version
0.40.3

unit is Sonoff ZBDongle-P

Do you need any more information ?

Outsidewall, 2024-04-26:
@tomtom13 I did wonder if you were using a dongle-e, as I have had many issues with it, fixed now with firmware 7.4.2

tomtom13, 2024-06-13:
So I can confirm that:
- after updating to newest vestion of firmware for Sonoff ZBDongle-P I've got exactly the same issues.
- I've changed the coordinator to SLZB-06 and it's running 20240315 with exactly the same problems.
And the BSD29_1 still disconnects when update interval is set sub 2 seconds. 
As I can see, there is not much traction on those issues - alto the BSD29_1 did improve in a manner that settings are more likely to be applied.

closed 2025-01-06 by github-actions[bot]