Koenkk/zigbee2mqtt #22309: Temporarily "disable" devices still pinged

DenDeXTeR51, 2024-04-25:
### What happened?

I disable some device for some period of time and then enable it when I need to use it. Z2M still try to ping them, when devices already marked as "disabled" in UI.

### What did you expect to happen?

When device marked as "disabled" - turn off ping.

### How to reproduce it (minimal and precise)

Add some Lights (like Gledopto Lamp) use it for some time, then disable device, power if off (from 220v) and Z2M will still try to ping them till the time you power on device and "enable" it again.

### Zigbee2MQTT version

1.36.1 commit: ffc2ff1

### Adapter firmware version

20230507

### Adapter

electrolama zig-a-zig-ah

### Setup

Z2M in docker on raspberrypi

### Debug log

Failed to ping 'TreeCherry' (attempt 1/1, ZCL command 0xa4c138aae4e94ba4/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 47806 - 11 - 18 - 0 - 1 after 10000ms))

Active device 'TreeCherry' was last seen '834.05' minutes ago.

[image]

andreasbrett, 2024-05-01:
Agreed, this is counter intuitive and unnecessarily congests the zigbee network with pings. Also it's really annoying as the "could not ping" dialog messages flood up z2m_frontend.

Koenkk, 2024-05-02:
Did you restart z2m after disabling the device?

andreasbrett, 2024-05-03:
> Did you restart z2m after disabling the device?

I did not know I had to restart after disabling a device. Just know did I notice the restart required endpoint published via MQTT. I raised a feature request for the frontend to show this restart requirement. I think it is helpful to see this right in the frontend rather than some time later in a different UI like Home Assistant.

https://github.com/nurikk/zigbee2mqtt-frontend/issues/2014

DenDeXTeR51, 2024-05-05:
I disable device via mqtt from openhab script and as far as I see I can restart z2m only through UI... Is there mqtt command to restart z2m?

closed 2024-12-02 by github-actions[bot]