Koenkk/zigbee2mqtt #22310: TuYa TH02Z does not delivery sensor data

Tockra, 2024-04-25:
### What happened?

I paired a sensor which was detected as TH02Z: https://www.zigbee2mqtt.io/devices/TH02Z.html#tuya-th02z 
I'm not 100% sure if it is really the sensor, but the image seems to fit.

After I pair the sensor the `Details` page shows 0 °C und 0% Humidity:
[image]

I had similar problems with 3 other Tuya sensors, but this is the only one where I don't get data even after some new pairings with the HA.

The sensor is laying next to my raspberry pi where I've installed home assistant. There I installed zigbee2mqtt through the addons store (foreign repo). 

I don't know where to check logs for a further inspection. 

its related to sensor: 0xa4c138e1b60c52a5

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus,Zigbee Gateway TI CC2652P + CP2102N Zigbee USB Stick,Zigbee Hub für ZHA in Home Assistant or Zigbee2MQTT,Open HAB etc.

### Setup

Add-on on Home Assistant OS on Raspberry Pi 5

### Debug log

[.....]
'{"from":"0xa4c1383f3042e037","homeassistant_rename":true,"to":"Bad - Temperatursensor","transaction":"vyk1k-2"}'
info  2024-04-25 13:54:51: MQTT publish: topic 'zigbee2mqtt/0xa4c1383f3042e037', payload ''
debug 2024-04-25 13:54:51: Refreshing Home Assistant discovery topic for 'Bad - Temperatursensor'
info  2024-04-25 13:54:51: MQTT publish: topic 'homeassistant/sensor/0xa4c1383f3042e037/battery/config', payload 'null'
info  2024-04-25 13:54:51: MQTT publish: topic 'homeassistant/sensor/0xa4c1383f3042e037/temperature/config', payload 'null'
info  2024-04-25 13:54:51: MQTT publish: topic 'homeassistant/sensor/0xa4c1383f3042e037/humidity/config', payload 'null'
info  2024-04-25 13:54:51: MQTT publish: topic 'homeassistant/sensor/0xa4c1383f3042e037/voltage/config', payload 'null'
info  2024-04-25 13:54:51: MQTT publish: topic 'homeassistant/sensor/0xa4c1383f3042e037/linkquality/config', payload 'null'
info  2024-04-25 13:54:51: MQTT publish: topic 'zigbee2mqtt/Bad - Temperatursensor', payload '{"battery":100,"humidity":46,"linkquality":72,"temperature":20.97,"voltage":3000}'
info  2024-04-25 13:54:51: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/rename', payload '{"data":{"from":"0xa4c1383f3042e037","homeassistant_rename":true,"to":"Bad - Temperatursensor"},"status":"ok","transaction":"vyk1k-2"}'
debug 2024-04-25 13:54:51: Received MQTT message on 'homeassistant/sensor/0xa4c1383f3042e037/battery/config' with data ''
debug 2024-04-25 13:54:51: Received MQTT message on 'homeassistant/sensor/0xa4c1383f3042e037/temperature/config' with data ''
debug 2024-04-25 13:54:51: Received MQTT message on 'homeassistant/sensor/0xa4c1383f3042e037/humidity/config' with data ''
debug 2024-04-25 13:54:51: Received MQTT message on 'homeassistant/sensor/0xa4c1383f3042e037/voltage/config' with data ''
debug 2024-04-25 13:54:51: Received MQTT message on 'homeassistant/sensor/0xa4c1383f3042e037/linkquality/config' with data ''
debug 2024-04-25 13:54:51: Received MQTT message on 'homeassistant/sensor/0xa4c1383f3042e037/voltage/config' with data ''
debug 2024-04-25 13:54:51: Received MQTT message on 'homeassistant/sensor/0xa4c1383f3042e037/linkquality/config' with data ''
info  2024-04-25 13:54:53: MQTT publish: topic 'homeassistant/sensor/0xa4c1383f3042e037/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c1383f3042e037"],"manufacturer":"TuYa","model":"Temperature and humidity sensor (TH02Z)","name":"Bad - Temperatursensor","via_device":"zigbee2mqtt_bridge_0x00124b002c3ab132"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","object_id":"bad_-_temperatursensor_battery","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Bad - Temperatursensor","unique_id":"0xa4c1383f3042e037_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info  2024-04-25 13:54:53: MQTT publish: topic 'homeassistant/sensor/0xa4c1383f3042e037/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c1383f3042e037"],"manufacturer":"TuYa","model":"Temperature and humidity sensor (TH02Z)","name":"Bad - Temperatursensor","via_device":"zigbee2mqtt_bridge_0x00124b002c3ab132"},"device_class":"temperature","enabled_by_default":true,"object_id":"bad_-_temperatursensor_temperature","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Bad - Temperatursensor","unique_id":"0xa4c1383f3042e037_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
info  2024-04-25 13:54:53: MQTT publish: topic 'homeassistant/sensor/0xa4c1383f3042e037/humidity/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c1383f3042e037"],"manufacturer":"TuYa","model":"Temperature and humidity sensor (TH02Z)","name":"Bad - Temperatursensor","via_device":"zigbee2mqtt_bridge_0x00124b002c3ab132"},"device_class":"humidity","enabled_by_default":true,"object_id":"bad_-_temperatursensor_humidity","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Bad - Temperatursensor","unique_id":"0xa4c1383f3042e037_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'
info  2024-04-25 13:54:53: MQTT publish: topic 'homeassistant/sensor/0xa4c1383f3042e037/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c1383f3042e037"],"manufacturer":"TuYa","model":"Temperature and humidity sensor (TH02Z)","name":"Bad - Temperatursensor","via_device":"zigbee2mqtt_bridge_0x00124b002c3ab132"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","object_id":"bad_-_temperatursensor_voltage","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Bad - Temperatursensor","unique_id":"0xa4c1383f3042e037_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}'
info  2024-04-25 13:54:53: MQTT publish: topic 'homeassistant/sensor/0xa4c1383f3042e037/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c1383f3042e037"],"manufacturer":"TuYa","model":"Temperature and humidity sensor (TH02Z)","name":"Bad - Temperatursensor","via_device":"zigbee2mqtt_bridge_0x00124b002c3ab132"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"bad_-_temperatursensor_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Bad - Temperatursensor","unique_id":"0xa4c1383f3042e037_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
debug 2024-04-25 13:54:53: Received MQTT message on 'homeassistant/sensor/0xa4c1383f3042e037/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c1383f3042e037"],"manufacturer":"TuYa","model":"Temperature and humidity sensor (TH02Z)","name":"Bad - Temperatursensor","via_device":"zigbee2mqtt_bridge_0x00124b002c3ab132"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","object_id":"bad_-_temperatursensor_battery","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Bad - Temperatursensor","unique_id":"0xa4c1383f3042e037_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
debug 2024-04-25 13:54:53: Received MQTT message on 'homeassistant/sensor/0xa4c1383f3042e037/temperature/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mq [... cut ...]
debug 2024-04-25 14:01:55: No converter available for 'ZG-227Z' with cluster 'msTemperatureMeasurement' and type 'attributeReport' and data '{"measuredValue":2020}'
debug 2024-04-25 14:01:55: Received Zigbee message from 'Growbox - Sensor', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":4250}' from endpoint 1 with groupID 0
debug 2024-04-25 14:01:55: No converter available for 'ZG-227Z' with cluster 'msRelativeHumidity' and type 'attributeReport' and data '{"measuredValue":4250}'

sdrev, 2024-08-10:
Hi, I have the same problem, have you been able to fix it?

Tockra, 2024-08-11:
I threw it away. Maybe the sensor was damaged or not supported...

closed 2025-01-06 by github-actions[bot]