Koenkk/zigbee2mqtt #22311: TuYa Motion sensor  ZY-M100-S_2

woodex1981, 2024-04-25:
### What happened?

Logs is filling up pretty fast with data, having 10 motion sensors it takes a few hours to eat up all RAM and slow down HA.

Debug 2024-04-25 14:06:46Received Zigbee message from 'Motion sensor / bathroom 1 st floor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":9}],"seq":6656}' from endpoint 1 with groupID 0
Info 2024-04-25 14:06:46MQTT publish: topic 'zigbee2mqtt/Motion sensor / bathroom 1 st floor', payload '{"detection_delay":0,"fading_time":120,"illuminance_lux":88,"linkquality":140,"maximum_range":5.55,"minimum_range":0,"presence":false,"radar_sensitivity":9,"target_distance":0}'
Debug 2024-04-25 14:06:48Received Zigbee message from 'Motion sensor / bathroom 1 st floor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":9}],"seq":6912}' from endpoint 1 with groupID 0
Info 2024-04-25 14:06:48MQTT publish: topic 'zigbee2mqtt/Motion sensor / bathroom 1 st floor', payload '{"detection_delay":0,"fading_time":120,"illuminance_lux":88,"linkquality":144,"maximum_range":5.55,"minimum_range":0,"presence":false,"radar_sensitivity":9,"target_distance":0}'

### What did you expect to happen?

Not to report to log debug info and not to flood logs

### How to reproduce it (minimal and precise)

Connect zigbee motion sensor, and check the log of zigbee2mqtt

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

20240404.2

### Adapter

Home assistant yellow

### Setup

Home assistant and zigbee2mqtt add on

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/15109695/log.txt)

elterko, 2024-06-14:
I second this, this model absolutely spams the entire network, can something be done on Z2M's side to avoid this ?

Koenkk, 2024-06-15:
Besides debounce, which reduces the rate at which MQTT messages are published (look at the z2m docs), z2m cannot reduce the amount of Zigbee messages sent by this device

closed 2024-06-15 by Koenkk