Koenkk/zigbee2mqtt #22315: [New device support]: Aqara Intelligent Bathroom Heater T1 [ZNYB01LM]

Kevin6872, 2024-04-25:
### Link

https://www.aqara.com/Thermostat_overview

### Database entry

{"id":42,"type":"Router","ieeeAddr":"0x54ef441000b4edf7","nwkAddr":10653,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.bhf_light.acn001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[514,513,5,4,3,0,768,8,6,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.bhf_light.acn001","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":29,"stackVersion":27,"hwVersion":1,"dateCode":"20230511"}},"manuSpecificLumi":{"attributes":{"223":{"type":"Buffer","data":[22,33,1,0,23,33,0,0,12,33,0,0,24,33,0,0]},"229":{"type":"Buffer","data":[16,15,1,7,240,37,73]},"238":329501,"247":{"type":"Buffer","data":[130,32,1,131,65,12,192,217,45,60,194,24,100,3,1,0,0,0]},"591":"0xffff099a0ffffffc","672":{"type":"Buffer","data":[192,217,45,60,194,24]},"1299":6500,"1300":2700}},"lightingColorCtrl":{"attributes":{"245":36281600,"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370,"colorTemperature":200}},"genOnOff":{"attributes":{"245":36281600,"onOff":0}},"genLevelCtrl":{"attributes":{"245":36281600,"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":29,"stackVersion":27,"hwVersion":1,"dateCode":"20230511","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1714057444725}

### Comments

Hello, I bought an Aqara Intelligent Bathroom Heater T1. The following picture shows all functions supported in Aqara Home. Currently, I use automatically generated external definitions, which can adjust the color temperature and brightness of the light normally. These links are related to AqaraGateway's addition of support for it, so hopefully they can be referenced. [
Add the support of Aqara YuBa T1](https://github.com/niceboygithub/AqaraGateway/commit/45b0515c79d8f6810063b1f0f86bfec19b3c1a15)
[Add support for bathroom heater T1 增加智能浴霸T1支持](https://github.com/niceboygithub/AqaraGateway/issues/218)
[image]
[image]

### External definition

```shell
const {identify, light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['lumi.bhf_light.acn001'],
    model: 'lumi.bhf_light.acn001',
    vendor: 'Aqara',
    description: 'Automatically generated definition',
    extend: [identify(), light({"colorTemp":{"range":[153,370]}})],
    meta: {},
};

module.exports = definition;
```

jejunpark, 2024-07-17:
please add device...!!

jejunpark, 2024-07-23:
@Koenkk can you add this device? in korea, Many people use this device and many people want to use this device in z2m.

Koenkk, 2024-07-24:
@jejunpark I first need a working converter before I can add it.

closed 2025-01-06 by github-actions[bot]