Koenkk/zigbee2mqtt #22319: Onesti Products AS, EasyAccess doorlock , battery report

stefancn, 2024-04-26:
### What happened?

After last update , the battery level was correctly changed from 200% to 100%, but it is stucked there, even if the doorlock warns of low battery....

### What did you expect to happen?

Show a decline in battery level as time goes by...

### How to reproduce it (minimal and precise)

Just look at the data each week

### Zigbee2MQTT version

1.36.1-1 

### Adapter firmware version

Dont know

### Adapter

Assistant SkyConnect

### Setup

Addon Home Assistant

stefancn, 2024-08-20:
Any news on the matter ....

bjornfor, 2024-12-29:
Bad bot.

bjornfor, 2025-03-01:
Bad bot! Blindly closing issues is a bad idea.

bjornfor, 2025-05-01:
Bad bot.

closed 2025-08-05 by github-actions[bot]

Alikzus, 2025-08-21:
I also have problem with a Nimly Touch Pro constantly reporting a battery level of 200% until the batteries are depleted.

The firmware version of my Connect module is 20220614.

The coordinator is a TI CC2652RB with zStack3x0.

I am currently running Z2M version 2.2.1, but this problem have been there for a year.