Koenkk/zigbee2mqtt #22321: [New device support]: GIRIER smart curtain switch 1CH

marcopaggioro, 2024-04-26:
### Link

https://it.aliexpress.com/item/1005005225077998.html?spm=a2g0o.order_list.order_list_main.10.6e143696EdOWAe&gatewayAdapt=glo2ita

### Database entry

{"id":13,"type":"Router","ieeeAddr":"0xa4c13872579760cf","nwkAddr":5683,"manufId":4417,"manufName":"_TZ3210_dwytrmda","powerSource":"Mains (single phase)","modelId":"TS130F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[4,5,258,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"closuresWindowCovering":{"attributes":{"32768":0,"tuyaMovingState":1,"currentPositionLiftPercentage":0,"moesCalibrationTime":100,"tuyaMotorReversal":0,"tuyaCalibration":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1714124493148}

### Comments

Hi,
I bought this Smart curtain switch module from Aliexpress. I tried to pair with Z2M but it is recognized as https://www.zigbee2mqtt.io/devices/TS130F.html#tuya-ts130f

The problem is that are parameters that has no sense like "Backlight mode" and there are parameters like "Calibration time" that are not editable.

### External definition

```shell
const {windowCovering} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS130F'],
    model: 'TS130F',
    vendor: '_TZ3210_dwytrmda',
    description: 'Automatically generated definition',
    extend: [windowCovering({"controls":["lift","tilt"]})],
    meta: {},
};

module.exports = definition;
```

marcopaggioro, 2024-04-26:
Maybe something like (inspired by https://github.com/Koenkk/zigbee2mqtt/discussions/8781)

```const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
	fingerprint: [
		{modelID: 'TS130F', manufacturerName: '_TZ3210_dwytrmda'}
	],
	model: 'Smart Curtain Switch',
	vendor: 'Girier',
	description: 'Curtain motor controller',
	fromZigbee: [fz.cover_position_tilt, fz.tuya_cover_options],
	toZigbee: [tz.cover_state, tz.moes_cover_calibration, tz.cover_position_tilt, tz.tuya_cover_calibration, tz.tuya_cover_reversal],
	meta: {coverInverted: true},
	exposes: [e.cover_position(), exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']),
	exposes.binary('calibration', ea.ALL, 'ON', 'OFF'), exposes.binary('motor_reversal', ea.ALL, 'ON', 'OFF'),
	exposes.numeric('calibration_time', ea.ALL).withValueMin(1).withValueMax(100)]
};

module.exports = definition;
```

tajnymag, 2024-05-09:
Just for a future reference, in Czechia, the same module can be bought under the name [CEL-TEC L120Z-C](https://www.cel-tec.cz/cel-tec-l120z-c-zigbee-ovladac-zaluzii-a-rolet-p3033). I've just now installed two modules and they both look the same and they share the same listed attributes.

dhalidi, 2024-07-24:
I had the same issue with the same module and to resolve the calibration time issue i just had to go into device settings in zigbee2mqtt.
1. Go to Dev console and select Closures in cluster then choose moescalibrationtime in attribute and put the calibration time manually in MoesCalibrationTime. If you enter 300 it will be 30 seconds.
There you go, you finally updated the calibration time manually

xanfia, 2024-11-03:
I confirm that @marcopaggioro's code is working perfectly for this device. Can it be merged into the main branch? It's still being sold in [Aliexpress ](https://www.aliexpress.com/item/1005005225077998.html?spm=a2g0o.order_list.order_list_main.16.5638194dpwFKSt&gatewayAdapt=glo2esp) being a solid and cheap option, once this patch is applied...

Koenkk, 2024-11-05:
Added!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `TS130F_GIRIER.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2024-11-05 by Koenkk

xanfia, 2024-11-05:
Done! Thank you so much for this project; I’m truly happy to have contributed, even if it’s just with this small detail.

xarly, 2024-11-11:
Hello, could you add support to "2 gang" version? Thank you!
https://it.aliexpress.com/item/1005005225077998.html?spm=a2g0o.order_list.order_list_main.10.6e143696EdOWAe&gatewayAdapt=glo2ita

max930710, 2024-12-20:
Hey, 
thank you for the general support here. 
Do you guys don’t have problems with interlock function when stopping the curtain manually half way? 
I can’t use these modules right now, because when I stop at some point and then want to close it to 0% the modules switch on for the full calibration time again. One module switches both outlets on, when the curtain is at 0% and the module is still on because of calibration time. The other one switches both outlets on when it’s moving down and is just half way down! So interlock isn’t working on both modules which I ordered for testing. 

I‘m scared they crash my motors. So I installed again the Sonoff, they just work perfectly. But unfortunately they are to big to install behind my switches. So I need to put them by the motor instead of the switches. These girier ones would be the perfect alternative, but for me they don’t work and may even grill my motors.

cheeseburgereddy1987, 2025-02-21:
Hey,

> Maybe something like (inspired by [#8781](https://github.com/Koenkk/zigbee2mqtt/discussions/8781))
> 
> ```
> // Add the lines below
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> 
> const definition = {
> 	fingerprint: [
> 		{modelID: 'TS130F', manufacturerName: '_TZ3210_dwytrmda'}
> 	],
> 	model: 'Smart Curtain Switch',
> 	vendor: 'Girier',
> 	description: 'Curtain motor controller',
> 	fromZigbee: [fz.cover_position_tilt, fz.tuya_cover_options],
> 	toZigbee: [tz.cover_state, tz.moes_cover_calibration, tz.cover_position_tilt, tz.tuya_cover_calibration, tz.tuya_cover_reversal],
> 	meta: {coverInverted: true},
> 	exposes: [e.cover_position(), exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']),
> 	exposes.binary('calibration', ea.ALL, 'ON', 'OFF'), exposes.binary('motor_reversal', ea.ALL, 'ON', 'OFF'),
> 	exposes.numeric('calibration_time', ea.ALL).withValueMin(1).withValueMax(100)]
> };
> 
> module.exports = definition;
> ```

Hello, I had exactly the same problem and was able to solve it with the code from xxx. 

For the correct image, simply enter the model 'TS130F_GIRIER' and the image will be loaded from the database

SlKlElT, 2025-05-06:
Hello,
does anyone know how to change the switch type for this device from toggle to momentary? Is there a way to add this option?