Koenkk/zigbee2mqtt #22322: Spam from devices

env-dxb, 2024-04-26:
### What happened?

Hi
I have an issue with devices reporting, once I send command to device, device reporting 3-6 times.
It's that why I have issue with LED strip controller, I send {"state":"ON"}, controller send me 5 states "ON" and (maybe because of this spam) 6th state returns me "OFF", so strip blinks and remains off

Other devices IKEA Tradfri etc also reporting 3-6 times after publishing request "0xdevice/set"

### What did you expect to happen?

Instead of I expecting "One request - One report"

### Zigbee2MQTT version

1.36.0 commit: 9c505fd7

### Adapter firmware version

7.3.1.0 build 176

### Adapter

ZBDongle-E

### Setup

Raspberry, ZBDongle-E, z2m, Homebridge, Node-Red

### Debug log

Send:
`{"state":"ON","color_temp":330,"brightness":254,"transition":2}`

Receive:
```
Info 2024-04-26 12:59:22MQTT publish: topic 'zigbee2mqtt/0xa4c138cc852c0166', payload '{"brightness":254,"color":{"hue":0},"color_mode":"color_temp","color_temp":330,"linkquality":188,"state":"ON"}'
Info 2024-04-26 12:59:22MQTT publish: topic 'zigbee2mqtt/0xa4c138cc852c0166', payload '{"brightness":254,"color":{"hue":0},"color_mode":"color_temp","color_temp":330,"linkquality":172,"state":"ON"}'
Info 2024-04-26 12:59:22MQTT publish: topic 'zigbee2mqtt/0xa4c138cc852c0166', payload '{"brightness":1,"color":{"hue":0},"color_mode":"color_temp","color_temp":330,"linkquality":172,"state":"ON"}'
Info 2024-04-26 12:59:23MQTT publish: topic 'zigbee2mqtt/0xa4c138cc852c0166', payload '{"brightness":1,"color":{"hue":0},"color_mode":"color_temp","color_temp":330,"linkquality":172,"state":"ON"}'
Info 2024-04-26 12:59:23MQTT publish: topic 'zigbee2mqtt/0xa4c138cc852c0166', payload '{"brightness":1,"color":{"hue":0},"color_mode":"color_temp","color_temp":330,"linkquality":172,"state":"ON"}'
Info 2024-04-26 12:59:23MQTT publish: topic 'zigbee2mqtt/0xa4c138cc852c0166', payload '{"brightness":1,"color":{"hue":0},"color_mode":"color_temp","color_temp":330,"linkquality":172,"state":"OFF"}'
```

env-dxb, 2024-04-30:
upd
other devices also flooding 
```
Info 2024-05-01 00:29:52MQTT publish: topic 'zigbee2mqtt/0x00158d000303d58e', payload '{"battery":100,"linkquality":64,"occupancy":true,"power_outage_count":161,"voltage":3075}'
Info 2024-05-01 00:29:52MQTT publish: topic 'zigbee2mqtt/0x00158d000303d58e', payload '{"battery":100,"linkquality":64,"occupancy":true,"power_outage_count":161,"voltage":3075}'
Info 2024-05-01 00:29:52MQTT publish: topic 'zigbee2mqtt/0x00158d000303d58e', payload '{"battery":100,"linkquality":80,"occupancy":true,"power_outage_count":161,"voltage":3075}'
```

LaurentChardin, 2024-05-11:
Do you have multiple reporting defined for the device ?

There is also a debounce configuration available at the device level that you could test: if the devices sent multiple messages on the zigbee network, the debounce can help publish less events to mqtt (not solving the zigbee part but can help cleaning a bit the mqtt part).

Usually, i have seen multiple messages when either many reporting were configured for similar attribute, or it could be faulty behavior from the device itself.

closed 2024-12-08 by github-actions[bot]