Koenkk/zigbee2mqtt #22327: Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'motDEtat'

RastaChaum, 2024-04-27:
### What happened?

Noted in the logs, this recurring message appears:

"Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'motDEtat'"

### What did you expect to happen?

This message should not be encountered, as it suggests the presence of a coding issue.

### How to reproduce it (minimal and precise)

No specials configurations. This is the default setup

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20230507

### Adapter

zStack3x0

### Setup

Docker installation on raspberry PI 2B

### Debug log

**Docker Z2M version**
REPOSITORY                               TAG          IMAGE ID       CREATED         SIZE
koenkk/zigbee2mqtt                       latest       60a241badfae   3 weeks ago     104MB

**Lixee details**
Constructeur	Modèle	        Date du firmware	Version du firmware	
LiXee               ZLinky_TIC	20230214	        4000-0013

RastaChaum, 2024-04-27:
The message in question occurs each time information is retrieved, accompanied by the following message:

```
zlinky config: historique, single_phase, false, H PLEINE/CREUSE
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'motDEtat'

```

ghostsam45, 2024-05-03:
same issue 

`2024-05-03 17:47:50 Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'motDEtat'`

Version de Zigbee2MQTT
[1.37.0] commit: [46f34c8]
Type de coordinateur
ConBee2/RaspBee2
Révision du coordinateur
0x26780700

Lixee ZLinky 20230214

ceduderino, 2024-05-03:
> same issue
>
> `2024-05-03 17:47:50 Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'motDEtat'`
>
> Version de Zigbee2MQTT [1.37.0] commit: [...]

same configuration, same issue

Koenkk, 2024-05-04:
@mildis Could you maybe check this ?

mildis, 2024-05-04:
Well, to my understanding, motDEtat is tied to Linky legacy mode.
If there’s a mismatch between Linky mode and how Lixee try to parse data, this might occurs.

@RastaChaum @ceduderino @ghostsam45 Are your Linky in standard mode or « historique » (legacy mode) ? Does forcing the Lixee parameter to the correct value instead of « auto » helps ?

ceduderino, 2024-05-04:
> Well, to my understanding, motDEtat is tied to Linky legacy mode. If there’s a mismatch between Linky mode and how Lixee try to parse data, this might occurs.
>
> @RastaChaum @ceduderino @ghostsam45 [...]

My Linky is in "historique" mode.
I have now forced the Lixee "Linky mode" parameter to "historique", but I get the same message.
`2024-05-04 12:49:11Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'motDEtat'`

mildis, 2024-05-04:
@ceduderino Do you have a line like "zlinky config: historique, single_phase, false, H PLEINE/CREUSE" ?

ceduderino, 2024-05-04:
> @ceduderino Do you have a line like "zlinky config: historique, single_phase, false, H PLEINE/CREUSE" ?

I have no such thing in the log (I am logging at info level).
But my configuration.yaml is as follows:
`    friendly_name: ZLinky_TIC
    measurement_poll_interval: ''
    kWh_precision: 3
    linky_mode: historique
    energy_phase: single_phase
    tarif: Historique - BASE
`

BAKSLASH971, 2024-05-04:
hello , i have the same problem 
[image]
[image]
[image]
[image]

mildis, 2024-05-04:
Well, I see nothing obvious in the code and as I'm using the Standard mode, I can't help.
Maybe @SilentT-FR who implemented this can help.
Also, have you tried to rollback to koenkk/zigbee2mqtt:1.36.1 to see if the problem is still here ? Or at least logged ?

RastaChaum, 2024-05-04:
Bonjour à tous et merci pour vos réponses que je viens uniquement de voir. 
J'ai vérifié ma configuration. Je suis effectivement en mode auto mais bien avec koenkk/zigbee2mqtt:1.36.1.
Avant de changer quoi que ce soit, j'ai de nouveau regardé mes traces. Et là, surprise ! Je ne vois plus ce message depuis le 28 avril fin de matinée !

Or, je ne me souviens pas d'avoir touché à quoi que ce soir ce jour. En revanche, j'ai effectivement coupé à plusieurs reprise mon compteur électrique en voulant bricoler sur mes radiateurs.... Peut être que cela a eu un effet sur le module Lixee ?? Ce n'est qu'une hypothèse très approximative.... Mais a @ceduderino  ou @ghostsam45 ca coute pas cher de couper votre compteur, voir débrancher/rebranche le module Lixee et de voir si le problème se règle aussi comme par magie... ! :)

ghostsam45, 2024-05-05:
> Well, I see nothing obvious in the code and as I'm using the Standard mode, I can't help. Maybe @SilentT-FR who implemented this can help. Also, have you tried to rollback to koenkk/zigbee2mqtt:1.36.1 to see if the problem is still here ? Or at least logged ?

looking in the logs of version 1.36.1, I did not have this error

BAKSLASH971, 2024-05-05:
No error log  with 1.36.1

remidebette, 2024-05-13:
Reading the latest release notes, is this issue fixed by https://github.com/Koenkk/zigbee2mqtt/issues/21825 ?

lecourtb, 2024-06-20:
Same issue with Z2M 1.38.0, "historique" mode, triphasé, measurement poll chunk = 1

closed 2025-01-06 by github-actions[bot]