Koenkk/zigbee2mqtt #22328: Tuya TS0026

Sam-Electrician, 2024-04-27:
### What happened?

The device shows supported however does not function in HA or Z2M 

### What did you expect to happen?

I expected the scene buttons to show up in HA to use in automations etc.

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20210708

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus_5

### Setup

Rpi5

### Debug log

2024-04-27 21:48:24Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 1 with groupID 0
Debug 2024-04-27 21:48:24No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
Debug 2024-04-27 21:48:24Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 1 with groupID 0
Debug 2024-04-27 21:48:24No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
Debug 2024-04-27 21:48:24Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 1 with groupID 0
Debug 2024-04-27 21:48:24No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
Debug 2024-04-27 21:48:24Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 1 with groupID 0
Debug 2024-04-27 21:48:24No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
Debug 2024-04-27 21:48:24Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 1 with groupID 0
Debug 2024-04-27 21:48:24No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
Debug 2024-04-27 21:48:24Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 1 with groupID 0
Debug 2024-04-27 21:48:24No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
Debug 2024-04-27 21:48:24Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 1 with groupID 0
Debug 2024-04-27 21:48:24No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
Debug 2024-04-27 21:48:24Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 1 with groupID 0
Debug 2024-04-27 21:48:24No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
Debug 2024-04-27 21:48:24Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 1 with groupID 0
Debug 2024-04-27 21:48:24No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
Debug 2024-04-27 21:48:24Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 1 with groupID 0
Debug 2024-04-27 21:48:24No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
Debug 2024-04-27 21:48:24Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 1 with groupID 0
Debug 2024-04-27 21:48:24No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
Debug 2024-04-27 21:48:24Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 1 with groupID 0
Debug 2024-04-27 21:48:24No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
Debug 2024-04-27 21:48:24Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 1 with groupID 0
Debug 2024-04-27 21:48:24No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
Debug 2024-04-27 21:48:24Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 1 with groupID 0
Debug 2024-04-27 21:48:24No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
Debug 2024-04-27 21:48:24Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandTuyaAction', cluster 'genOnOff', data '{"value":0}' from endpoint 1 with groupID 0
Debug 2024-04-27 21:48:24No converter available for 'TS0026' with cluster 'genOnOff' and type 'commandTuyaAction' and data '{"value":0}'
Debug 2024-04-27 21:49:22Received Zigbee message from 'TL1-LT1', type 'attributeReport', cluster 'genBasic', data '{"65506":48,"65508":0,"appVersion":101}' from endpoint 1 with groupID 0

Koenkk, 2024-04-28:
- How many buttons are there on your device?
- Could you provide the debug log when pressing each of them.
- 
See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

Sam-Electrician, 2024-04-28:
Hi there is 6 buttons on the device. 

Debug:

/2024-04-28 20:09:59Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 3 with groupID 0
2024-04-28 20:09:59No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:09:59Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 3 with groupID 0
2024-04-28 20:09:59No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:09:59Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 3 with groupID 0
2024-04-28 20:09:59No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:09:59Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 3 with groupID 0
2024-04-28 20:09:59No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:09:59Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 3 with groupID 0
2024-04-28 20:09:59No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:10:01Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 4 with groupID 0
2024-04-28 20:10:01No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:10:01Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 4 with groupID 0
2024-04-28 20:10:01No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:10:01Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 4 with groupID 0
2024-04-28 20:10:01No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:10:01Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 4 with groupID 0
2024-04-28 20:10:01No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:10:01Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 4 with groupID 0
2024-04-28 20:10:01No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:10:01Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 4 with groupID 0
2024-04-28 20:10:01No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:10:01Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 4 with groupID 0
2024-04-28 20:10:01No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:10:01Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 4 with groupID 0
2024-04-28 20:10:01No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:10:01Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 4 with groupID 0
2024-04-28 20:10:01No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:10:01Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 4 with groupID 0
2024-04-28 20:10:01No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:10:01Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 4 with groupID 0
2024-04-28 20:10:01No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:10:01Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 4 with groupID 0
[... cut ...]
2024-04-28 20:10:03Received Zigbee message from 'Switch 1 Formal Lounge', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":0}' from endpoint 6 with groupID 0
2024-04-28 20:10:03No converter available for 'TS0026' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":0}'
2024-04-28 20:10:16Received Zigbee message from 'TL1-LT1', type 'attributeReport', cluster 'genBasic', data '{"65506":48,"65508":0,"appVersion":101}' from endpoint 1 with groupID 0
[image]

Koenkk, 2024-04-28:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/aa96a254776bec8277db8f86b4655aa1): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Sam-Electrician, 2024-04-28:
Hi I have had this error

/const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/lib/fromZigbee');

const e = exposes.presets;
const ea = exposes.access;

module.exports = {
    zigbeeModel: ['TS0026'],
    model: 'TS0026',
    vendor: 'TuYa',
    description: '6 button scene wall switch',
    fromZigbee: [fz.command_recall],
    meta: {multiEndpoint: true},
    exposes: [e.action(['scene_1', 'scene_2', 'scene_3', 'scene_4', 'scene_5', 'scene_6'])],
    toZigbee: [],
};

Koenkk, 2024-04-29:
Can you provide the debug log when starting z2m, maybe the converter doesn't load.

Sam-Electrician, 2024-04-30:
/[21:48:17] INFO: Preparing to start...
[21:48:18] INFO: Socat not enabled
[21:48:18] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2024-04-30 21:48:20: Failed to load external converter file 'ext_converter.js' (Cannot find module 'zigbee-herdsman-converters/lib/fromZigbee'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js)
Zigbee2MQTT:error 2024-04-30 21:48:20: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
Zigbee2MQTT:error 2024-04-30 21:48:20: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device
Zigbee2MQTT:debug 2024-04-30 21:48:20: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2024-04-30 21:48:20: Logging to console and directory: '/config/zigbee2mqtt/log/2024-04-30.21-48-20' filename: log.txt
Zigbee2MQTT:debug 2024-04-30 21:48:20: Removing old log directory '/config/zigbee2mqtt/log/2024-04-26.15-01-36'
Zigbee2MQTT:info  2024-04-30 21:48:20: Starting Zigbee2MQTT version 1.36.1 (commit #unknown)
Zigbee2MQTT:info  2024-04-30 21:48:20: Starting zigbee-herdsman (0.40.3)
Zigbee2MQTT:debug 2024-04-30 21:48:20: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[17,206,252,165,84,216,196,68],"networkKey":"HIDDEN","panID":25350},"serialPort":{"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_5cc03959ada4ed118509f3a32981d5c7-if00-port0"}}'
Zigbee2MQTT:info  2024-04-30 21:48:21: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2024-04-30 21:48:21: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:debug 2024-04-30 21:48:21: Zigbee network parameters: {"channel":11,"extendedPanID":"0x11cefca554d8c444","panID":25350}
Zigbee2MQTT:info  2024-04-30 21:48:21: Currently 2 devices are joined:
Zigbee2MQTT:info  2024-04-30 21:48:21: Switch 1 Formal Lounge (0x385cfbfffe2cbf7a): TS0026 - TuYa 6 button scene wall switch (Router)
Zigbee2MQTT:info  2024-04-30 21:48:21: TL1-LT1 (0xf082c0fffeb1554c): TS0502B - TuYa Light controller (Router)
Zigbee2MQTT:info  2024-04-30 21:48:21: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2024-04-30 21:48:21: Connecting to MQTT server at mqtt://core-mosquitto:1883
Zigbee2MQTT:debug 2024-04-30 21:48:21: Using MQTT login with username: addons
Zigbee2MQTT:info  2024-04-30 21:48:21: Connected to MQTT server
Zigbee2MQTT:info  2024-04-30 21:48:21: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2024-04-30 21:48:21: Started frontend on port 8099
Zigbee2MQTT:info  2024-04-30 21:48:21: MQTT publish: topic

Koenkk, 2024-05-01:
Updated https://gist.github.com/Koenkk/aa96a254776bec8277db8f86b4655aa1, if it doesn't work, provide the debug logging agian.

Sam-Electrician, 2024-05-02:
Thats great, I really appreciate your help with this and its working now.

Thanks

Sam

Sam-Electrician, 2024-05-02:
> What are all the possible actions, recall_1, recall_2, etc. ?

Yes so I can recall 1-6 for the buttons in HA for automations

Koenkk, 2024-05-02:
Integrated the fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-02 by Koenkk