Koenkk/zigbee2mqtt #22329: Zigbee2MQTT crashes every few days

xcomputerman, 2024-04-28:
### What happened?

I've been getting periodic crashes. This happened when I was using ezsp and is still happening after switching to ember. I'm using a SLZB-06M coordinator over TCP. Log attached.

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

7.4.1 [GA]

### Adapter

SMLight SLZB-06M

### Setup

Home Assistant add-on, about 38 devices

### Debug log

[zigbeelog.txt](https://github.com/Koenkk/zigbee2mqtt/files/15140776/zigbeelog.txt)

xcomputerman, 2024-05-04:
Latest crash log.
[zigbeelog2.txt](https://github.com/Koenkk/zigbee2mqtt/files/15210744/zigbeelog2.txt)

Nerivec, 2024-05-16:
Can you confirm this still happens with latest core firmware from SMLight (2.2.x, with latest emberznet firmware too)?

Loic691, 2024-05-27:
I have the same behaviour after swtch from Sonoff E to SMLight SLZB-06M.
For me all is up to date with the last core and zigbee firmware (wich is known as 7.4.1 GA in Z2M)

xcomputerman, 2024-05-27:
I updated my SLZB-06M ESP firmware to the latest beta. I have had only one crash since then, but that is much better than before which was literally every few days. I think I have now gone at least two weeks without a crash. Keeping an eye on it.

Loic691, 2024-05-28:
> I updated my SLZB-06M ESP firmware to the latest beta. I have had only one crash since then, but that is much better than before which was literally every few days. I think I have now gone at least two weeks without a crash. Keeping an eye on it.
> […](#)

Thx. I will try.
I have too another problem wirh this dongle.
If I restart whole HA OS, the z2m don't stary correctly. I have after restart to stop Z2m add on. Cycle PoE and qo restart again s2m.

Do you have this same bahaviour ?

Loic691, 2024-05-28:
You talk about core firmware 2.2.8.dev ?

Loic691, 2024-05-28:
@Nerivec 
Can we use firmware_builds/slzb-06m on the dongle with OTA update from the core firmware GUI ?
What's is the best or official firmware  ? I'm confused with all this test and firmware...

Nerivec, 2024-05-28:
You should stay on the one recommended by smlight.
Can you elaborate on what you mean by "z2m doesn't start correctly"?

@darkxst Can you take a look at [this](https://github.com/Koenkk/zigbee2mqtt/issues/22329#issuecomment-2134266461) once we have some more details?

darkxst, 2024-05-29:
> If I restart whole HA OS, the z2m don't stary correctly. I have after restart to stop Z2m add on. Cycle PoE and qo restart again s2m.

If possible can you get logs from serial console attached to USB (/dev/ttyUSB0 or COM#) on SLZB-06M

xcomputerman, 2024-05-29:
New crash today. Log is attached!

[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7 frmNum=6] Added to rxQueue
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:ezsp: <=== [FRAME: ID=21:"SET_MANUFACTURER_CODE" Seq=14 Len=5]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:ezsp: ===> [FRAME: ID=83:"SET_CONFIGURATION_VALUE" Seq=15 Len=8]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=7 frmRx=7]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0 frmNum=7] Added to rxQueue
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:ezsp: <=== [FRAME: ID=83:"SET_CONFIGURATION_VALUE" Seq=15 Len=6]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember: [EzspConfigId] SET "STACK_PROFILE" TO "2" with status=SUCCESS.
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:ezsp: ===> [FRAME: ID=83:"SET_CONFIGURATION_VALUE" Seq=16 Len=8]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=0 frmRx=0]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=0] Added to rxQueue
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:ezsp: <=== [FRAME: ID=83:"SET_CONFIGURATION_VALUE" Seq=16 Len=6]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember: [EzspConfigId] SET "SECURITY_LEVEL" TO "5" with status=SUCCESS.
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:ezsp: ===> [FRAME: ID=82:"GET_CONFIGURATION_VALUE" Seq=17 Len=6]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=1 frmRx=1]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=1] Added to rxQueue
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:ezsp: <=== [FRAME: ID=82:"GET_CONFIGURATION_VALUE" Seq=17 Len=8]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:ezsp: ===> [FRAME: ID=83:"SET_CONFIGURATION_VALUE" Seq=18 Len=8]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=2 frmRx=2]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=2] Added to rxQueue
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:ezsp: <=== [FRAME: ID=83:"SET_CONFIGURATION_VALUE" Seq=18 Len=6]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember: [EzspConfigId] SET "BINDING_TABLE_SIZE" TO "32" with status=SUCCESS.
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:ezsp: ===> [FRAME: ID=83:"SET_CONFIGURATION_VALUE" Seq=19 Len=8]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=3 frmRx=3]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=3] Added to rxQueue
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:ezsp: <=== [FRAME: ID=83:"SET_CONFIGURATION_VALUE" Seq=19 Len=6]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember: [EzspConfigId] SET "KEY_TABLE_SIZE" TO "0" with status=SUCCESS.
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:ezsp: ===> [FRAME: ID=83:"SET_CONFIGURATION_VALUE" Seq=20 Len=8]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=4 frmRx=4]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=4] Added to rxQueue
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:ezsp: <=== [FRAME: ID=83:"SET_CONFIGURATION_VALUE" Seq=20 Len=6]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember: [EzspConfigId] SET "MAX_END_DEVICE_CHILDREN" TO "32" with status=SUCCESS.
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:ezsp: ===> [FRAME: ID=83:"SET_CONFIGURATION_VALUE" Seq=21 Len=8]
[2024-05-29 09:16:10] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=5 frmRx=5]
[... cut ...]
[2024-05-29 09:16:12] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3]
[2024-05-29 09:16:12] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=2] Added to rxQueue
[2024-05-29 09:16:12] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3]
[2024-05-29 09:16:13] [34mdebug[39m: 	zh:ember:ezsp: <=== [FRAME: ID=83:"SET_CONFIGURATION_VALUE" Seq=26 Len=6]
[2024-05-29 09:16:15] [34mdebug[39m: 	zh:ember: [EzspConfigId] SET "RETRY_QUEUE_SIZE" TO "16" with status=SUCCESS.

closed 2024-10-01 by Nerivec