Koenkk/zigbee2mqtt #22336: Z2M does not reset Tamper for Bosch RFDL-ZB-MS

Ltek, 2024-04-28:
### What happened?

Past 2 days I installed 5 Bosch RFDL-ZB-MS on Z2M, moving them from Deconz. 

Problem: Z2M does not sets Tamper alert back to FALSE after is been TRUE. I have let them sit over 24 hrs, still do not reset vs Deconz they reset immediately when tamper switch is correctly pressed (its in the wall mount sleeve).

Debug log shows its sending FALSE...

024-04-28 08:45:04MQTT publish: topic 'zigbee2mqtt/Sensor Theater Bosch Motion', payload '{"battery":100,"battery_low":false,"illuminance":9094,"illuminance_lux":8,"linkquality":136,"occupancy":false,"tamper":false,"temperature":19.21,"voltage":3000}'
Debug 2024-04-28 08:45:07Received Zigbee message from 'Sensor Garage Bosch Motion Door', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":10169}' from endpoint 1 with groupID 0
Info 2024-04-28 08:45:07MQTT publish: topic 'zigbee2mqtt/Sensor Garage Bosch Motion Door', payload '{"battery":100,"battery_low":false,"illuminance":10169,"illuminance_lux":10,"linkquality":188,"occupancy":false,"tamper":true,"temperature":17.54,"voltage":3000}'
Debug 2024-04-28 08:45:08Received Zigbee message from 'Sensor Stairs Aqara Motion bottom', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3025,"10":0,"100":0,"11":4,"3":20,"4":5032,"5":61,"6":[0,2]}}' from endpoint 1 with groupID 0
Debug 2024-04-28 08:45:08RTCGQ11LM: Processed data into payload {"voltage":3025,"battery":100,"device_temperature":20,"power_outage_count":60,"illuminance":4,"illuminance_lux":4}
Debug 2024-04-28 08:45:08RTCGQ11LM: Processed data into payload {"voltage":3025,"battery":100,"device_temperature":20,"power_outage_count":60,"illuminance":4,"illuminance_lux":4}
Info 2024-04-28 08:45:08MQTT publish: topic 'zigbee2mqtt/Sensor Stairs Aqara Motion bottom', payload '{"battery":100,"device_temperature":20,"illuminance":4,"illuminance_lux":4,"linkquality":196,"occupancy":false,"power_outage_count":60,"voltage":3025}'
Debug 2024-04-28 08:45:09Received Zigbee message from 'Sensor Office Bosch Motion', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":17571}' from endpoint 1 with groupID 0
Info 2024-04-28 08:45:09MQTT publish: topic 'zigbee2mqtt/Sensor Office Bosch Motion', payload '{"battery":80,"battery_low":false,"illuminance":17571,"illuminance_lux":57,"linkquality":104,"occupancy":false,"tamper":false,"temperature":20.81,"voltage":2900}'
Debug 2024-04-28 08:45:10Received Zigbee message from 'Sensor Living Room Bosch Motion LR Kitchen', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":17292}' from endpoint 1 with groupID 0
Info 2024-04-28 08:45:10MQTT publish: topic 'zigbee2mqtt/Sensor Living Room Bosch Motion LR Kitchen', payload '{"battery":100,"battery_low":false,"illuminance":17292,"illuminance_lux":54,"linkquality":236,"occupancy":false,"tamper":false,"temperature":19.75,"voltage":3000}'
Debug 2024-04-28 08:45:13Received Zigbee message from 'Sensor Mudroom Bosch Motion', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":7786}' from endpoint 1 with groupID 0
Info 2024-04-28 08:45:13MQTT publish: topic 'zigbee2mqtt/Sensor Mudroom Bosch Motion', payload '{"battery":80,"battery_low":false,"illuminance":7786,"illuminance_lux":6,"linkquality":172,"occupancy":false,"tamper":false,"temperature":20.32,"voltage":2900}'
Debug 2024-04-28 08:45:14Received Zigbee message from 'Sensor Theater Bosch Motion', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":8954}' from endpoint 1 with groupID 0
Info 2024-04-28 08:45:14MQTT publish: topic 'zigbee2mqtt/Sensor Theater Bosch Motion', payload '{"battery":100,"battery_low":false,"illuminance":8954,"illuminance_lux":8,"linkquality":224,"occupancy":false,**"tamper":false**,"temperature":19.21,"voltage":3000}'
Debug 2024-04-28 08:45:17Received Zigbee message from 'Sensor Garage Bosch Motion Door', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":10309}' from endpoint 1 with groupID 0
Info 2024-04-28 08:45:17MQTT publish: topic 'zigbee2mqtt/Sensor Garage Bosch Motion Door', payload '{"battery":100,"battery_low":false,"illuminance":10309,"illuminance_lux":11,"linkquality":212,"occupancy":false,"tamper":true,"temperature":17.54,"voltage":3000}'
Debug 2024-04-28 08:45:19Received Zigbee message from 'Sensor Office Bosch Motion', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":16294}' from endpoint 1 with groupID 0
Info 2024-04-28 08:45:19MQTT publish: topic 'zigbee2mqtt/Sensor Office Bosch Motion', payload '{"battery":80,"battery_low":false,"illuminance":16294,"illuminance_lux":43,"linkquality":92,"occupancy":false,"tamper":false,"temperature":20.81,"voltage":2900}'

### What did you expect to happen?

Z2M to set the correct state

### How to reproduce it (minimal and precise)

install one of these sensors.

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

 EZSP 7.x - whatever the latest on web flasher, just did it

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

### Setup

Homeseer on Windows

### Debug log

024-04-28 08:45:04MQTT publish: topic 'zigbee2mqtt/Sensor Theater Bosch Motion', payload '{"battery":100,"battery_low":false,"illuminance":9094,"illuminance_lux":8,"linkquality":136,"occupancy":false,"tamper":false,"temperature":19.21,"voltage":3000}'
Debug 2024-04-28 08:45:07Received Zigbee message from 'Sensor Garage Bosch Motion Door', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":10169}' from endpoint 1 with groupID 0
Info 2024-04-28 08:45:07MQTT publish: topic 'zigbee2mqtt/Sensor Garage Bosch Motion Door', payload '{"battery":100,"battery_low":false,"illuminance":10169,"illuminance_lux":10,"linkquality":188,"occupancy":false,"tamper":true,"temperature":17.54,"voltage":3000}'
Debug 2024-04-28 08:45:08Received Zigbee message from 'Sensor Stairs Aqara Motion bottom', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3025,"10":0,"100":0,"11":4,"3":20,"4":5032,"5":61,"6":[0,2]}}' from endpoint 1 with groupID 0
Debug 2024-04-28 08:45:08RTCGQ11LM: Processed data into payload {"voltage":3025,"battery":100,"device_temperature":20,"power_outage_count":60,"illuminance":4,"illuminance_lux":4}
Debug 2024-04-28 08:45:08RTCGQ11LM: Processed data into payload {"voltage":3025,"battery":100,"device_temperature":20,"power_outage_count":60,"illuminance":4,"illuminance_lux":4}
Info 2024-04-28 08:45:08MQTT publish: topic 'zigbee2mqtt/Sensor Stairs Aqara Motion bottom', payload '{"battery":100,"device_temperature":20,"illuminance":4,"illuminance_lux":4,"linkquality":196,"occupancy":false,"power_outage_count":60,"voltage":3025}'
Debug 2024-04-28 08:45:09Received Zigbee message from 'Sensor Office Bosch Motion', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":17571}' from endpoint 1 with groupID 0
Info 2024-04-28 08:45:09MQTT publish: topic 'zigbee2mqtt/Sensor Office Bosch Motion', payload '{"battery":80,"battery_low":false,"illuminance":17571,"illuminance_lux":57,"linkquality":104,"occupancy":false,"tamper":false,"temperature":20.81,"voltage":2900}'
Debug 2024-04-28 08:45:10Received Zigbee message from 'Sensor Living Room Bosch Motion LR Kitchen', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":17292}' from endpoint 1 with groupID 0
Info 2024-04-28 08:45:10MQTT publish: topic 'zigbee2mqtt/Sensor Living Room Bosch Motion LR Kitchen', payload '{"battery":100,"battery_low":false,"illuminance":17292,"illuminance_lux":54,"linkquality":236,"occupancy":false,"tamper":false,"temperature":19.75,"voltage":3000}'
Debug 2024-04-28 08:45:13Received Zigbee message from 'Sensor Mudroom Bosch Motion', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":7786}' from endpoint 1 with groupID 0
Info 2024-04-28 08:45:13MQTT publish: topic 'zigbee2mqtt/Sensor Mudroom Bosch Motion', payload '{"battery":80,"battery_low":false,"illuminance":7786,"illuminance_lux":6,"linkquality":172,"occupancy":false,"tamper":false,"temperature":20.32,"voltage":2900}'
Debug 2024-04-28 08:45:14Received Zigbee message from 'Sensor Theater Bosch Motion', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":8954}' from endpoint 1 with groupID 0
[... cut ...] ,"temperature":17.54,"voltage":3000}'
Debug 2024-04-28 08:45:19Received Zigbee message from 'Sensor Office Bosch Motion', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":16294}' from endpoint 1 with groupID 0
Info 2024-04-28 08:45:19MQTT publish: topic 'zigbee2mqtt/Sensor Office Bosch Motion', payload '{"battery":80,"battery_low":false,"illuminance":16294,"illuminance_lux":43,"linkquality":92,"occupancy":false,"tamper":false,"temperature":20.81,"voltage":2900}'

Ltek, 2024-05-08:
still not working properly in Z2M v1.37

Ltek, 2024-08-27:
Warning to Users : Z2M does not work/pair properly with this. When you can get it to paid (after like 10-15 times) it usually falls off hours or days later.

Ltek, 2024-12-31:
still doesn work properly

Ltek, 2025-03-03:
not dead yet

Ltek, 2025-03-13:
@Koenkk would appreciate if you could look at this. I have 20 of these, 10 on Deconz & 10 on Z2M. 2 of the Z2M never reset tamper. I have removed and reincluded them, it doesnt help. When I put them on Deconz, they work fine. I can use HA to set the Tamper to Off / Clear but after a bit it comes back.  I'd like to switch fully to Z2M but device stability issues like this are stopping me from doing so.

thank you

Koenkk, 2025-03-15:
With the latest z2m, could you provide the debug log of the following sequence:
- Trigger tamper -> tamper: true published
- Un-trigger tamper -> no tamper: false published.

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Ltek, 2025-03-15:
I got it working but it is really odd, check this out.  I set the UI to capture a Debug log and it didnt write any file. Also, oddly, it has not logged anything in 2 days (see screenshot)... 

[image]

BUT after trying a few times to get Z2M to interview or reconfig (via dashboard buttons + sensor in pairing mode). This failed showing the below errors. 

So on a whim I set Z2M to "Permit Join (All)" and set sensor in pairing mode, then clicked Interview and Configure buttons one immediately after the other... it WORKED.  They interviewed properly and the Tamper set to Clear immediately. Keep in mind these sensors were in the same place they have been for 2 years -- 1.5 yrs working perfectly on Deconz which had the old Conbee 1 in the same location (2 feet away from each other) as the Z2M Sonoff Dongle-E

Not sure if this gives you any clue why interviews and reconfig failed. Maybe the specific sequence helps explain why?

`Z2M Server: z2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of 'Sensor Garage Bosch Motion 1' (0x000d6f0011c74f23) failed: Error: Interview failed because can not get active endpoints ('0x000d6f0011c74f23')'
3/15/2025 4:02:35 PM

Z2M Server: z2m: Failed to configure 'Sensor Backyard Bosch Patio Motion', attempt 3 (Error: Bind 0x000d6f0011c8d507/1 msTemperatureMeasurement from '0xe0798dfffef42cde/1' failed (Delivery failed for '59803'.) at EmberOneWaitress.deliveryFailedFor (C:\Homeseer\Data\ZigbeePlus\zigbee2mqtt\node_modules\.pnpm\zigbee-herdsman@3.2.7\node_modules\zigbee-herdsman\src\adapter\ember\adapter\oneWaitress.ts:100:31) at EmberAdapter.onMessageSent (C:\Homeseer\Data\ZigbeePlus\zigbee2mqtt\node_modules\.pnpm\zigbee-herdsman@3.2.7\node_modules\zigbee-herdsman\src\adapter\ember\adapter\emberAdapter.ts:457:42) at Ezsp.emit (node:events:519:28) at Ezsp.ezspMessageSentHandler (C:\Homeseer\Data\ZigbeePlus\zigbee2mqtt\node_modules\.pnpm\zigbee-herdsman@3.2.7\node_modules\zigbee-herdsman\src\adapter\ember\ezsp\ezsp.ts:5055:14) at Ezsp.callbackDispatch (C:\Homeseer\Data\ZigbeePlus\zigbee2mqtt\node_modules\.pnpm\zigbee-herdsman@3.2.7\node_modules\zigbee-herdsman\src\adapter\ember\ezsp\ezsp.ts:746:26) at Ezsp.onAshFrame (C:\Homeseer\Data\ZigbeePlus\zigbee2mqtt\node_modules\.pnpm\zigbee-herdsman@3.2.7\node_modules\zigbee-herdsman\src\adapter\ember\ezsp\ezsp.ts:396:22) at UartAsh.emit (node:events:519:28) at Immediate. (C:\Homeseer\Data\ZigbeePlus\zigbee2mqtt\node_modules\.pnpm\zigbee-herdsman@3.2.7\node_modules\zigbee-herdsman\src\adapter\ember\uart\ash.ts:1171:45) at processImmediate (node:internal/timers:478:21))`

closed 2025-05-23 by github-actions[bot]