Koenkk/zigbee2mqtt #22338: Nous A1Z stop

StefanGruell, 2024-04-28:
### What happened?

When trying to toggle the on/off button of the Nous A1Z in the Zigbee2Mqtt menu nothing happens and a timeout message pops up. The reported linkquality was always around 100.

### What did you expect to happen?

That the plug switches it state

### How to reproduce it (minimal and precise)

Set up a network with the sonoff adapter and the Nous A1Z with lqi of 100. After a few days no commands can be sent to the plug anymore.

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ZBDongle-E 3.0 USB Dongle Plus

### Setup

Plain on a Raspberry Pi 4 8GB RAM

### Debug log

Publish 'set' 'state' to 'Strommesser1' failed: 'Error: Command 0xa4c138c1ed2dc73b/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 24684 - 1 - 36 - 6 - 11 after 10000ms)'

gummifisch24, 2024-05-12:
Had the same problem, tried to remove and add the plug again but now it wont even show up when trying to add it to Zigbee2MQTT in the Dashboard...

korttoma, 2024-05-21:
I am getting the same with one of my Nous A1Z now also (LQI=116) no communication anymore:

error: 	z2m: Publish 'get' 'state' to 'Garage_Fan' failed: 'Error: ZCL command 0xa4c138bb98b1f08b/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54701 - 1 - 162 - 6 - 1 after 10000ms)'

projectfreak, 2024-07-07:
I can confirm that. After a few days, several of my Nous A1Z Smart Plugs can no longer be switched. Energy measurement etc. continue to work. 
Restarting Zigbee2MQTT does not bring any improvement.
Sometimes it helps to disconnect a smart plug from the power supply and plug it back in after a while. But this rarely works.
Removing and reconnecting usually works.

Error message after unsuccessful switching:
`Error 2024-07-07 20:38:58Publish 'set' 'state' to 'Nous A1Z (01)' failed: 'Error: ZCL command 0xa4c138c8e8fb8baf/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 24036 - 1 - 6 - 6 - 11 after 10000ms)'`

Fehlermeldung nach Restart von Zigbee2MQTT:
`Error 2024-07-07 20:43:12Failed to configure 'Nous A1Z (01)', attempt 2 (Error: ZCL command 0xa4c138c8e8fb8baf/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 24036 - 1 - 6 - 0 - 1 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:485:47) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:766:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:401:29) at Object.configureMagicPacket (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:380:9) at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:4251:13) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:172:21) at Configure.configure (/app/lib/extension/configure.ts:121:13))`

Zigbee2MQTT version [1.37.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.37.1) commit: [ea39d86](https://github.com/Koenkk/zigbee2mqtt/commit/ea39d86)
Coordinator type EZSP v8
Coordinator revision 6.10.3.0 build 297

NothingWeAre, 2024-07-07:
In my case this looks like hardware problem
socket works fine if i place it right beside coordinator
but same error if it's even 50cm apart

net1957, 2024-07-14:
I have the same problem with  2  NOUS A1Z

one with original firmware, the second upgraded to version 192
manual on/off work on the plug.

HA 2024.7.2
Z2M 1.39.0
Coordinator : SLZB-06M  zigbee:  20240510, core:  v2.3.6,  ha driver: ember

devironglove, 2024-08-22:
Also yet happened to me today. I've 15+ NOUS A1Z and this is the first time I've this with one of them.
It seems impossible to re-add it after having it forced removed because I couldn't get it to work.
I Suspect an hardware error :(

honkbig, 2024-09-07:
I also have one out of four Nous A1Z that stopped reporting. I thought it would magically be better by updating to Ember but it made no difference. It works as a switch (pressing the button) but it does not connect or report to network. I tried to reset it (hold power button until it flashes). Did not work. 
Versions:
Zigbee2mqtt  1.40-1.1
Sonoff usb stick v2 with firmware ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl 
It is run in Home Assistant on a Raspberry pi

korttoma, 2024-09-09:
Nous sent me a replacement device as according to them the device is faulty. I have 8 of them and only one has failed since I started to use them about 9 months ago.

phifre, 2024-09-09:
FYI, out of the 14 "Nous A1Z" I bought, 2 were faulty (around 15%, similar ration as @korttoma ) and they replaced them also (after some discussion).

honkbig, 2024-09-19:
> I also have one out of four Nous A1Z that stopped reporting. I thought it would magically be better by updating to Ember but it made no difference. It works as a switch (pressing the button) but it [...]

My device was replaced ==> problem solved.

McTristan, 2024-10-15:
Me too but my devices are brand new - yesterday all of them worked, today 2 of them stopped working. Weird ...

zoneme99, 2024-10-16:
Two has dropped off for me but got them from my friend so can't get them replaced:(

zoneme99, 2024-10-16:
I don't know but I have a sonoff dongle E I think

motionist, 2024-11-12:
I have the same issue.
Controller is SonOff.

chevdor, 2024-11-22:
I am running into the same issue.

I have a total of 6 devices. Initially all worked fine, no problem whatsoever.
A few weeks into using them, some dropped from the network and so far I never got them back in.  My failure ratio is 4/6, which is very high and the issue showed up around 2-3 weeks after installing a "brand new" device they sent me as replacement. 

I described the issue [here](https://community.openhab.org/t/nous-a1z-network-drop-forever-bricked/160165/15).

Initially NOUS replaced my first faulty devices, not skipping to put the blame on the devices plugged into the socket (which btw was a standard ventilator, CE Labelled and working very well with other sockets...) or on the power source (I am using an inverter on **some**). They ignored all the infos I sent, and sent a replacement (it took a while...).
Besides NOUS accusing everything surrdounding their devices (Z2M, consummer device, power source), I have tested enough to know it is not valid since I ran into the issue with other software than Z2M too, other power sources, and other consummers. Surprisingly, I am NOT running into the issue on devices powering bigger machines such as washing machine and dryers.

I was hoping they would be willing to troubleshoot the issue to understand it but they did not seem to care much.

Now they start playing stupid, despite all my tests and attemps to troubleshoot the issue for them... currently playing the clock and postponing replacements under warranty.

I tested with 3 constellations:
- OpenHab + Zigbee Bindng and Aeotech Zi-Stick
- eWelink + Sonoff ZP Bridge -P
- Sonoff Zigbee 3.0 USB Plus -E with Z2M

No matter what I do with my 4 "dead" devices, none of the constellations above see ANY ZigBee comm.
The relay and the button on the 4 devices does work. So they are not totally fried but the ZigBee part appears to be dead.
I installed Z2M hoping I could do an OTA but for that, the devices would have to be recognized.

I tested a few  "valid" NOUS A1Z and they paired totally fine. I could run the OTA and noticed the firmware going from 78 to 192.

Those devices are the only ones I am having trouble with.

My current stand is:
- the devices are (parlty ?) fine: the relay/power part works
- pressing 10s on the button does show entering the reset/pair mode, the led blinks
- the zigbee software / hardware appears to be faulty
- devices **not** directly connected to the controller seem to be less at risk ?
- all faulty devices had an older firwmware
- I have updated the working devices, time will tell if they break too after a while or not
- I am not sure if the devices are recoverable (opening those seems to be tricky due to the glue they use)
- my Sonoff dongle has been freshly updated (like 2d ago...), I am now on SDK 4.4.4, EZSP 7.4.4.0 and using `ember`

cyberden, 2025-01-21:
I have the same issue with a Sonoff plug which used to work fine for months on my Z2M network. I removed it and can't add it back. I think it's dead.

I had another one lying around that joined the network without issue.

chevdor, 2025-01-21:
I suspect that NOUS knows about the issue. They kept sending me replacements wihout asking much despite my proposal to provide quite some details. The last time they sent me replacements, they insisted that they cannot garantee it will work without issue and the issue will no happen again.

It **seems** that applying the lastest firwmare helps a bit but I had devices die on me on the latest firmware as well.
I had the issue now on some 4+ devices. I never got a faulty one back into a Zigbee network.

NOUS keep on putting the fault on my devices... but that is hilarious at this point as it sounds like most devices whether Ventinaltor, LED lighting, and in some case no plugged in device at all... would cause the issue....

I fear the best we can do is contact NOUS and ask for replacements... if they get tired, they may fix the product or the firmware and provide an update.

I don't have the bandwidth atm to debug those products but it would be nice to know if the device is actually dead or if it is somehow recoverable via some serial update/reset.

sleveille4, 2025-02-07:
hello,

found this thread as i have the same issue with one of my A1Z.

but i might have found a solution!

like you, pairing was not possible most of the time and when possible, device not responding after a few seconds.

what i found tho is when i plug it on top of another zigbee plug, i could pair it like normal every time

then it worked, but as soon as i moved it even 5cm => dropped
 (like if this was a signal strenght issue)

so i put it back, it worked and i looked at the "state" tab in Z2MQTT.

and noticed it returned FW version "-1" (instead of 192)

sure thing, => OTA tab, FW check => update available
so i "updated" it (actually it just reflashed the same FW in it).

and so far, it seems to be back working as expected (but only since a few hours, so to be confirmed, but thats alot of improvement already)

tldr of what i did:
1- plug it directly on another (working) zigbee plug
2- pair it,
3- check FW version (it reported -1 in states)
4- update FW

chevdor, 2025-02-07:
That's VERY interesting. I had one of them left that was "dead".
I tried your trick and plugged it into another one that was turned on. I triggered a reset and it did pair and show up !!!
Unfortunately, I already did flash a new firmware and I don't see a new version.

Countdown however was reset to `0` and I get errors saying `countdown` cannot be fetched.
My firmware version remained:
```
 "update": {
        "installed_version": 192,
        "latest_version": 192,
        "state": "idle"
    },
```

I am searching for a trick to re-apply a firmware update no matter what.

That looks very much like a bug in their firmware that is probably resetting some params du default.

Many thanks for sharing your findings @sleveille4 !

chevdor, 2025-02-07:
Adding that with the A1Z 1m away from the other working plug, it starts failing again, but as soon as on the back of the other, it works again. I think the radio settings are getting messed up and it looks like a reset does not reset them.

chevdor, 2025-02-07:
No idea if that will work but I found how we may be able to re-flash:
Send a MQTT request to `zigbee2mqtt/bridge/request/device/ota_update/check/downgrade` with the content `{"id": "ZB073"}` (`ZB073` is my friendly name).

The UI changes to:

[image]

but the upgrade fails as no new image is available.

sleveille4, 2025-02-07:
> Adding that with the A1Z 1m away from the other working plug, it starts failing again, but as soon as on the back of the other, it works again. I think the radio settings are getting messed up and it looks like a reset does not reset them.

came here again to post this exact same conclusion as you, will try to have zigbee network rebuild by shutting down HA for 30m the starting it back up, but i don't have much hope from this....

chevdor, 2025-02-08:
Annying in the story is that NOUS tends to put the blame on the users or the devices...
At least knowing more about the issue can help other users recognise it.
There may still hope if we find a wayt to reflash the device. Hopefully we can do that without openming it and using serial.

chevdor, 2025-04-10:
keep open

boscorelly, 2025-05-13:
> No idea if that will work but I found how we may be able to re-flash: Send a MQTT request to `zigbee2mqtt/bridge/request/device/ota_update/check/downgrade` with the content `{"id": "ZB073"}` (`ZB073` is my friendly name).
> 
> The UI changes to:
> <img alt="Image" width="910" src="https://private-user-images.githubusercontent.com/738724/410937482-7eb2935e-23ee-4939-90cf-bdb098def9f8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDcxNzg1NjgsIm5iZiI6MTc0NzE3ODI2OCwicGF0aCI6Ii83Mzg3MjQvNDEwOTM3NDgyLTdlYjI5MzVlLTIzZWUtNDkzOS05MGNmLWJkYjA5OGRlZjlmOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNTEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDUxM1QyMzE3NDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jNTE5YjNhMTRmOTMyYjZkYjUyMGUxNDFiNzJiZTY0NzVkODFhNTZmNDliMGJhNmUyYTk5MzFmMDlkYmNmNTc2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.um0KH52ACmX6te8nnAo2bhKvjfagQPq2uTx0qLC-XYY">
> 
> but the upgrade fails as no new image is available.

hi,

how do you do it ? in dev console ?

boscorelly, 2025-05-13:
> hello,
> 
> found this thread as i have the same issue with one of my A1Z (stopped responding, almost impossible to pair again, and when paired, stopped working in seconds).
> 
> but i might have found a solution!
> 
> like you, pairing was not possible most of the time and when possible, device not responding after a few seconds.
> 
> what i found tho is when i plug it on top of another zigbee plug, i could pair it like normal every time
> 
> then it worked, but as soon as i moved it even 5cm => dropped (like if this was a signal strenght issue)
> 
> so i put it back, it worked and i looked at the "state" tab in Z2MQTT.
> 
> and noticed it returned FW version "-1" (instead of 192)
> 
> sure thing, => OTA tab, FW check => update available so i "updated" it (actually it just reflashed the same FW in it).
> 
> and so far, it seems to be back working as expected (but only since a few hours, so to be confirmed, but thats alot of improvement already)
> 
> tldr of what i did: 1- plug it directly on another (working) zigbee plug 2- pair it, 3- check FW version (it reported -1 in states) 4- update FW

i can confirm it works, except the part where i must update firmware.
the report say "version: -1" but i don't know how to upgrade it as i don't have any firmware available.

boscorelly, 2025-05-13:
i found how to send the message, but i get an error :

[image]

johnyb0y, 2025-07-31:
Hi, just a warning for anyone stumbling over this issue. I now had 2 out of 4 of these failing with the exact symptoms described here. Plugging them in only a few centimeters from the hub brings them online again - but that defeats their purpose. 
They sent me a replacement for one of the failed ones - the replacement failed with the same symptoms two weeks later.
I upgraded all of them to the newest firmware "192", doesn't matter.
Something is seriously wrong with this model - buy something else.

boscorelly, 2025-08-01:
shittiest zigbee device ever :(

johnyb0y, 2025-08-05:
The current Firmware "192" is rather old. I have them in two other Nouz devices since at least a year. These are still working fine. 
Only the two I bought recently crapped out after a few days, both updated to "192". 
This could also be due to something like a new device revision with some undisclosed hardware changes. Just a theory of mine.

johnyb0y, 2025-08-05:
The GitHub app doesn't let me edit my last comment, so another comment from me. 
Not really relevant for this issue, but maybe helpful to some here. 
As an alternative I bought two "Innr SP240". Both work fine with Z2M, firmware update worked as well, and they seem to do a good job as routers in my zigbee network.

1969kowal-boop, 2025-12-01:
Szukałem pomocy i natrafiłem na ten post. Dostałem zwrot pieniędzy za część 6 gniazd jednak za 4 nie  i chciałem je naprawić co skończyło się totalną destrukcją. Te gniazda są wadliwe i tak jak wspomnieliście firma nic sobie z tego nie robi. Aktualizacja nic nie daje ,a wręcz szkodzi. Gniazda powodują silne "zapychanie" sieci zigbee dlatego gdy już nie działają nie należy ich podłączać ani pozostawić w zasilaniu!. To ostatnie które dzisiaj "skończyło swoje życie" Jeśli możecie trzeba po prostu bojkotować tak złej jakości urządzenie Zigbee  bo psuje całą markę temu rozwiązaniu.

[image]

[image]

C-Duv, 2026-08-28:
Still an issue as no development was done to address it (that I'm aware of).

boscorelly, 2026-08-28:
Tu peux jeter cette daube.

C-Duv, 2026-10-29:
Not-a-bot creating activity here.

andrei-lazarov, 2026-10-29:
Please keep the issue open only if it's something we can fix on the Z2M side.
Make sure the coordinator firmware is updated and your network is healthy (stable, no spam, no interference).. 
Otherwise replace the plug 🙁

chevdor, 2026-10-31:
Unfortunately a fair answer.
Nothing points in the direction of issues in Z2M itself.

It is nice for this issue to exist to help people find some answers but I doubt Z2M can fix anything there.
This issue should be closed IMO.

boscorelly, 2026-11-02:
anyone has an export of the firmware ? I wannt to look inside it