Koenkk/zigbee2mqtt #22347: Problem with WHD-02, offline

Ant118, 2024-04-29:
### What happened?

Tuya WHD02 switch offline.
If you press the button directly on the device, it appears online again

### Zigbee2MQTT version

1.36.1-dev commit: 1544317

### Adapter firmware version

7.4.1 [GA]

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2 (Dongle E)

### Setup

Home Assistant addon

### Debug log

[2024-04-29 13:54:44] warning: 	z2m: Failed to ping 'ShkafLight' (attempt 1/1, ZCL command 0x70b3d52b6007fcf0/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (ERROR_INVALID_CALL))
[2024-04-29 13:54:47] debug: 	z2m: Active device 'ShkafLight' was last seen '30.32' minutes ago.

[2024-04-29 14:04:48] error: 	zh:ember:ezsp: [SEND COMMAND] Cannot send second one before processing response from first one.
[2024-04-29 14:04:48] debug: 	zh:controller:endpoint: ZCL command 0x70b3d52b6007fcf0/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (ERROR_INVALID_CALL)
[2024-04-29 14:04:48] warning: 	z2m: Failed to ping 'ShkafLight' (attempt 1/1, ZCL command 0x70b3d52b6007fcf0/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (ERROR_INVALID_CALL))
[2024-04-29 14:04:51] debug: 	z2m: Active device 'ShkafLight' was last seen '40.39' minutes ago.
[2024-04-29 14:07:30] debug: 	z2m: Saving state to file /config/zigbee2mqtt/state.json

Ant118, 2024-04-29:
when I press the button on the device, the device does not always appear

Ant118, 2024-04-29:
[image]

Ant118, 2024-04-30:
It is problem adapter firmware version 7.4.1.
6.10.7.0 build 435 - OK.

closed 2024-04-30 by Ant118

htppseb, 2025-01-07:
Hey for me its not fully clear what the fix actually was. can you explain a bit more? is the problem the firmware of the ZigBee stick or the smart switch?