Koenkk/zigbee2mqtt #22349: Dimmer range from min to max brightness ignored?

UsefulVid, 2024-04-29:
### What happened?

I set the dimmer min and max range from min 10 to max 150.
These values seem to be ignored. I am using HA and the values with the percent slider are always calculated as if 254 is the max value.
In HA my slider is only working in the lower part and in the upper part it is of no use.
TS0601_dimmer_5

### What did you expect to happen?

I expected that I could dim precisely in the min and max range and not under or above.

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

EZSP v12 Sonoff

### Setup

Addon on HA

Domochip, 2024-09-07:
I have the same problem with : 
Zigbee Model: TS0601
Zigbee Manufacturer: _TZE204_5cuocqty
Description: Zigbee 1 channel Dimmer

Here is my z2m infos : 

Zigbee2MQTT version: 1.40.1 403d3c0
Coordinator type: zStack3x0
Coordinator revision: 20240710

do you still have it?

UsefulVid, 2024-09-07:
Yes problem still present

UsefulVid, 2024-12-29:
no stale please :)

UsefulVid, 2025-03-01:
Still present

closed 2025-05-09 by github-actions[bot]

vitalybibikov, 2025-06-03:
Still present