Koenkk/zigbee2mqtt #22355: ZBMINIL2 can't be added to group

martkaczmarek, 2024-04-29:
### What happened?

Sonoff ZBMINIL2 can't be added to a group; 
When trying to add it to group, it fails. 
Clusters of this device don't contain GenGroups, as far as I know it's related. 

### What did you expect to happen?

I expect ZBMINIL2 to join the group

### How to reproduce it (minimal and precise)

1. Join Sonoff ZBMINIL2 to your network
2. Create a group
3. Try adding Sonoff ZBMINIL2 to your group

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

6.7.8.0 build 373

### Adapter

zbbridge Pro

### Setup

Addon on full HA, HP T520

numminorih-git, 2024-05-14:
somehow ZBMINIL2 added to group, but doesn't react for group command. And I can't delete it now...

swinkapetunia, 2024-06-22:
Same issue, version 1.38.0. I can add ZBMINIL2 to a group as well as control them individually when previewing the group, but they don't respond to commands (other devices from the same group do). They can't be removed from the group and the group cannot be deleted.

closed 2025-01-05 by github-actions[bot]