Koenkk/zigbee2mqtt #22361: Zemismart blind ZM25RX-08/30 stopped responding 2 weeks ago

TRusselo, 2024-04-30:
### What happened?

My smart blind no longer reacts to commands from zigbee2mqtt nor home assistant.
Shows online.  Last seen updates time often.

wont go up/down when commanded.

HA -diags
[mqtt-d7c42d6885b84ea688201fd4c5799055-Living Room Shade-2469241cee2e2c6823ae8ff07573ccd7.json](https://github.com/Koenkk/zigbee2mqtt/files/15162766/mqtt-d7c42d6885b84ea688201fd4c5799055-Living.Room.Shade-2469241cee2e2c6823ae8ff07573ccd7.json)

tried to "reconfigure"
got error:
2024-04-30 05:21:27Failed to configure 'Living Room Shade', attempt 1 (Error: ZCL command 0x70ac08fffe12a11b/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 9091 - 1 - 15 - 0 - 1 after 10000ms) at Object.start (/app/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at EZSPAdapter.sendZclFrameToEndpointInternal (/app/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:492:47) at Queue.execute (/app/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28) at Endpoint.read (/app/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:404:24) at Object.configureMagicPacket [as configure] (/app/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:377:9) at Configure.configure (/app/zigbee2mqtt/lib/extension/configure.ts:122:13) at Configure.onMQTTMessage (/app/zigbee2mqtt/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/zigbee2mqtt/lib/eventBus.ts:174:17))

### Zigbee2MQTT version

 1.36.1 

### Adapter firmware version

7.1.1.0 build 273

### Adapter

skyconnect

### Setup

docker on unraid

julianfs, 2024-05-01:
Sounds like you'll need to re-pair the device, possibly needing to delete it from Z2M before going through the process.

TRusselo, 2024-05-03:
that worked.

closed 2024-05-03 by TRusselo