Koenkk/zigbee2mqtt #22364: [New device support]: Immax NEO 07523L Soil temperature & humidity sensor

slush0, 2024-04-30:
### Link

https://www.immax.eu/immax-neo-smart-soil-moisture-and-temperature-sensor-zigbee-p18721/

### Database entry

{"id":15,"type":"EndDevice","ieeeAddr":"0xa4c138146dc0c51f","nwkAddr":4539,"manufId":4417,"manufName":"_TZE200_2se8efxh","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"G\u0011�-iM\u0011�-i\u0000\u0000\u0000\u0000\u0011\u0000\u0000\u0000\u0000\u0011","65506":56,"65508":1,"65534":0,"stackVersion":0,"dateCode":"","appVersion":72,"manufacturerName":"_TZE200_2se8efxh","zclVersion":3,"modelId":"TS0601","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1714502975079}

### Comments

I'm attaching working definition, it exports battery level, temperature and humidity.

### External definition

```shell
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_2se8efxh',
        },
    ],
    model: '07523L',
    vendor: 'Immax',
    description: 'NEO Soil temperature & humidity sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [e.battery(), e.temperature(), e.humidity()],
    meta: {
        tuyaDatapoints: [
            [3, 'humidity', tuya.valueConverter.raw],
            [5, 'temperature', tuya.valueConverter.raw],
            [15, 'battery', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

Koenkk, 2024-05-01:
`humidity` is the `soil_moisture` right? Does the following work?

```
        exposes: [e.temperature(), e.soil_moisture(), tuya.exposes.temperatureUnit(), e.battery(), tuya.exposes.batteryState()],
        meta: {
            tuyaDatapoints: [
                [3, 'soil_moisture', tuya.valueConverter.raw],
                [5, 'temperature', tuya.valueConverter.raw],
                [9, 'temperature_unit', tuya.valueConverter.temperatureUnit],
                [14, 'battery_state', tuya.valueConverter.batteryState],
                [15, 'battery', tuya.valueConverter.raw],
            ],
        },
```

supaeasy, 2024-05-02:
From the looks I have the exact device (bought from Aliexpress). Z2M recognizes this as https://www.zigbee2mqtt.io/devices/TS0601_soil.html

It is absolute garbage though. Soil moisture is utterly wrong and there is no way to calibrate it.

Koenkk, 2024-05-02:
Added it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-02 by Koenkk

slush0, 2024-05-04:
> `humidity` is the `soil_moisture` right? Does the following work?
>
> ```
>         exposes: [e.temperature(), e.soil_moisture(), tuya.exposes.temperatureUnit(), e.battery(), [...]

Yes, this definition works for my device, thanks!

victor-vc, 2025-07-17:
I have just paired such a sensor and see in the log:
`info 2025-07-17 21:22:19z2m: Successfully interviewed '0xa4c13862400b0d7b', device has successfully been paired`
`warning 2025-07-17 21:22:19z2m: Device '0xa4c13862400b0d7b' with Zigbee model 'TS0601' and manufacturer name '_TZE284_2se8efxh' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html`

Koenkk, 2025-07-18:
@victor-vc added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

victor-vc, 2025-07-20:
Could you please update the device pictur for this model? Zigbee2MQTT Edge has correctly detected the soil moisture sensor but its icon is the Zigbee logo. Thanks again.