Koenkk/zigbee2mqtt #22368: Aqura WSDCGQ11LM disconnects without any indication

lanwin, 2024-05-01:
### What happened?

I have an Sonoff ZBDongle-E adapter and currently planing with different firmware versions, multipan and EZSP/Ember backend. While doing this it happens that my Aqura WSDCGQ11LM sensors disconnect. This is a known problem so far. My problem is that there is no indication that this happens. The only way to see it is that in Home Assist the value graph shows a straight line. 

### What did you expect to happen?

Even when Aqura sensors did not impl Zigbee correctly. A sensor is expected to send a value at least one in a specific time. I think it would be a good idea to consider a device offline when there is no value update in a given time. This would also clear the values in HA.

### How to reproduce it (minimal and precise)

-

### Zigbee2MQTT version

1.36.1-dev

### Adapter firmware version

7.4.2 [GA]

### Adapter

Sonoff ZBDongle-E

### Setup

Docker

lux73, 2024-05-01:
Aqara Temperature Devices are known to send only "new" Values...

this can be within 5min or 2h interval - there's nothing that can do z2m

if the Sensors means "now i send a new Value" he does ist - z2m could not force him

lanwin, 2024-05-01:
@lux73 thank you for your answer. That is true. But it in example it is very very unlikely that a room temperature sensor did not send any update in 3h+.

lux73, 2024-05-01:
i agree - this is why i replace all my Inhouse Aqaras with [Danfoss Room Sensors](https://store.danfoss.com/de/de/Climate-Solutions-W%C3%A4rmetechnik/Smart-heating/Danfoss-Ally-%E2%80%93-smart-heating/Thermostats-and-Sensors/Zigbee/p/014G2480)

these send a new Value at least all 30min - or quicker

the Problem with Aqara: even when the Temperature Difference is big, sometimes they won't update. Thats weird

Anonym-tsk, 2024-08-12:
The same problem with my 2 WSDCGQ11LM sensors. They worked for years, but I had 5 disconnects last month.

jariha, 2024-11-19:
Same, something has broken as I have 3 of those sensors and now those are dropping offline.

closed 2025-01-27 by github-actions[bot]