Koenkk/zigbee2mqtt #22375: zigbee2mqtt will not start with conbee 3 because Error "Cannot read properties of undefined"

FrazerDungeon, 2024-05-01:
### What happened?

Hello,
I try to install and start zigbee2mqtt on iobroker with raspi 5 8 GB.
npm start or DEBUG=zigbee-herdsman* npm start fails with conbee 3

Zigbee2MQTT:info  2024-05-01: Starting Zigbee2MQTT version 1.36.1 (commit #8c8a6a3)
Zigbee2MQTT:info  2024-05-01: Starting zigbee-herdsman (0.40.3)

### What did you expect to happen?

That I can start it.

### How to reproduce it (minimal and precise)

npm start or DEBUG=zigbee-herdsman* npm start

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

2.13.6

### Adapter

dresden electronics conbee 3

### Setup

raspi 5 

### Debug log

pi@iobroker:/opt/zigbee2mqtt $ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.36.1 start
> node index.js

Zigbee2MQTT:debug 2024-05-01: Loaded state from file /opt/zigbee2mqtt/data/state                                                                                                                     .json
Zigbee2MQTT:info  2024-05-01: Logging to console and directory: 'data/log/2024-0                                                                                                                     5-01.19-25-58' filename: log.txt
Zigbee2MQTT:debug 2024-05-01: Removing old log directory 'data/log/2024-04-30.22                                                                                                                     -19-44'
Zigbee2MQTT:info  2024-05-01: Starting Zigbee2MQTT version 1.36.1 (commit #8c8a6                                                                                                                     a3)
Zigbee2MQTT:info  2024-05-01: Starting zigbee-herdsman (0.40.3)
Zigbee2MQTT:debug 2024-05-01: Using zigbee-herdsman with settings: '{"adapter":{                                                                                                                     "concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqt                                                                                                                     t/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/data                                                                                                                     base.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"c                                                                                                                     hannelList":[11],"extendedPanID":[118,224,97,209,38,36,194,169],"networkKey":"HI                                                                                                                     DDEN","panID":4823},"serialPort":{"adapter":"deconz","path":"/dev/ttyUSB0"}}'
  zigbee-herdsman:deconz:driver Set delay to 0 +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyD                                                                                                                     istribute":false,"networkKey":[127,157,126,146,19,117,194,61,217,167,144,120,131                                                                                                                     ,218,218,41],"panID":4823,"extendedPanID":[118,224,97,209,38,36,194,169],"channe                                                                                                                     lList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"deconz"},"databasePa                                                                                                                     th":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/d                                                                                                                     ata/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.j                                                                                                                     son","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:deconz:driver Opening with /dev/ttyUSB0 +2ms
  zigbee-herdsman:deconz:driver Serialport opened +10ms
  zigbee-herdsman:controller:log Started with result 'resumed' +10ms
  zigbee-herdsman:controller:log Injected database: true, adapter: true +1ms
  zigbee-herdsman:controller:device:log Request Queue (38): default expiration t                                                                                                                     imeout set to 0 +0ms
  zigbee-herdsman:controller:database:log Writing database to '/opt/zigbee2mqtt/                                                                                                                     data/database.db' +0ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1                                                                                                                      paramId: 1 +7ms
  zigbee-herdsman:deconz:adapter checking coordinator simple descriptor +0ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x0 EP:0 SeqNr. 2 reque                                                                                                                     st id: 1 +3s
  zigbee-herdsman:deconz:driver Timeout for request - CMD: 0xa seqNr: 1 +7s
Zigbee2MQTT:error 2024-05-01: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-05-01: Failed to start zigbee
Zigbee2MQTT:error 2024-05-01: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-05-01: Exiting...
Zigbee2MQTT:error 2024-05-01: undefined
  zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 3 +6ms
pi@iobroker:/opt/zigbee2mqtt $ sudo journalctl -u zigbee2mqtt.service -f
Apr 30 22:25:45 iobroker node[1165]:                     ^
Apr 30 22:25:45 iobroker node[1165]: TypeError: Cannot read properties of undefined (reading 'end')
Apr 30 22:25:45 iobroker node[1165]:     at MQTT.disconnect (/opt/zigbee2mqtt/lib/mqtt.ts:130:21)
Apr 30 22:25:45 iobroker node[1165]:     at Controller.stop (/opt/zigbee2mqtt/lib/controller.ts:214:9)
Apr 30 22:25:45 iobroker node[1165]:     at stop (/opt/zigbee2mqtt/index.js:111:5)
Apr 30 22:25:45 iobroker node[1165]:     at process.handleQuit (/opt/zigbee2mqtt/index.js:117:9)
Apr 30 22:25:45 iobroker systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Apr 30 22:25:45 iobroker systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Apr 30 22:25:45 iobroker systemd[1]: Stopped zigbee2mqtt.service - zigbee2mqtt.
Apr 30 22:25:45 iobroker systemd[1]: zigbee2mqtt.service: Consumed 3.020s CPU time.

taulen, 2024-05-07:
Did you ever get it working ?
Seeing the same thing here

root@zigbee2mqtt:/opt/zigbee2mqtt# cd /opt/zigbee2mqtt && DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.37.0-dev start
> node index.js

[2024-05-07 19:38:44] info:     z2m: Logging to console, file (filename: log.log)
[2024-05-07 19:38:44] debug:    z2m: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
[2024-05-07 19:38:45] info:     z2m: Starting Zigbee2MQTT version 1.37.0-dev (commit #3544d517)
[2024-05-07 19:38:45] info:     z2m: Starting zigbee-herdsman (0.46.4)
[2024-05-07 19:38:45] debug:    z2m: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[20],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":5771},"serialPort":{"adapter":"deconz","path":"/dev/serial/by-id/usb-dresden_elektronik_ConBee_III_DE03216377-if00-port0"}}'
[2024-05-07 19:38:45] debug:    zh:deconz:driver: Set delay to 0
[2024-05-07 19:38:45] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[92,203,83,131,143,94,4,148,143,15,105,74,168,50,209,89],"panID":5771,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[20]},"serialPort":{"path":"/dev/serial/by-id/usb-dresden_elektronik_ConBee_III_DE03216377-if00-port0","adapter":"deconz"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
[2024-05-07 19:38:45] debug:    zh:deconz:driver: Opening with /dev/serial/by-id/usb-dresden_elektronik_ConBee_III_DE03216377-if00-port0
[2024-05-07 19:38:45] debug:    zh:deconz:driver: Serialport opened
[2024-05-07 19:38:45] debug:    zh:controller: Started with result 'resumed'
[2024-05-07 19:38:45] debug:    zh:controller: Injected database: true, adapter: true
[2024-05-07 19:38:45] debug:    zh:controller:database: Writing database to '/opt/zigbee2mqtt/data/database.db'
[2024-05-07 19:38:45] debug:    zh:deconz:driver: send read parameter request from queue. seqNr: 1 paramId: 1
[2024-05-07 19:38:48] debug:    zh:deconz: checking coordinator simple descriptor
[2024-05-07 19:38:48] debug:    zh:deconz:driver: DATA_REQUEST - destAddr: 0x0 EP:0 SeqNr. 2 request id: 1
[2024-05-07 19:38:55] debug:    zh:deconz:driver: Timeout for request - CMD: 0xa seqNr: 1
[2024-05-07 19:38:55] error:    z2m: Error while starting zigbee-herdsman
[2024-05-07 19:38:55] error:    z2m: Failed to start zigbee
[2024-05-07 19:38:55] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-07 19:38:55] error:    z2m: Exiting...
[2024-05-07 19:38:55] error:    z2m: undefined
root@zigbee2mqtt:/opt/zigbee2mqtt#

taulen, 2024-05-07:
For future me, had to add "baudrate: 115200" under the serial part :)

FrazerDungeon, 2024-05-08:
I updated on 1.37 and added Baudrate: 155200
This worked for me now. Thanks

closed 2024-05-08 by FrazerDungeon