Koenkk/zigbee2mqtt #22384: Update Home Assistant entity ID is missing

rhug123, 2024-05-01:
### What happened?

When editing a name of a device, in the previous version it included a toggle to change the entity id as well.

### What did you expect to happen?

The toggle to be included when renaming a device

### How to reproduce it (minimal and precise)

Go to rename a device and there should be no toggle. See link below for example.
https://community.home-assistant.io/t/force-entity-id-update-for-discovered-entities/227167

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

0

### Adapter

Sonoff Zigbee Stick E

### Setup

Homeassistant in VM

prokudin07, 2024-05-01:
+1
in mobile version absent "Rename device"

bilfive, 2024-05-01:
The same problem. I have updated one instance.. And one not. At 1.36.1 this:
[image]
and now at 1.37.0 only this:
[image]

harvindhillon, 2024-05-02:
I just did a quick test. I added a new device and renamed it. I see that the name in HA is updated as well automatically. I renamed it again after and it updated as well.

So maybe this is now the default behaviour and kind of makes sense to me :) I like it being someone who forgets to set it

harvindhillon, 2024-05-02:
Okay silly me, all it does it renames the device, but not the entities, apologies, yes this is an annoying bug now 🙍‍♂️

FWeinb, 2024-05-02:
This regression was already raised in https://github.com/nurikk/zigbee2mqtt-frontend/pull/2010. Will probably be fixed in the next release.

Korvenwin, 2024-05-02:
@FWeinb until the next release, is there any other way to setup the name in Home Assitant?

stefangries, 2024-05-02:
@FWeinb You can manually publish a message via MQTT to trigger a rename:
https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request

Koenkk, 2024-05-02:
Fixed by @danroc , many thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-02 by Koenkk