Koenkk/zigbee2mqtt #22385: Doesn't work function to change friendly name in mobile app and mobile browser

zvezdochet2004, 2024-05-01:
### What happened?

The button to change friendly name has disappeared in the mobile app and mobile version browser Chrome

### How to reproduce it (minimal and precise)

Open the device panel or device card

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20210813

### Adapter

zStack3x0

### Setup

Addon in home assistant supervised

tsloms, 2024-05-02:
Same problem here

jstaro, 2024-05-02:
Likewise, in the desktop version, you can no longer set "Description".

timwnl, 2024-05-02:
Same issue here in iOS safari browsers, running in docker. Works fine in desktop browsers

Koenkk, 2024-05-02:
@nurikk could you check this?

serlinGi, 2024-05-02:
with android phone this feature rewards you by placing the phone horizontally

TheRealWaldo, 2024-05-02:
The same is occurring in the Home Assistant application.

Milinator, 2024-05-02:
> The same is occurring in the Home Assistant application.

Just found out: If you put your phone into horizontal mode the edit button appears. Strange because that wasn't the case before this update and I don't understand why that'd be a better approach. It worked like this.

TheRealWaldo, 2024-05-02:
Also note: the 'also rename home assistant' switch is gone on both mobile AND web.

Koenkk, 2024-05-04:
> Also note: the 'also rename home assistant' switch is gone on both mobile AND web.

This has already been fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2024-11-01 by Koenkk