Koenkk/zigbee2mqtt #22386: Problem changing colors with z2m groups

wastez, 2024-05-01:
### What happened?

Since I while I’m using z2m group for a group of lights (6 spots). 
The problem is that if I change colors using the color pallete that it is not working good. Sometimes they are changing the color really slow or some of them do not change the color. 
If i use Home Assistant groups I don’t have the problem, it is working as it should. 

My stick is a sonoff dongle e with ember. I‘m using the z2m 1.37. 0. 

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

ezsp ncp 7.4.2

### Adapter

Sonoff dongle e

### Setup

Addon on Home Assistant

wastez, 2024-05-03:
@Nerivec 
What do you think about it?
The problem was already before 1.37. 
I don’t think the problem is in the driver.

Nerivec, 2024-05-04:
Can you provide some `debug`-level logs when this happens?

wastez, 2024-05-10:
Ok sorry missed this message

wastez, 2024-05-10:
Ok here it is. 
Network is to busy because of broadcasting, you will see it in the logs. 
[debug.log](https://github.com/Koenkk/zigbee2mqtt/files/15274756/debug.log)

wastez, 2024-05-10:
@Nerivec 
Any thoughts?

Nerivec, 2024-05-10:
The Zigbee spec prevents more than 8 broadcasts over a 9-second period.
[Silabs docs](https://community.silabs.com/s/article/how-are-different-tables-managed-in-emberznet-x?language=en_US) (see paragraph "Broadcast Table").

<details>
<summary>Filtered logs</summary>

```logs
[2024-05-10 14:04:11] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=0 messageTag=8 status=SUCCESS]
[2024-05-10 14:04:11] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=1 messageTag=9 status=SUCCESS]
[2024-05-10 14:04:12] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=2 messageTag=10 status=SUCCESS]
[2024-05-10 14:04:12] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=3 messageTag=11 status=SUCCESS]
[2024-05-10 14:04:13] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=4 messageTag=12 status=SUCCESS]
[2024-05-10 14:04:13] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=5 messageTag=13 status=SUCCESS]
[2024-05-10 14:04:14] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=6 messageTag=14 status=SUCCESS]
[2024-05-10 14:04:14] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=7 messageTag=15 status=SUCCESS]
[2024-05-10 14:04:15] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=8 messageTag=16 status=NETWORK_BUSY]
[2024-05-10 14:04:15] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=9 messageTag=17 status=NETWORK_BUSY]
[2024-05-10 14:04:16] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=10 messageTag=18 status=NETWORK_BUSY]
[2024-05-10 14:04:16] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=11 messageTag=19 status=NETWORK_BUSY]
[2024-05-10 14:04:17] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=12 messageTag=20 status=NETWORK_BUSY]
[2024-05-10 14:04:17] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=13 messageTag=21 status=NETWORK_BUSY]
[2024-05-10 14:04:18] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=14 messageTag=22 status=NETWORK_BUSY]
[2024-05-10 14:04:19] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=15 messageTag=23 status=NETWORK_BUSY]
[2024-05-10 14:04:19] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=16 messageTag=24 status=NETWORK_BUSY]
[2024-05-10 14:04:20] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=17 messageTag=25 status=NETWORK_BUSY]
[2024-05-10 14:04:20] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=18 messageTag=26 status=NETWORK_BUSY]
[2024-05-10 14:04:21] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=19 messageTag=27 status=NETWORK_BUSY]
[2024-05-10 14:04:21] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=20 messageTag=28 status=SUCCESS]
[2024-05-10 14:04:21] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=21 messageTag=29 status=SUCCESS]
[2024-05-10 14:04:21] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=22 messageTag=30 status=NETWORK_BUSY]
[2024-05-10 14:04:22] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=23 messageTag=31 status=NETWORK_BUSY]
[2024-05-10 14:04:22] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=24 messageTag=32 status=NETWORK_BUSY]
[2024-05-10 14:04:23] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=25 messageTag=33 status=NETWORK_BUSY]
[2024-05-10 14:04:23] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=26 messageTag=34 status=NETWORK_BUSY]
[2024-05-10 14:04:24] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=27 messageTag=35 status=NETWORK_BUSY]
[2024-05-10 14:04:24] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=28 messageTag=36 status=NETWORK_BUSY]
[2024-05-10 14:04:25] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=29 messageTag=37 status=NETWORK_BUSY]
[2024-05-10 14:04:25] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=30 messageTag=38 status=NETWORK_BUSY]
[2024-05-10 14:04:26] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=31 messageTag=39 status=NETWORK_BUSY]
[2024-05-10 14:04:26] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=32 messageTag=40 status=SUCCESS]
[2024-05-10 14:04:26] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=33 messageTag=41 status=SUCCESS]
[2024-05-10 14:04:26] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=34 messageTag=42 status=SUCCESS]
[2024-05-10 14:04:26] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=35 messageTag=43 status=SUCCESS]
[2024-05-10 14:04:26] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=36 messageTag=44 status=SUCCESS]
[2024-05-10 14:04:26] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=37 messageTag=45 status=SUCCESS]
[2024-05-10 14:04:27] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=39 messageTag=47 status=NETWORK_BUSY]
[2024-05-10 14:04:27] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=40 messageTag=48 status=NETWORK_BUSY]
[... cut ...]
[2024-05-10 14:05:16] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=163 messageTag=42 status=NETWORK_BUSY]
```

</details>

That's ~160 broadcasts in roughly 1 minute.... (_broadcast/multicast, same limit applies_) So averaged that's 2.67, when the limit is 0.89.
_Definitely too much._ 😅 

I've been wanting to run some tests on this, see how we could maybe "optimize" it (though definitely not to that extent), but haven't had the time yet, plus I don't have enough devices in my testnet to run this effectively at the moment...

wastez, 2024-05-10:
@Nerivec 

Ok i can understand that this is much but then z2m groups are nothing to change the colors.
If i use home assistant groups then it is working without problems. (clear no multicasting)
Can i help you with the tests?

diegocjorge, 2024-05-14:
I have the same issue. I have a total of 95 devices, with 57 of them organized into 14 groups (each device belongs to only one group).

My log is full of errors such as:
```
error: zh:ember: Delivery of MULTICAST failed for '65533' [apsFrame={"profileId":260,"clusterId":768,"sourceEndpoint":1,"destinationEndpoint":255,"options":256,"groupId":13,"sequence":145} messageTag=32]
info: zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for '5742'.
```

Rarely, I also encounter this:
```
info: zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for '22550'.
```

Is there anything I could do to improve the situation? I already use channel 25 and have my 2.4GHz WiFi on channel 1.

I have a spare Sonoff dongle-P. Do you think it would be better to split my network into two?

Let me know if you would like for me to send some logs.

Nerivec, 2024-05-14:
@diegocjorge That would not be the same problem. Yours are failing because it looks like the network isn't very stable, and it can't reach devices properly all the time. Try putting your logs (`info` level is enough) through this little tool I made https://nerivec.github.io/z2m-ember-helper/ and see what you get (you need several hours worth of logs to get decent feedback, the more the better).

diegocjorge, 2024-05-16:
> @diegocjorge That would not be the same problem. Yours are failing because it looks like the network isn't very stable, and it can't reach devices properly all the time. Try putting your logs [...]

Thanks @Nerivec, I ran the tool and encountered a few issues:

1. **Broadcast Table Full Detected** - This can indicate the network is relying too heavily on broadcasts (messages to the whole network or to groups): Could this be why some of my grouped lights don’t always change color correctly?

2. **ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE**: These errors are mostly occurring with Gledopto’s GL-C-001P (5-in-1 Smart LED Controller).

3. **ROUTE_ERROR_ADDRESS_CONFLICT**: The tool detected 26 instances of this error, but the device hex does not match any device in my network. Do you have any idea what could be causing this? Could it be an old device still lingering in the database?

4. **CCA Failure Count is High**: This is odd because I ran an RF environment scan with my Ubiquiti AP, and it reported that WiFi channel 13 (frequency 2462~2482 MHz) has only 13% utilization and -87 dBm interference. Therefore, Zigbee channel 25 should be mostly clear, and CCA failure shouldn’t be so high.

There might be something broken with my database. I’m considering completely removing Z2M files and starting over from scratch. I'm just pondering whether to keep everything in one network or split it with my old Sonoff Dongle-P. Any suggestions?

Nerivec, 2024-05-16:
1. It gets problematic if you then see `NETWORK_BUSY` returned for messages. The broadcast table can hold 15 messages (standard size), you need to reach over that limit before it starts actually being a problem. [See "Broadcast table" if you want more details](https://community.silabs.com/s/article/how-are-different-tables-managed-in-emberznet-x?language=en_US)
2. Some of these is okay, too many can indicate a problem with a router/an area (depends where they originate).
3. This is weird if the device doesn't match anything in your network (doesn't involve the Z2M database though, this is detected by the adapter directly).
4. The count has to be at least 10 times higher than the "ideal" before you see this message (15 times if it's red), so something appears to be messing with your 2.4GHz band. _Although the "ideal" is from a network with nearly non-existent interferences._ What's the actual average per device (check the top table, it's near the right-end)?

Are you using the default Z2M network config (i.e. didn't used `GENERATE` when you created the network originally)?
Any chance there's another Zigbee network around you?

_Find me on Discord if you can, when you have some time, we can run some tests, figure out what's best. It should be fine with one network, but you could put "problem routers/devices" in another one for example so they don't mess with the rest._

wastez, 2024-05-25:
@Nerivec 
Cool I saw you made a special configuration file for the adapter, so the problem could be solved. 
man’s I saw there is a special multicast option for it. 
So does that mean the adapter driver now knows the difference between multicast and broadcast?

Nerivec, 2024-05-25:
That PR was not for this issue. It's only to allow very specific cases to configure memory allocation slightly differently (for example, a very large network might allocate more towards `SOURCE_ROUTE_TABLE_SIZE`). Since each has a cost (see [SiLabs UG100 - 2.3.1](https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf)), and networks can have very different purposes, a generic allocation, although pretty well tested by now, is not ideal for all cases, hence the ability to configure some of it with that PR, for more advanced users. **As mentioned in the docs PR, these configs can easily break a network if improperly handled.** These configs have always been there, just hardcoded.

`BROADCAST_TABLE_SIZE`: The maximum number of broadcasts during a single broadcast timeout period.
`MULTICAST_TABLE_SIZE`: The maximum number of multicast groups that the device may be a member of.
Note that it's not advised to change the default table size (15) for broadcasts since it might bring more problems, without also having full control of that same setting on your routers, which you don't have, since you don't use only homemade devices where you control the firmware.

_Do not experiment on a live network, especially if you don't **fully** understand how a setting affects a network._

wastez, 2024-05-25:
@Nerivec 
No there is no other network, the problem only appears if I use the color pallete to change colors, there are much colors which are changed in a short time. 
But then nobody thought on this case while using multicasts for lights. 
Maybe i use just use Home Assistant groups for this case. 
Or forget z2m groups for my color lights.

Nerivec, 2024-05-25:
Yes, spamming is always a big issue. Zigbee groups have their advantages and their inconveniences for sure, like most of the Zigbee specification... _Broadcasts definitely weren't meant for sending updates at huge rates._

In the color/dimmer scenario, the biggest issue I saw was the lack of a proper throttling mechanism in those devices, which means they just send every darn change along the way. There are some plans to implement such a mechanism on Z2M side (though that will only affect commands sent _from Z2M_, since Z2M has no control over how fast commands are sent from devices...) but it's not exactly an easy task to try to maximize efficiency without hindering user experience, and staying in line with Zigbee spec, and covering all usage cases, and hundreds of devices... 😅

closed 2024-12-28 by github-actions[bot]