Koenkk/zigbee2mqtt #22387: ZNJLBL01LM show as Unknown after 1.37.0

clowncracker, 2024-05-02:
### What happened?

After updating to 1.37.0 from 1.36.1, the blinds no longer show Open/Closed, they only show as Unknown.

### What did you expect to happen?

I expect the blinds to show Closed or Open, not Unknown.

### How to reproduce it (minimal and precise)

Upgrade to 1.37.0 and I get this issue.  Downgraded back to 1.36.1 and the state came back.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus Gateway

### Setup

Add-on on Unraid VM

10Thirty, 2024-05-02:
My ZNJLBL01LM Is also showing the same issue on a UZG-01 with firmware 20240316. I also tried connecting it to an aqara hub to see if there was a newer firmware as [#22299](https://github.com/Koenkk/zigbee2mqtt/issues/22299#issuecomment-2075502488)  references fw 09-12-2023 but there isn't one available so it could be there is a newer revision that is slightly different or a difference between Chinese and EU ones.

cromelex, 2024-05-02:
As mention on my comment on https://github.com/Koenkk/zigbee2mqtt/issues/22397

> Same issue here. Not sure if this is with Zigbee2MQTT 1.37.0 or Home Assistant 2024.05.0
> 
> To be clear, Z2M seems to have all the info, but in Home Assistant the state of the cover.entity is "unknown"
> 
> { "battery": 43, "charging_status": false, "device_temperature": 23, "last_seen": "2024-05-02T11:14:40+01:00", "linkquality": 116, "motor_speed": "medium", "motor_state": "pause", "position": 36, "power_outage_count": 2, "running": false, "state": "OFF" }
> 
> ![image](https://private-user-images.githubusercontent.com/96779452/327393982-e4be99cd-721e-4883-889b-0100b034aed8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ2NDU4MjEsIm5iZiI6MTcxNDY0NTUyMSwicGF0aCI6Ii85Njc3OTQ1Mi8zMjczOTM5ODItZTRiZTk5Y2QtNzIxZS00ODgzLTg4OWItMDEwMGIwMzRhZWQ4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTAyVDEwMjUyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJjNWJiM2E5ZDc1ZTQ5NjVkNGVhYzkyYzVhYzRjMzM0ZTFiOTFiMGJlNTc3YjFmMjI2N2E4YTMwZmE5ZmRkMWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.1Mct6U41Ce9d8MDUjrhFNYDAOufNrgBNYBO4MS7aZS8)
> 
> Looking at some automations I had, the attribute in HASS was previously "**position**", but now it is "**current_position**".
> 
> Was this changed in 1.37.0 ? That could be the cause of this issue.

The attribute in HASS was previously "**position**", but now it is "**current_position**" for the cover.entity. I think this might be the issue.

frops83, 2024-05-02:
same problem here, hoping for a fix

Koenkk, 2024-05-02:
Pushed a fix, let me know if it works.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

golles, 2024-05-02:
> Pushed a fix, let me know if it works.
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Image `koenkk/zigbee2mqtt:latest-dev` (Last pushed 41 minutes ago by koenkk) fixes this issue for me!

Dillton, 2024-05-02:
> Hi @Koenkk, this is also a similar issue for the ZNCLDJ12LM, line 2626 in `src/lib/lumi.ts`. Should be 'stopped' and not 'stop'?
> 
> ```
> if (['ZNCLDJ12LM', 'ZNCLDJ14LM'].includes(model.model)) lookup = {0: 'closing', 1: 'opening', 2: 'stop'};
> ```

Hello @Koenkk I can confirm this for ZNCLDJ12LM. The state for this one is still opening or closing even when the motor is currently stopped. Please push fix for this also. Thanks in advance.

golles, 2024-05-03:
Besides everything working fine, the HA log has many entries like `WARNING (MainThread) [homeassistant.components.mqtt.cover] Payload is not supported (e.g. open, closed, opening, closing, stopped): pause`

Koenkk, 2024-05-04:
I will probably do a hotfix release in the coming days, but good that this is fixed now!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-04 by Koenkk

clowncracker, 2024-05-05:
@Koenkk I just tried using the dev branch and it looks like this issue is fixed, however I've run into another problem.  I'm now seeing this in my logs when I only open my blinds partway:

`Payload is not supported (e.g. open, closed, opening, closing, stopped): pause`

Should I open a new ticket or should we reopen this one since it's related to the update?

clowncracker, 2024-05-05:
> > I just tried using the dev branch and it looks like this issue is fixed, however I've run into another problem. I'm now seeing this in my logs when I only open my blinds partway:
> > `Payload is not supported (e.g. open, closed, opening, closing, stopped): pause`
> > Should I open a new ticket or should we reopen this one since it's related to the update?
> 
> @clowncracker What's the exact `zigbeeModel` of your device?

lumi.curtain.acn002

clowncracker, 2024-05-06:
> > > > I just tried using the dev branch and it looks like this issue is fixed, however I've run into another problem. I'm now seeing this in my logs when I only open my blinds partway:
> > > > `Payload is not supported (e.g. open, closed, opening, closing, stopped): pause`
> > > > Should I open a new ticket or should we reopen this one since it's related to the update?
> > > 
> > > 
> > > @clowncracker What's the exact `zigbeeModel` of your device?
> > 
> > 
> > lumi.curtain.acn002
> 
> And the exact commit/version of your Z2M install?

1.37.0-dev commit: [358e2ee](https://github.com/Koenkk/zigbee2mqtt/commit/358e2ee)

fir3drag0n, 2024-05-06:
With latest 1.37.0-dev commit: [358e2ee](https://github.com/Koenkk/zigbee2mqtt/commit/358e2ee) the state is again not showed anymore.

Koenkk, 2024-05-06:
> Payload is not supported (e.g. open, closed, opening, closing, stopped): pause

When is this logged?

clowncracker, 2024-05-06:
> > Payload is not supported (e.g. open, closed, opening, closing, stopped): pause
> 
> When is this logged?

I'm no longer having this issue.  I think that @burmistrzak was correct, I think the status was just cached.  Moving the blinds back to closed and then opening them again partway solved the issue.