Koenkk/zigbee2mqtt #22388: Cover toggle no longer works for zigbee2mqtt blinds

berserk88, 2024-05-02:
### What happened?

Cover toggle no longer works for zigbee2mqtt blinds. The current state and position of the blinds were showing as unknown. I deleted and readded the blinds, now they are correctly reported, however cover.toggle will only close the blinds, it no longer toggles to stop or open as it used to.

### What did you expect to happen?

The cover.toggle command should toggle through closing, stopping and opening the blinds on each run. 

### How to reproduce it (minimal and precise)

Run cover.toggle

### Zigbee2MQTT version

1.37

### Adapter firmware version

6.7.9.0 build 405

### Adapter

Sonos zigbee hub

### Setup

Addon for home assistant

DannyFeliz, 2024-05-02:
I can confirm this; for some reason, that didn't happen to me but to my parents.

golles, 2024-05-02:
Maybe duplicate of #22387 ?

rasika-jay, 2024-05-09:
Can confirm, I have the same issue with Aqara blinds. I also see 'Payload is not supported (e.g. open, closed, opening, closing, stopped):...' in HA logs.

closed 2024-12-06 by github-actions[bot]